platformio/platformio-core #4863: Add Sparkfun Thing Plus Matter - MGM240P support

ruddycanuck, 2024-02-21:
- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 10

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 6.1.13

### Description of problem
Arduino, Sparkfun and Silicon Labs have a new Matter project: https://github.com/SiliconLabs/arduino/tree/main/libraries/Matter

I would like to use VS Code to program the Sparkfun Things Plus Matter ( Model DEV-20270 ) dev board using the Silicon Labs developed Arduino Libraries. (Arduino compiling is painfully slow.)
https://www.sparkfun.com/products/20270

I cannot find this board in the Platformio device list.

#### Steps to Reproduce

1. In VS Code - Platformio select 'Import Arduino Project'
2. Attempt to find the Sparkfun Thing Plus Matter - MGM240P

### Actual Results
Product is not listed.

### Expected Results
Product listed and Platformio able to use the Matter libraries from Arduino

### Additional info
This board is available in Arduino - board info is from https://siliconlabs.github.io/arduino/package_arduinosilabs_index.json
The board comes with all of the appropriate Matter Arduino libraries for this Sparkfun and the Silicon Labs xG24 Explorer Kit dev boards.

LorbusChris, 2024-05-01:
This is the same family of MCUs that is going to be used in the upcoming Arduino Nano Matter (which uses the MGM240S) and which'll also be supported in https://github.com/SiliconLabs/arduino

MitchTalmadge, 2024-05-13:
More info about Arduino Nano Matter: https://store.arduino.cc/pages/nano-matter

LorbusChris, 2024-05-15:
This draft PR might be a good starting point for anyone looking to add support for the Sparkfun Thing Plus Matter and Arduino Nano Matter boards and the [Silabs Arduino core](https://github.com/SiliconLabs/arduino):

https://github.com/platformio/platform-siliconlabsefm32/pull/8

LorbusChris, 2024-05-18:
Official PlatformIO support for EFR32 / MGM240 seems to be in the works :tada: : https://github.com/SiliconLabs/arduino/issues/52#issuecomment-2117415159