platformio/platformio-core #4872: Platform.io detecting wrong platform on Windows

sghillie, 2024-03-11:
I am having an eerily similar bug to the issue reported in 2022, but in reverse. I am running on an intel processor, but platform.io is detecting it as amd64. 
https://github.com/platformio/platformio-core/issues/4353
Below is my console output

> Resolving OpenIMU300RI dependencies...

> Tool Manager: Installing platformio/toolchain-gccarmnoneeab @ >=1.60301.0,<1.80000.0

> UnknownPackageError: Could not find the package with 'platformio/toolchain-gccarmnoneeab @ >=1.60301.0,<1.80000.0' requirements for your system 'windows_amd64'

sghillie, 2024-03-11:
PS C:\Users\shaun\Documents\GitHub\IMU300RI> pio system info
--------------------------  ------------------------------------------------------
PlatformIO Core             6.1.14b1
Python                      3.11.7-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1252
PlatformIO Core Directory   C:\Users\shaun\.platformio
PlatformIO Core Executable  C:\Users\shaun\.platformio\penv\Scripts\platformio.exe
Python Executable           C:\Users\shaun\.platformio\penv\Scripts\python.exe    
Global Libraries            0
Development Platforms       1
Tools & Toolchains          4
--------------------------  ------------------------------------------------------

sghillie, 2024-03-11:
no vm is running, and in python this is what it detects
```
PS C:\Users\shaun\Documents\GitHub\IMU300RI> python
Python 3.11.7 (tags/v3.11.7:fa7a6f2, Dec  4 2023, 19:24:49) [MSC v.1937 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import platform
>>> platform.architecture()
('64bit', 'WindowsPE')
>>> platform.machine()
'AMD64'
>>> platform.system() 
'Windows'
>>>
```

Jason2866, 2024-03-14:
This is correct.

closed 2024-03-14 by ivankravets