platformio/platformio-core #4884: Can't have configuration sections that are empty in 6.1.6+

nomis, 2024-03-24:
What kind of issue is this?

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Ubuntu 20.04

**PlatformIO Version** (`platformio --version`): 6.1.14

### Description of problem

I have configuration sections that are empty because I have a common "app" repository for all the boilerplate code that I want in every application and use `extra_configs` to load it. In order to insert extra dependencies into all environments I have empty sections in the "app" repository config file used by all the environments that I can then use to override part of the configuration.

This worked on 6.1.5 but changes made in 17360b0ed2fc213200571d275b23b6d34bba0c10 for #4480 have broken this in 6.1.6.

#### Steps to Reproduce

1. `git clone https://github.com/nomis/ggroohauga` (9b900c59938a57d4a1ce52ea1f0fe93b5960c956)
3. `git submodule update --init`
4. `platformio run`

### Actual Results

Can't build the project because it doesn't like the empty section:
`InvalidProjectConfError: Invalid 'ggroohauga/platformio.ini' (project configuration file): 'No option 'lib_deps' in section: 'app:common''`

### Expected Results

Both environments build successfully.

### Additional info

It is no longer possible to use 6.1.5 because the version of tool-scons it requires has been deleted from the registry.

nomis, 2024-03-24:
Some of these errors occur with 6.1.5 too, but then 6.1.14 makes them worse.

For example:
1. `git clone https://github.com/nomis/ggroohauga`
2. `git checkout 6a28ad362d12525974ee4ab44d73bcf59afbb9fb`
3. `git submodule update --init`
4. `platformio run`

```
$ platformio run
InvalidProjectConfError: Invalid 'ggroohauga/platformio.ini' (project configuration file): 'No option 'build_flags' in section: 'app:native_common''
```

That doesn't happen with 6.1.5, but this does:
`Error: Invalid 'ggroohauga/platformio.ini' (project configuration file): 'No option 'build_flags' in section: 'app:common''`

When I add `build_flags` I then get:
`Error: Invalid 'ggroohauga/platformio.ini' (project configuration file): 'No option 'lib_deps' in section: 'app:common''`

When I add `lib_deps` I then get:
`Error: Invalid 'ggroohauga/platformio.ini' (project configuration file): 'No option 'extra_scripts' in section: 'app:common''`

Eventually after adding `extra_scripts` it will then build.

I'm unable to test versions prior to 6.1.5 because those versions may have dependencies that are no longer available.

ivankravets, 2024-03-26:
Please provide a simple project to reproduce this issue. You can attach it via ZIP in the comment to this issue.

nomis, 2024-03-29:
[platformio-core-4884.zip](https://github.com/platformio/platformio-core/files/14808782/platformio-core-4884.zip)

This won't build with 6.1.5 unless everything in the `platformio.ini` section `[app:common]` is uncommented:
`Error: Invalid 'platformio-core-4884/platformio.ini' (project configuration file): 'No option 'build_flags' in section: 'app:common''`
`Error: Invalid 'platformio-core-4884/platformio.ini' (project configuration file): 'No option 'lib_deps' in section: 'app:common''`
`Error: Invalid 'platformio-core-4884/platformio.ini' (project configuration file): 'No option 'extra_scripts' in section: 'app:common''`
```
[app:common]
build_flags = ${env.build_flags}
lib_deps = ${env.lib_deps}
extra_scripts = ${env.extra_scripts}
```

With 6.1.14 it looks like it's trying to enforce this check for unreferenced sections too but it's only complaining about one of the options:
`InvalidProjectConfError: Invalid 'platformio-core-4884/platformio.ini' (project configuration file): 'No option 'build_flags' in section: 'app:native_common''`
```
[app:native_common]
build_flags = ${env.build_flags}
```

Nothing in the [documentation](https://docs.platformio.org/en/latest/projectconf/sections/env/options/advanced/extends.html) states that any custom section must have specific options so something unexpected is happening with the environments/use of extends.

ivankravets, 2024-04-24:
Sorry, I can't understand your logic. The project which you shared has broken INI configuration:

```
        for s in self.sections():
            print(s)
            print(self.items(s))
````

not all items can be resolved. The internal Python ConfigParser produces ParsingError.

We can't calculate project checksum. See https://github.com/platformio/platformio-core/blob/develop/platformio/project/helpers.py#L98

closed 2024-04-24 by ivankravets

nomis, 2024-04-24:
Ok, I see what happened now.

Use of those variables is necessary to allow them to be modified in the main `platformio.ini`, and when I've tried to upgrade to a newer PlatformIO I also updated older projects to the latest version of my "app" submodule which had new variables which I have to define or it won't work, but it's been a couple of years since I've done anything with these and forgot how it was supposed to work.

ivankravets, 2024-04-24:
Could we mark this issue as resolved?

nomis, 2024-04-24:
Yes