platformio/platformio-core #4885: PIO Core 6.1.14 Issue compiling with EspAsyncWebServer

MicroframeDev1, 2024-03-25:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Win 11 Pro 23H2

**PlatformIO Version** (`platformio --version`): 6.1.14

### Description of problem
Project compiled fine on Friday.  I opened the project today and PIO automatically installed a new version without my consent.  It appears PIO Core has updated to 6.1.14.  Now my code won't compile.  I suspect this is related to changes to LDF.

#### Steps to Reproduce

1. Unsure how to provide a minimal example.  Our code is quite complex.

### Actual Results
Compile fails with:
.pio/libdeps/MFSB-DB/ESPAsyncWebServer-Microframe/src/AsyncEventSource.h:26:10: fatal error: AsyncTCP.h: No such file or directory
AsyncTCP library is installed.

### Expected Results
Linker finds installed libraries and compiles correctly.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env]
board = featheresp32
monitor_speed = 115200
;platform = espressif32
platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.02.00/platform-espressif32.zip
framework = arduino
check_skip_packages = yes
monitor_filters = time, colorize, esp32_exception_decoder
lib_deps = 
    ;https://docs.platformio.org/en/latest/projectconf/sections/env/options/library/lib_deps.html
    https://github.com/MicroframeDev1/ESPAsyncWebServer.git#v3.1.0                  ;Was esphome/ESPAsyncWebServer-esphome@2.1.0
	bblanchon/ArduinoJson@^6.19.3                                                   ;Allows 6.x.x
board_build.partitions = partitions_custom.csv

[env:MFSB-Release]
build_flags = 
	-D CORE_DEBUG_LEVEL=1

[env:MFSB-DB]
build_type = debug
build_flags =
	-D CORE_DEBUG_LEVEL=3
	-D APPDEBUG=1
	-Wl,--print-memory-usage
```
Compile fails with both MFSB-DB and MFSB-Release.
Doing a Full Clean and then Build results in the same failure.

**Source file to reproduce issue:**
```cpp
Unable to share proprietary code.
```

### Additional info
Build log:
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/featheresp32.html
PLATFORM: Espressif 32 (2023.2.0) > Adafruit ESP32 Feather
[... 125 lines cut ...]
In file included from .pio/libdeps/MFSB-DB/ESPAsyncWebServer-Microframe/src/WebServer.cpp:21:
.pio/libdeps/MFSB-DB/ESPAsyncWebServer-Microframe/src/ESPAsyncWebServer.h:33:10: fatal error: AsyncTCP.h: No such file or directory

******************************************************************
* Looking for AsyncTCP.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:AsyncTCP.h"
* Web  > https://registry.platformio.org/search?q=header:AsyncTCP.h
*
******************************************************************

 #include <AsyncTCP.h>
          ^~~~~~~~~~~~
compilation terminated.
compilation terminated.
compilation terminated.
*** [.pio\build\MFSB-DB\libd7c\ESPAsyncWebServer-Microframe\WebHandlers.cpp.o] Error 1
*** [.pio\build\MFSB-DB\libd7c\ESPAsyncWebServer-Microframe\WebRequest.cpp.o] Error 1
*** [.pio\build\MFSB-DB\libd7c\ESPAsyncWebServer-Microframe\WebResponses.cpp.o] Error 1
*** [.pio\build\MFSB-DB\libd7c\ESPAsyncWebServer-Microframe\WebServer.cpp.o] Error 1
============================================================================================================== [FAILED] Took 6.95 seconds ==============================================================================================================

Environment    Status    Duration
-------------  --------  ------------
MFSB-DB        FAILED    00:00:06.953

MicroframeDev1, 2024-03-25:
Note we branched ESPAsyncWebServer and AsyncTCP to our own repository to workaround a bad update in AsyncTCP.cpp released by ESPHome.  We've been using our branched/patched version for 2 months without issues.

ivankravets, 2024-03-25:
Could you provide a simple project (ZIP archive attached to this issue as a comment) to reproduce the problem?

MicroframeDev1, 2024-03-25:
Adding `lib_ldf_mode = deep+` to build flags makes no difference.  It still can't find AsyncTCP.h.

MicroframeDev1, 2024-03-25:
> Could you provide a simple project (ZIP archive attached to this issue as a comment) to reproduce the problem?

I'll attempt to strip the project down.

MicroframeDev1, 2024-03-25:
Update: it appears the failing project references a library in platformio.ini but doesn't call it in code yet.  I have a similar project that does use `#include <ESPAsyncWebServer.h>` and it compiles properly.

MicroframeDev1, 2024-03-25:
[MFSB2.zip](https://github.com/platformio/platformio-core/files/14748329/MFSB2.zip)

Note the project references a custom branch of ESPAsyncWebServer but doesn't actually call the include header.  This project will fail to build.

ivankravets, 2024-03-26:
Thanks for the provided project. The issue was linked with the nested dependency declared via VCS repository. See https://github.com/MicroframeDev1/ESPAsyncWebServer/blob/v3.1.0/library.json#L28

The solution is to avoid using VCS declarations, especially for nested dependencies. You can try a direct branch https://github.com/MicroframeDev1/ESPAsyncWebServer , it contains valid dependencies declared via semantic versioning.

ivankravets, 2024-03-26:
You can also re-test this issue via `pio upgrade --dev`.

closed 2024-03-26 by ivankravets

MicroframeDev1, 2024-03-26:
`pio upgrade --dev` failed but gave instructions on how to upgrade from command line.  Success.  Now running PIO Core 6.1.15a1.  Project compiles correctly when referencing our branch of ESPAsyncWebServer.  Thank you for the quick fix!

commit c475578d (2024-03-26): Resolved an issue where the |LDF| couldn't locate a library dependency declared via version control system repository // Resolve #4885