platformio/platformio-core #4895: AddPreAction doesn't work as expected

briandilley, 2024-04-10:
Using `AddPreAction("buildprog", ...)` does not result in the action being beformed BEFORE compilation as expected. In fact, it seems impossible to have something happen _before_ compilation (ie: generating code, etc.). Here's a sample script that generates a website that is built into the program:

```python

import shutil

Import("env")

def build_web_progmem(source, target, env):
    print("running build_web_progmem")
    PROJECT_DIR = env.subst("$PROJECT_DIR")
    env.Execute(f"npm --prefix={PROJECT_DIR}/../web install")
    env.Execute(f"npm --prefix={PROJECT_DIR}/../web run awot-static")

env.AddCustomTarget(
    name="build_web",
    dependencies=None,
    actions=[build_web_progmem],
    title="Build Web",
    description="Build the web interface"
)
env.AddPreAction("buildprog", build_web_progmem)

```

ivankravets, 2024-04-18:
For the "pregeneration" before building stage, please use PRE script https://docs.platformio.org/en/latest/scripting/actions.html

closed 2024-04-18 by ivankravets

jmamma, 2025-07-11:
> For the "pregeneration" before building stage, please use PRE script https://docs.platformio.org/en/latest/scripting/actions.html

The PRE script doesn't seem to have access to any of the environment variables.