platformio/platformio-core #4896: Unable to remove library from VS Code

gojimmypi, 2024-04-11:
I'm logged into Windows 11 Pro 22H2 (22621.3447) as a regular user (not a member of local `Administrators` group) and using PlatformIO Core 6.1.14·Home 3.4.4 for VS Code 1.88.0, specifically:

[image]

I'm working on wolfSSL for PlatformIO as described in https://github.com/platformio/platformio-registry/issues/85

The _install_ of the wolfSSL library works well, but the _removal_ (or update) from VS Code fails with this error:

[image]

Here's the full error:

```
WARNING: This command is deprecated and will be removed in the next releases.
Please use `pio pkg uninstall` instead.
Resolving esp32dev dependencies...
Removing wolfSSL @ 5.6.6-test7
wolfSSL@5.6.6-test7 has been removed!
[WinError 5] Access is denied: 'C:\\Users\\gojimmypi\\Documents\\PlatformIO\\Projects\\ESP_IDF_Hello_World\\.pio\\libdeps\\esp32dev\\wolfSSL\\src\\src'
Please manually remove the file `C:\Users\gojimmypi\Documents\PlatformIO\Projects\ESP_IDF_Hello_World\.pio\libdeps\esp32dev\wolfSSL\src\src`
[WinError 5] Access is denied: 'C:\\Users\\gojimmypi\\Documents\\PlatformIO\\Projects\\ESP_IDF_Hello_World\\.pio\\libdeps\\esp32dev\\wolfSSL\\src\\wolfcrypt\\src\\port\\Espressif'
Please manually remove the file `C:\Users\gojimmypi\Documents\PlatformIO\Projects\ESP_IDF_Hello_World\.pio\libdeps\esp32dev\wolfSSL\src\wolfcrypt\src\port\Espressif`
[WinError 5] Access is denied: 'C:\\Users\\gojimmypi\\Documents\\PlatformIO\\Projects\\ESP_IDF_Hello_World\\.pio\\libdeps\\esp32dev\\wolfSSL\\src\\wolfcrypt\\src\\port'
Please manually remove the file `C:\Users\gojimmypi\Documents\PlatformIO\Projects\ESP_IDF_Hello_World\.pio\libdeps\esp32dev\wolfSSL\src\wolfcrypt\src\port`
[WinError 5] Access is denied: 'C:\\Users\\gojimmypi\\Documents\\PlatformIO\\Projects\\ESP_IDF_Hello_World\\.pio\\libdeps\\esp32dev\\wolfSSL\\src\\wolfcrypt\\src'
Please manually remove the file `C:\Users\gojimmypi\Documents\PlatformIO\Projects\ESP_IDF_Hello_World\.pio\libdeps\esp32dev\wolfSSL\src\wolfcrypt\src`
[WinError 5] Access is denied: 'C:\\Users\\gojimmypi\\Documents\\PlatformIO\\Projects\\ESP_IDF_Hello_World\\.pio\\libdeps\\esp32dev\\wolfSSL\\src\\wolfcrypt'
Please manually remove the file `C:\Users\gojimmypi\Documents\PlatformIO\Projects\ESP_IDF_Hello_World\.pio\libdeps\esp32dev\wolfSSL\src\wolfcrypt`
[WinError 5] Access is denied: 'C:\\Users\\gojimmypi\\Documents\\PlatformIO\\Projects\\ESP_IDF_Hello_World\\.pio\\libdeps\\esp32dev\\wolfSSL\\src\\wolfssl\\openssl'
Please manually remove the file `C:\Users\gojimmypi\Documents\PlatformIO\Projects\ESP_IDF_Hello_World\.pio\libdeps\esp32dev\wolfSSL\src\wolfssl\openssl`
[WinError 5] Access is denied: 'C:\\Users\\gojimmypi\\Documents\\PlatformIO\\Projects\\ESP_IDF_Hello_World\\.pio\\libdeps\\esp32dev\\wolfSSL\\src\\wolfssl\\wolfcrypt'
Please manually remove the file `C:\Users\gojimmypi\Documents\PlatformIO\Projects\ESP_IDF_Hello_World\.pio\libdeps\esp32dev\wolfSSL\src\wolfssl\wolfcrypt`
[WinError 5] Access is denied: 'C:\\Users\\gojimmypi\\Documents\\PlatformIO\\Projects\\ESP_IDF_Hello_World\\.pio\\libdeps\\esp32dev\\wolfSSL\\src\\wolfssl'
Please manually remove the file `C:\Users\gojimmypi\Documents\PlatformIO\Projects\ESP_IDF_Hello_World\.pio\libdeps\esp32dev\wolfSSL\src\wolfssl`
[WinError 5] Access is denied: 'C:\\Users\\gojimmypi\\Documents\\PlatformIO\\Projects\\ESP_IDF_Hello_World\\.pio\\libdeps\\esp32dev\\wolfSSL\\src'
Please manually remove the file `C:\Users\gojimmypi\Documents\PlatformIO\Projects\ESP_IDF_Hello_World\.pio\libdeps\esp32dev\wolfSSL\src`
[WinError 5] Access is denied: 'C:\\Users\\gojimmypi\\Documents\\PlatformIO\\Projects\\ESP_IDF_Hello_World\\.pio\\libdeps\\esp32dev\\wolfSSL'
Please manually remove the file `C:\Users\gojimmypi\Documents\PlatformIO\Projects\ESP_IDF_Hello_World\.pio\libdeps\esp32dev\wolfSSL`
```

I'm able to otherwise manually delete the library as noted in the error message, but that is of course not ideal.

See example staging versions: https://registry.platformio.org/libraries/wolfssl-staging/wolfSSL/versions

The example above was with [5.6.6-test6](https://registry.platformio.org/libraries/wolfssl-staging/wolfSSL?version=5.6.6-test6) installed, attempting to remove and upgrade.

The error seems related to permissions and/or antivirus (not the library itself), as I also see this error:

[image]

Does PlatformIO modify attributes at registry library install time?

This is not limited to my wolfSSL library, as I also see the error for this Adafruit library:

[image]

Thanks for taking a look at this.

ivankravets, 2024-04-11:
Do you use any antivirus tools?

gojimmypi, 2024-04-12:
Update on this topic. It appears that the anti-virus software is complaining about exactly what the message indicates: "_attempted to modify attributes_". The odd thing isthat the operation seems to have been partially successful, and is also the problem with library _removal_. 

Specifically, it appears the `platformio.exe` has a good intention to flag things as read only. But it fails. The result in the read-only flag apparently neither set nor cleared. Note the checkbox for `read-only` status:

[image]

The checkbox looks like this when `read-only`:

[image]

and like this when not read-only:

[image]

At library removal time, all the _files_ are successfully removed, but as seen in the error message [above](https://github.com/platformio/platformio-core/issues/4896#issue-2238227101), the _directories_ are the things that were not deleted.

Perhaps this could be fixed with something as easy as a config or prompt asking to flag things as read-only or not? I suspect library removal would proceed as desired if the read-only flag had not been touched.

ivankravets, 2024-04-18:
Could you provide an output of
```
pio system info
```

See the related code https://github.com/platformio/platformio-core/blob/develop/platformio/fs.py#L212

gojimmypi, 2024-04-23:
Hi @ivankravets 

I've confirmed that if I manually flag all the files and all the directories in the wolfSSL directory as R/W, later removal and/or upgrade works just fine. 

Specifically this directory:

```
C:\\Users\\gojimmypi\\Documents\\PlatformIO\\Projects\\ESP_IDF_Hello_World\\.pio\\libdeps\\esp32dev\\wolfSSL\\
```

> Could you provide an output of `pio system info`

```
C:\Users\gojimmypi>pio system info
--------------------------  ----------------------------------------------------------
PlatformIO Core             6.1.14
Python                      3.11.7-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1252
PlatformIO Core Directory   C:\Users\gojimmypi\.platformio
PlatformIO Core Executable  C:\Users\gojimmypi\.platformio\penv\Scripts\platformio.exe
Python Executable           C:\Users\gojimmypi\.platformio\penv\Scripts\python.exe
Global Libraries            0
Development Platforms       1
Tools & Toolchains          19
--------------------------  ----------------------------------------------------------

```

We are providing two different Official wolfSSL libraries: standard and another specifically for Arduino:

- [wolfssl/wolfSSL](https://registry.platformio.org/libraries/wolfssl/wolfSSL)

- [wolfssl/Arduino-wolfSSL](https://registry.platformio.org/libraries/wolfssl/Arduino-wolfSSL)

There are also two different versions: the stable release versions (above) and these staging updates, with the latest post-release changes.

- [wolfssl-staging/wolfSSL](https://registry.platformio.org/libraries/wolfssl-staging/wolfSSL)

- [wolfssl-staging/Arduino-wolfSSL](https://registry.platformio.org/libraries/wolfssl-staging/Arduino-wolfSSL)