platformio/platformio-core #4900: Include path doesn't always contain the dependencies from a library.json file

sylque, 2024-04-19:
### Configuration

**Operating system**: Windows

**PlatformIO Version**: 6.1.14

### Description of problem

If you put this in a `library.json` file:
```
{
  "dependencies": {
    "SparkFun_BMI270": "https://github.com/sylque/SparkFun_BMI270_Arduino_Library.git"
  }
}
```
***and*** file `ImuBMI270.cpp:` of your library includes `SparkFun_BMI270_Arduino_Library.h`,

 ***but*** your project never includes directly `SparkFun_BMI270_Arduino_Library.h`,

***then*** you will get a compile error:
```
C:/Users/.../ImuBMI270.cpp:4:10: fatal error: SparkFun_BMI270_Arduino_Library.h: No such file or directory
```
The library is actually loaded and built, but the include path is not found.

### Expected Results

As I added "SparkFun_BMI270_Arduino_Library" explicitly in a manifest, I expect it to be reachable.

### Additional info

A workaround is to use `lib_ldf_mode = deep`.

ivankravets, 2024-04-24:
If you add `#include <SparkFun_BMI270_Arduino_Library.h>` to your `name_of_your_library.h or cpp` it will resolve all dependencies. Otherwise, you can set `lib_ldf_mode = deep` for your library. Docs:

- https://docs.platformio.org/en/latest/librarymanager/ldf.html#dependency-finder-mode
- https://docs.platformio.org/en/latest/manifests/library-json/fields/build/libldfmode.html

sylque, 2024-04-24:
Thanks @ivankravets .

Unfortunately, adding `#include <SparkFun_BMI270_Arduino_Library.h>` to `name_of_your_library.cpp` doesn't work for me.

My problem seems to be both simpler and stranger that what I originally described. Please see this repo of a minimal library:
https://github.com/sylque/Platformio_bugLib

And here is a simple project that includes this library:

platformio.ini:
```
[platformio]
default_envs = esp32

[env:esp32]
framework = arduino
platform = espressif32
board = m5stick-c
lib_deps = 
  https://github.com/sylque/Platformio_bugLib.git
```
main.cpp:
```c++
#include <Arduino.h>
void setup() {
}
void loop() {
}
```
Compilation result:
```
Scanning dependencies...
Dependency Graph
|-- Platformio_bugLib @ 1.0.0+sha.24e33b8
Building in release mode
Compiling .pio\build\esp32\lib254\Platformio_bugLib\Platformio_bugLib.cpp.o
.pio/libdeps/esp32/Platformio_bugLib/Platformio_bugLib.cpp:1:10: fatal error: SparkFun_BMI270_Arduino_Library.h: No such file or directory
```

It makes no sense to me. I've just spent two hours deleting .pio folders, doing Clean, Full Clean, reinstalling platformio, trying with symlink:, file:, etc. Sometimes it has worked, most of the time it hasn't. Any clue is welcome.

ivankravets, 2024-04-24:
> main.cpp:

Have you added `#include <Platformio_bugLib.h>` to the one of the project files?

sylque, 2024-04-24:
I provided you with all the code I'm using. There's no hidden project file, the only one is `main.cpp`.

Or do you mean that I *should* add `#include <Platformio_bugLib.h>` to `main.cpp`? That wouldn't be good. I provide a library: it's up to the users of my library to decide if they want to include a file from my library or not. It should work the same in both cases.

Unfortunately, I cannot test whether adding `#include <Platformio_bugLib.h>` to `main.cpp` solves anything, because now it works properly! (without any change) The issue seems to occur randomly when you add or remove stuff in `lib_deps`. For example, I've just changed from `https://github.com/sylque/Platformio_bugLib.git` to `symlink://../../../bugs/Platformio_bugLib` (exact same library), and now it works! But I'm sure the issue will be back soon.

May I ask what's the behavior on your side with the code I provided?

valeros, 2024-04-24:
Hi @sylque, strange, but I cannot reproduce the issue. I took your `platformio.ini` and `main.cpp`, then just run the project. Although there was an issue with missing SPI and Wire libraries, but after adding them to `lib_deps` everything compiled just fine, here is the dependency graph from that project:
```
Dependency Graph
|-- Platformio_bugLib @ 1.0.0+sha.24e33b8
|   |-- SparkFun BMI270 Arduino Library @ 1.0.2+sha.09d6aaf 
|   |   |-- SPI @ 2.0.0
|   |   |-- Wire @ 2.0.0
|-- Wire @ 2.0.0
|-- SPI @ 2.0.0
```
Here is my test project if you're interested: [platformio-core-issue-4900.zip](https://github.com/platformio/platformio-core/files/15098167/platformio-core-issue-4900.zip)

Am I missing something?

sylque, 2024-04-24:
@valeros , here is the full output when I compile your project:
```
 *  Executing task in folder platformio-core-issue-4900: C:\Users\sylque\.platformio\penv\Scripts\platformio.exe run 

Processing esp32 (framework: arduino; platform: espressif32; board: m5stick-c)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/m5stick-c.html
PLATFORM: Espressif 32 (6.6.0) > M5Stick-C
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Platformio_bugLib @ 1.0.0+sha.24e33b8
|-- Wire @ 2.0.0
|-- SPI @ 2.0.0
Building in release mode
Compiling .pio\build\esp32\src\main.cpp.o
Building .pio\build\esp32\bootloader.bin
Generating partitions .pio\build\esp32\partitions.bin
Compiling .pio\build\esp32\lib210\Platformio_bugLib\Platformio_bugLib.cpp.o
.pio/libdeps/esp32/Platformio_bugLib/Platformio_bugLib.cpp:1:10: fatal error: SparkFun_BMI270_Arduino_Library.h: No such file or directory

*****************************************************************************************
* Looking for SparkFun_BMI270_Arduino_Library.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SparkFun_BMI270_Arduino_Library.h"
* Web  > https://registry.platformio.org/search?q=header:SparkFun_BMI270_Arduino_Library.h
*
*****************************************************************************************

 #include <SparkFun_BMI270_Arduino_Library.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
esptool.py v4.5.1
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Compiling .pio\build\esp32\libaff\Wire\Wire.cpp.o
*** [.pio\build\esp32\lib210\Platformio_bugLib\Platformio_bugLib.cpp.o] Error 1
==================================================================== [FAILED] Took 4.68 seconds ====================================================================

 *  The terminal process "C:\Users\Sylvain\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

valeros, 2024-04-24:
I believe your issue is probably a duplicate of https://github.com/platformio/platformio-core/issues/4885 that's already fixed in the dev branch. Please run `pio upgrade --dev` in the IDE terminal and try again.

sylque, 2024-04-24:
@valeros , I confirm the issue doesn't occur with the dev branch. Thanks a million!