platformio/platformio-core #4901: Silent failure when dependency path is incorrect in library.json

sylque, 2024-04-19:
### Configuration

**Operating system**: Windows

**PlatformIO Version**: 6.1.14

### Description of problem

If you put this in a `library.json` file:

```
{
  "dependencies": {
    "nonexistent_lib_1": "symlink:../hfzehfzeiohize",
    "nonexistent_lib_2": "file:../hrfiohgeuibz"
  }
}
```

PlatformIO will happily tell you:
```
Library Manager: Installing symlink:../hfzehfzeiohize
Library Manager: Installing file:../hrfiohgeuibz
```
even though none of those paths exist. This makes tracking path issues painful.

### Expected Results

I would expect:
```
Library Manager: Installing symlink:../hfzehfzeiohize
PackageException: Can not create a symbolic link for `../hfzehfzeiohize`, not a directory
```
or:
```
Library Manager: Installing symlink:../hfzehfzeiohize
VCSBaseException: VCS: Unknown repository type symlink:../hfzehfzeiohize
```

Mawasi1, 2026-09-18:
I reproduced this on current develop (6.2.1b1) and dug into the cause.

`symlink:../x` and `file:../x` aren't documented spellings (docs only show `symlink://` and `file://`), but the failure mode for them is bad in two different ways:

- `pio pkg install`: silently ignored. The spec parser sees one slash and splits `symlink:../x` into owner `symlink:..` / name `x`, so it becomes a registry lookup. That fails with `UnknownPackageError`, which `install_dependencies` swallows for deps with no `owner` key (the built-in-library guard from 346580d9).
- `pio run` with the same library under `lib/`: every build hard-fails with `UnknownPackageError: Could not find the package with 'symlink:../x' requirements`, since that loop keys on the parsed owner instead.

Separately, the correctly spelled `file://<missing dir>` crashes with a raw `FileNotFoundError` traceback and the "report this to the developers" banner, while `symlink://<missing dir>` already gives a clean `PackageException`.

I'd like to submit a small fix:
1. `PackageSpec._parse_owner`: skip the owner/name split when the text before the slash contains a colon. The string then reaches `install_from_uri` and fails with the existing `VCSBaseException: Unknown repository type symlink:../x`, consistently in both commands. Verified that normal specs (owner/name, owner/name@ver, name=owner/pkg, git/zip URLs, `file:///`, `symlink://`, bare names) parse identically.
2. `install_from_uri`: check the `file://` path exists and raise a `PackageException`, matching the symlink branch.
Plus offline tests for both.

ivankravets, 2026-09-18:
@Mawasi1, who are you? Have you ever heard/used the PlatformIO before?

ivankravets, 2026-09-18:
> "symlink:../hfzehfzeiohize"

Please check the docs on how to use symlinks https://docs.platformio.org/en/latest/core/userguide/pkg/cmd_install.html#local-folder

closed 2026-09-18 by ivankravets