platformio/platformio-core #4916: support for renesas risc-v mcu

gkarsai, 2024-05-19:
What kind of issue is this?
- [ ] **Feature Request**.
Renesas came out with a family of RISC-V MCU-s. It would be great to have support for these, in particular the evaluation board: https://www.renesas.com/us/en/products/microcontrollers-microprocessors/risc-v/fpb-r9a02g021-fpb-r9a02g021-risc-v-mcu-fast-prototyping-board
It seems most of the tools are already available (e.g. gcc toolchain).
Thanks!

ivankravets, 2024-07-06:
Duplicate of https://github.com/platformio/platformio-core/issues/4028

Please vote ( 👍  ) for https://github.com/platformio/platformio-core/issues/4028

closed 2024-07-06 by ivankravets

trombonehero, 2024-12-10:
#4028 seems to be about Renesas-RA boards, i.e., those with Arm Cortex processors. The Renesas RISC-V boards are a very different family that seems to have gotten lost over on #4028... should they have a different issue @ivankravets?