platformio/platformio-core #4918: Parse error in Azure from junit output

merijnrats, 2024-05-21:
What kind of issue is this?
- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

### Configuration

**Operating system**:
Windows
**PlatformIO Version**:
6.1.15
### Description of problem
there seems to be a bug with the formatting of the junit-output, there aren't any new lines or spacings and because of that Azure can't parse the xml.

#### Steps to Reproduce
1. pio test  --junit-output-path .

### Actual Results
[image]

### Expected Results
[image]

### Additional info
I have a fix and I will try to create merge request as suggestion for devs

ivankravets, 2024-05-21:
Spaces should not be a cause of the issue. Could you debug in depth this issue and find the problem?

merijnrats, 2024-05-22:
[image]
[image]
I want to run the unittest with junit output to insert the test results into the pipeline results.  But I get a utf-8 parse error and since the current results are in 1 line I figured that would lead to the parse error which in my tested fixed it after adding newlines and whitespaces.

But i've got a workaround where I output it to json and run a custom script to parse it to junit which i can insert into Azure so it's fine,

ivankravets, 2024-05-22:
> utf-8 parse error

UTF-8 is de-facto standard encoding. If the system does not support it, you need to tune your system. You can also add extra step and convert from UTF-8 to "encoding supported by OS".

closed 2024-05-22 by ivankravets