platformio/platformio-core #4924: Very very slow download (stuck at 0%)

paulrouget, 2024-05-29:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to the
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues) repository

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

macos

**PlatformIO Version** (`platformio --version`):

PlatformIO Core, version 6.1.15

### Description of problem

Stuck on this:

```
Tool Manager: Installing espressif/toolchain-xtensa-esp32s3
Downloading  [------------------------------------]    0%
```

It seems to actually download, but at 1kb/s.

### Additional info

I've read about CDN issues. But I tried to connect via a VPN, still stuck at 0%.

ivankravets, 2024-05-29:
Hello. Currently, we don’t experience any issues. See https://community.platformio.org/t/platformio-packages-download-is-slow/40037/10

What is your country?

paulrouget, 2024-05-29:
> What is your country?

Taiwan.

paulrouget, 2024-05-30:
I tried download this: https://dl.registry.platformio.org/download/platformio/tool/toolchain-xtensa/2.100300.220621/toolchain-xtensa-windows_amd64-2.100300.220621.tar.gz with wget, and I get 3k/s.

paulrouget, 2024-05-30:
Using a VPN from Korea solved the issue.

ivankravets, 2024-05-30:
Could you try again [this link](https://dl.registry.platformio.org/download/platformio/tool/toolchain-xtensa/2.100300.220621/toolchain-xtensa-windows_amd64-2.100300.220621.tar.gz)?

paulrouget, 2024-05-30:
Yes it's fast. thanks

ivankravets, 2024-05-30:
Great! Sorry again for the issue and thanks for the report. If you experience this issue next time, please inform us. The situation with wars/conflicts is difficult, and it is normal to expect network issues.

closed 2024-05-30 by ivankravets