platformio/platformio-core #4925: ESP.getFlashChipMode(); crashes on esp32s3

Rai-Sp, 2024-05-30:
What kind of issue is this?

- [x ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to the
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues) repository

------------------------------------------------------------------

### Description of problem

```
/* In version 6.7.0 of platform-espressif32, a call to ESP.getFlashChipMode();
   crashes on esp32s3. This was not the case in previous versions.
   The problem seems to be linked to the newly introduced file
   .....sdk/esp32s3/include/soc/esp32s3/include/soc/spi_reg.h and its usage.
   Unfortunately, my skills in MACRO++ programming are too poor
   to make a better guess. A small program for demonstration follows:

The program, executed on an ESP32-S3-DevKitC-1, delivers the following output:    
Chip: ESP32-S3
Revision: 0.1
SDK: v4.4.7-dirty
Incorrect return value = 0x8
Correct return value = 0x60003008

In case the comment is removed from the line with "ESP.getFlashChipMode();"
the output is:
----------------------------------------------------------------------------------
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x42021950  PS      : 0x00060630  A0      : 0x82001773  A1      : 0x3fcebbf0
A2      : 0x00000008  A3      : 0x67ab6cc2  A4      : 0x67ab6cc2  A5      : 0x3fcebbac
A6      : 0x3fc94d0c  A7      : 0x3fcebbac  A8      : 0x82001e46  A9      : 0x3fcebb50
A10     : 0x00000012  A11     : 0x3fcebbac  A12     : 0x00000012  A13     : 0x3fcebbf0
A14     : 0x3fcebb70  A15     : 0x00000008  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000008  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffd

Backtrace: 0x4202194d:0x3fcebbf0 0x42001770:0x3fcebc10 0x420034e6:0x3fcebc50
  #0  0x4202194d in EspClass::getFlashChipMode() at /home/user123/.platformio/packages/framework-arduinoespressif32/cores/esp32/Esp.cpp:356
  #1  0x42001770 in setup() at src/main.cpp
  #2  0x420034e6 in loopTask(void*) at /home/user123/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp
----------------------------------------------------------------------------------

   Any idea, how to fix the problem?
*/

#include <Arduino.h>
#include <soc/spi_reg.h>

void setup() {
  Serial.begin(115200);
  delay(3000);
  Serial.printf("Chip: %s\n", ESP.getChipModel());
  esp_chip_info_t chip_info;
  esp_chip_info(&chip_info);
  uint16_t rev = chip_info.full_revision;
  Serial.printf("Revision: %u.%u\n", rev / 100, rev % 100);
  Serial.printf("SDK: %s\n", ESP.getSdkVersion());

  // ESP.getFlashChipMode();   // Would crash in the new version at execution.
  // For esp32s3 this would contain:
  //   uint32_t spi_ctrl = REG_READ(SPI_CTRL_REG(0));
  //   To see, what SPI_CTRL_REG(0) returns in the new version:
  Serial.printf("Incorrect return value = 0x%X\n", SPI_CTRL_REG(0));
  // This prints: "Incorrect return value = 0x8". esp32s3 cannot read from address 0x8.
  // SPI_CTRL_REG is defined as:
  //   #define SPI_CTRL_REG(i) (REG_SPI_BASE(i) + 0x8)
  // In the new version we have:
  //   #define REG_SPI_BASE(i) (((i)>=2) ? (DR_REG_SPI2_BASE + (i-2) * 0x1000) : (0))
  //     (at .....sdk/esp32s3/include/soc/esp32s3/include/soc/spi_reg.h: line 122)
  // before it was:
  //   #define REG_SPI_BASE(i) (DR_REG_SPI1_BASE + (((i)>1) ? (((i)* 0x1000) + 0x20000) : (((~(i)) & 1)* 0x1000 )))
  //     (at ...../sdk/esp32/include/soc/esp32/include/soc/spi_reg.h: line 19)
  // So, go back to the old version:
  #undef REG_SPI_BASE
  #define REG_SPI_BASE(i) (DR_REG_SPI1_BASE + (((i)>1) ? (((i)* 0x1000) + 0x20000) : (((~(i)) & 1)* 0x1000 )))
  Serial.printf("Correct return value = 0x%X", SPI_CTRL_REG(0));
  // The NEWLY INTRODUCED FILE .....sdk/esp32s3/include/soc/esp32s3/include/soc/spi_reg.h contains at line 122:
  //   #define REG_SPI_BASE(i) (((i)>=2) ? (DR_REG_SPI2_BASE + (i-2) * 0x1000) : (0))
  //   For i == 0 result is 0!
  //   which delivers the wrong address in this case. By the way: MACRO++ is not my favorite programming language!
}

void loop() { delay(100); }

/* platformio.ini:
[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino

upload_port = /dev/ttyUSB8
upload_speed = 921600
monitor_port = /dev/ttyUSB8
monitor_speed = 115200

monitor_filters =
  default
  esp32_exception_decoder
*/

thelastoutpostworkshop, 2024-06-01:
It crash also using ESP32 Espressif Arduino framework with ESP32-S3

ivankravets, 2024-06-03:
Please forward issue to https://community.platformio.org/c/development-platforms/17

craiglink, 2024-06-09:
opened this issue with Espressif https://github.com/espressif/arduino-esp32/issues/9816