platformio/platformio-core #4926: Dependencies are being installed regardless of platform in some cases

nicklasb, 2024-06-01:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to the
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues) repository

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [X ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: MacOS

**PlatformIO Version** (`platformio --version`): 6.1.15

### Description of problem
```

 "dependencies": [
        {
            "name": "esp_modem",
            "version": "https://github.com/RobustoFramework/Robusto/releases/download/v1.0.0/esp_modem_1.1.0.zip",
            "platforms": ["espressif32"]
        },
        {
            "name": "lvgl/lvgl",
            "version": "~8.3.0",
            "platforms": ["espressif32"]

        },
        {
            "name": "esp_lvgl_port",
            "version": "https://github.com/RobustoFramework/Robusto/releases/download/v1.0.0/esp_lvgl_port_1.4.0.zip",
            "platforms": ["espressif32"]

        },
        {
            "name": "RadioLib",
            "version": "https://github.com/jgromes/RadioLib.git#6.4.2",
            "platforms": ["espressif32"]

        }
    ],
```
Similar to https://github.com/platformio/platformio-core/issues/4338 , dependencies of my library is being installed even though I have specified on what platforms they should be installed for.

#### Steps to Reproduce

1. Use a Native or Arduino platform project
2. Add a dependency to the [Robusto framework](https://registry.platformio.org/libraries/robusto/robusto)
```
lib_deps = 
    robusto

```
4. Let the dependencies install

### Actual Results

Several of the above dependencies of Robusto that shouldn't install, are installed:  esp_lvgl_port, esp_modem and Radiolib
Intererestingly the lvgl/lvgl library is *not* installed. It differs by having a short version "~8.3.0".

My possibilities to test different variants is limited as I do not want to upload a hundred versions of the library to the registry..

### Expected Results

Only non-platform-specific dependencies should install.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
I certainly do not hope that the rest of the config interacts with the dependency loading. 
The extra script just adds paths, omitted.
```ini

[env:attiny85]
platform = atmelavr
board = attiny85
framework = arduino

upload_protocol = stk500v1
upload_flags =
    -P$UPLOAD_PORT
    -b$UPLOAD_SPEED
    -v
upload_port = /dev/cu.usbmodem141201
upload_speed = 19200
board_fuses.lfuse = 0xE2
board_fuses.hfuse = 0xDF
board_fuses.efuse = 0xFF

[env:testing]
platform = native
build_flags = 
    -I/usr/local/include/libelf/
    -L/usr/local/lib/
    -lelf

extra_scripts = post:extra_script.py
platform_packages =
    platformio/tool-simavr
    platformio/toolchain-atmelavr

lib_deps = 
    robusto                                                                        #<<<<< Here it is
    https://github.com/nicklasb/robusto-platformio.git

test_framework = unity
test_speed = 9600
test_testing_command =
    ${platformio.packages_dir}/tool-simavr/bin/simavr
    -m
    attiny85
    -f
    16000000L
    ${platformio.build_dir}/attiny85/firmware.elf

[test]
framework = unity
```

**Source file to reproduce issue:** N/A

nicklasb, 2024-08-25:
@ivankravets Not sure about the labels here, wouldn't this be some kind of bug?

fariouche, 2024-10-06:
I think I have a related bug too:

I have 2 envs, one for esp32 and an other for native.
I'm using the native env for testings by generating an executable for my pc.
In lib/ I have a library named lib/platform-native and an other named lib/platform-esp32
Inside platform-esp32/library.json I have
  "platforms": "espressif32",
  "frameworks": "espidf",

And in platform-native/library.json I have
"platforms": "native",
    "frameworks": "*",
    "dependencies": {
        "armmbed/mbedtls": "^2.23.0"
    }

Each env have a lib_deps= platform-esp32 or platform-native

When I build the native env, I see mbedtls being pulled and compiled.
The problem is that for esp32 I have the same behavior. I see that it ignored platform-native, but why does it pick mbedtls? (for the esp32 I'm using the mbedtls from espidf, and I see it being compiled too)

Now the weird part:
If I change the syntax of platform-native/library.json like so:
"dependencies": 
  [
    { "name" : "armmbed/mbedtls", "version" : "^2.23.0", "platforms" : ["native"]}
  ]

I get a warning that mbedtls dependency is ignored! (Warning: Ignored 'armmbed/mbedtls' dependency for 'platform-native' library)

I think something is wrong in the dependency management of libraries.

For now I removed the dependency from library.json and put it in the env directly