platformio/platformio-core #4934: `pio run -t compiledb` does not generate compilation database for tests

UtkarshVerma, 2024-06-11:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to the
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues) repository

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Arch Linux

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 6.1.14

### Description of problem
I use pio run -t compiledb to generate a compilation database that clangd can utilise. However, I noticed that the generated database does not include the path to unity.h and hence I get linter errors in my editor (neovim) whenever I try to include unity.h.

[image]

#### Steps to Reproduce

1. `pio run -t compiledb`

### Actual Results
Compilation database only generated for project root.

### Expected Results
Compilation database should also be generated for tests.

ivankravets, 2024-07-06:
Could try to add Unity to `lib_deps` as described here https://registry.platformio.org/libraries/throwtheswitch/Unity/installation ?

luizgustavojunqueira, 2025-03-21:
I have the same problem and have Unity on my lib_deps, the path is included in the compile_commands.json, but the problem persists.

For some reason if I move the test file to the project root then it works properly, or even if a include Unity in any file on the src or include directories. But if its on the test directory it doesn't work.