platformio/platformio-core #4940: LDF might skip or replace internal Arduino's libs when doing recursive resolution dependency graph

vortigont, 2024-06-21:
- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

### Configuration

**Operating system**:
Ubuntu Linux

**PlatformIO Version** (`platformio --version`):
PlatformIO Core             6.1.15
platform: espressif32

### Description of problem
It looks like LDF in sometimes could exclude bundled platform libraries from the dependency graph in situations like:
 - a 3rd party lib `A` depends on some bundled Arduino library, i.e. `Wifi.h` or `FS.h`
 - this lib's library manifest does NOT have those bundled libs defined in section `"dependencies":  []`
 - another 3rd party lib `B` depends on lib `A` and has it defined in lib's manifest
 - user's project depends on a lib `A` and defines it's dependency in `platformio.ini`

During dependency graph LDF misses the dependency for lib `A` on bundled Arduino libs and when compiling lib `A` it fails with error like `fatal error: Update.h: No such file or directory` where `Update.h` is bundled lib.

The situation is described in [this](https://community.platformio.org/t/how-to-properly-include-libraries-in-custom-libraries/39480) community topic with AsyncServer. But I've faced similar with another lib [esp32-flashz](https://github.com/vortigont/esp32-flashz)

There are certain conditions that could prevent (or hide) this problem:
 1) lib `A` has defined bundled Arduino libs in manifest's `"dependencies":  []` section
 2) Including bundled lib's header files in some places could help there too

Usually it works fine with described conditions. However I faced really strange situation with GutHub's CI workflow that runs project builds with a lib and workaround 1) implemented.
LDF considers bundled lib names as some another third party lib names and pulls same named lib from the Platformio's registry, then build process obviously fails since this is a completely wrong lib.
In the CI job log it looks like this:
```
Library Manager: Installing git+https://github.com/vortigont/esp32-flashz#libdeps
git version 2.45.2
Cloning into '/home/runner/.platformio/.cache/tmp/pkg-installing-xwkyu2wm'...
Library Manager: esp32-flashz@1.1.4+sha.ce117d6 has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing Update
Library Manager: Installing Ticker
Library Manager: Warning! More than one package has been found by Ticker requirements:
Library Manager:  - sstaub/Ticker@4.4.0
Library Manager:  - jsc/Ticker@2.1.0
Library Manager: Please specify detailed REQUIREMENTS using package owner and version (shown above) to avoid name conflicts
```

Here LDF picks `Ticker` lib as some 3rd party instead of bundled Arduino's [Ticker](https://github.com/espressif/arduino-esp32/tree/master/libraries/Ticker). Then pio pulls it from PlatformIO's registry and tries to include it instead of bundled one. This obviously fails:

```
.pio/libdeps/esp32debug/esp32-flashz/src/flashz-http.cpp:122:37: error: no matching function for call to 'Ticker::Ticker()'
                             t = new Ticker;
                                     ^~~~~~
In file included from .pio/libdeps/esp32debug/esp32-flashz/src/flashz-http.hpp:39,
                 from .pio/libdeps/esp32debug/esp32-flashz/src/flashz-http.cpp:30:
.pio/libdeps/esp32debug/Ticker/Ticker.h:74:2: note: candidate: 'Ticker::Ticker(fptr, uint32_t, uint32_t, resolution_t)'
  Ticker(fptr callback, uint32_t timer, uint32_t repeat = 0, resolution_t resolution = MICROS);
  ^~~~~~
.pio/libdeps/esp32debug/Ticker/Ticker.h:74:2: note:   candidate expects 4 arguments, 0 provided
```

Same project build on my local ubuntu machine completes fine. Not sure why in GitHub LDF behaves differently.

On the other hand, if I remove `Ticker` and `Update` from lib's [manifest](https://github.com/vortigont/esp32-flashz/blob/81f3738e9f700713ea26404ba27b9abdc19bf220/library.json) the build fails with `fatal error: Update.h: No such file or directory` both locally and in GitHub.

Summary: somehow LDF might exclude bundled Arduino libs from recursive dependency graphs.

ivankravets, 2024-12-12:
Could you provide a simple project to reproduce this issue?

vortigont, 2024-12-14:
@ivankravets 
yes, sure. Here is the dummy [piobug4940](https://github.com/vortigont/piobug4940) project that does not builds with default core and builds fine with #4941

The dependency there is like this:

```
PROJECT
  | -- EmbUI (lib from a registry)
           | -- esp32-flashz (lib from a registry)
           |      | -- Update (bundled lib from Arduino core )
           |      | -- Ticker (bundled lib from Arduino core )
           | -- other 3rd party libs
```

commit 9be0a824 (2024-12-18): LDF: refresh lib dependency after recursive search (#4941)

LDF might mistakenly remove recursive dependency libs from a graph
usually platform bundled ones

Closes #4940
closed 2024-12-18 by ivankravets