platformio/platformio-core #4957: Library compiles when in lib folder, but not when in lib_deps

sylque, 2024-08-01:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Windows

**PlatformIO Version** (`platformio --version`): 6.1.15

### Description of problem

The "SimpleFOC" library compiles when in lib folder, but not when in lib_deps. When it is in lib_deps, at one point during compilation, relative paths within the library seem to stop working.

**The content of `platformio.ini`:**
```ini
[env:lolin_d32]
platform = https://github.com/tasmota/platform-espressif32/releases/download/2024.07.11/platform-espressif32.zip
board = lolin_d32
framework = arduino
lib_deps = askuric/Simple FOC
```

**Source file to reproduce issue:**
```cpp
#include <SimpleFOC.h>
void setup() {}
void loop() {}
```

### Additional info

Compilation of the "askuric/Simple FOC" library hangs at some point:
```
Processing lolin_d32 (platform: https://github.com/tasmota/platform-espressif32/releases/download/2024.07.11/platform-espressif32.zip; board: lolin_d32; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/lolin_d32.html
PLATFORM: Espressif 32 (2024.7.11) > WEMOS LOLIN D32
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.0.2+sha.9100566 
 - tool-esptoolpy @ 4.7.4 
 - tool-mklittlefs @ 3.2.0 
 - tool-riscv32-esp-elf-gdb @ 11.2.0+20220823 
 - tool-xtensa-esp-elf-gdb @ 11.2.0+20230208 
 - toolchain-xtensa-esp32 @ 12.2.0+20230208
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 26 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Simple FOC @ 2.3.4
Building in release mode
Compiling .pio\build\lolin_d32\libf27\Simple FOC\current_sense\hardware_specific\teensy\teensy4_mcu.cpp.o
Compiling .pio\build\lolin_d32\libf27\Simple FOC\drivers\hardware_specific\atmega\atmega2560_mcu.cpp.o
Compiling .pio\build\lolin_d32\libf27\Simple FOC\drivers\hardware_specific\atmega\atmega328_mcu.cpp.o
Compiling .pio\build\lolin_d32\libf27\Simple FOC\drivers\hardware_specific\atmega\atmega32u4_mcu.cpp.o
Compiling .pio\build\lolin_d32\libf27\Simple FOC\drivers\hardware_specific\due_mcu.cpp.o
Compiling .pio\build\lolin_d32\libf27\Simple FOC\drivers\hardware_specific\esp32\esp32_driver_mcpwm.cpp.o
Compiling .pio\build\lolin_d32\libf27\Simple FOC\drivers\hardware_specific\esp32\esp32_ledc_mcu.cpp.o
Compiling .pio\build\lolin_d32\libf27\Simple FOC\drivers\hardware_specific\esp32\esp32_mcpwm_mcu.cpp.o
In file included from .pio/libdeps/lolin_d32/Simple FOC/src/current_sense/hardware_specific/teensy/../../../drivers/hardware_specific/teensy/teensy_mcu.h:4,
                 from .pio/libdeps/lolin_d32/Simple FOC/src/current_sense/hardware_specific/teensy/../../../drivers/hardware_specific/teensy/teensy4_mcu.h:4,
                 from .pio/libdeps/lolin_d32/Simple FOC/src/current_sense/hardware_specific/teensy/teensy4_mcu.cpp:2:
.pio/libdeps/lolin_d32/Simple FOC/src/current_sense/hardware_specific/teensy/../../../drivers/hardware_specific/teensy/../../hardware_api.h:6:10: fatal error: ../communication/SimpleFOCDebug.h: No such file or directory
    6 | #include "../communication/SimpleFOCDebug.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```
This is odd, because ../communication/SimpleFOCDebug.h is really there. There's also a random factor, as sometimes the same error occurs with another include file. In fact, at one point, any attempt to include a file with a relative path seems to trigger an error. 

**If I remove the library from lib_deps and copy it the lib folder, it compiles without a glitch.**

I've done a Full Clean, deleted the .pio folder, uninstalled/reinstalled PlatformIO, but it didn't solve the problem.

vortigont, 2024-08-07:
it builds fine for me in linux. Chech your project's folder path, maybe you are getting too close to the 255 chars limit.
Try to move project folder to the root of your drive with short path. i.e. `c:\proj\`

sylque, 2024-08-07:
@vortigont you nailed it, thanks a lot!

Long paths are supposed to be enabled on my Windows (as described [here](https://superuser.com/a/1807797/399801)), so is the 255 chars limit a known limitation of PlatformIO build system?

vortigont, 2024-08-08:
not sure where is the root of this problem, probably in python's windows build not the platformio.
Or better say it's in Windows :))) It's 21st century, 3rd decade now and windoze still stuck with 40+ years old hard limit and some registry hacks >8-0

Jason2866, 2024-08-09:
Mhh, strange there should be a warning regarding whitespaces `Simple FOC` in path.
That's the next Windows 21st century, 3rd decade problem

vortigont, 2024-08-10:
I guess there a special place in hell for those ones who creates lib folders with spaces :)

Jason2866, 2024-08-10:
Yeah, probably the creator loves to have issues in his github repo ;-)

sylque, 2024-08-10:
@Jason2866 , the creator of the repo has nothing to do with it. `Simple FOC` is not a folder name, it's a library name. I think it's the build system that converted it to a folder name.

Jason2866, 2024-08-11:
@sylque Exactly that's why it is a bad idea. The library name is used to generate the folder name for the lib where it is placed.

ivankravets, 2024-08-28:
@sylque , have you resolved this issue? Might we close it?

sylque, 2024-08-28:
I've resolved my issue by moving the project folder to the root of my file system (so as to reduce path lengths). However, I don't think this is a proper fix.

At this stage, I believe we should identify the faulty compilation tool and file a bug report with its developers. Unfortunately, I'm not knowledgeable enough about the Espressif toolchain to investigate further.