platformio/platformio-core #4975: Problem finding package

AgainPsychoX, 2024-09-07:
What kind of issue is this?

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 10.0.19045.4780

**PlatformIO Core version**: 6.1.15

**VS Code extension version**: v3.3.3

**VS Code version**: 1.92.2

### Description of problem

I want to use my ESP32-S3 with Arduino framework, with newer GCC preferably (C++20), so I specify newer toolchain as `platform_packages`, but it doesn't work. Of course, unlike the error suggests, it is compatible with Windows x64, see https://registry.platformio.org/tools/espressif/toolchain-xtensa-esp32s3/compatibility .

#### Steps to Reproduce

1. Create project with specified `platformio.ini`
2. Try open the project (or save the file) to prompt packages installation

### Actual Results

Fails with error log:
```
Resolving transmitter dependencies...
Tool Manager: Installing espressif/^12.2.0+20230208
UnknownPackageError: Could not find the package with 'espressif/^12.2.0+20230208' requirements for your system 'windows_amd64'
```

Please note the error log looks weird: there is no proper package name in there!

### Expected Results

Installed 12.2.0 instead default 8.4.0. 

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:transmitter]
platform = espressif32
platform_packages =
	espressif/toolchain-xtensa-esp32s3@^12.2.0+20230208
board = my-esp32s3-n16r8
framework = arduino
```

**Source file to reproduce issue:** N/A (any will work).

**Custom board JSON** (`boards/my-esp32s3-n16r8.json`):
```json
{
	"build": {
		"arduino":{
			"ldscript": "esp32s3_out.ld",
			"partitions": "default_16MB.csv",
			"memory_type": "qio_opi"
		},
		"core": "esp32",
		"extra_flags": [
			"-DARDUINO_ESP32S3_DEV",
			"-DBOARD_HAS_PSRAM",
			"-DARDUINO_USB_MODE=1",
			"-DARDUINO_RUNNING_CORE=1",
			"-DARDUINO_EVENT_RUNNING_CORE=1"
		],
		"f_cpu": "240000000L",
		"f_flash": "80000000L",
		"flash_mode": "qio",
		"psram_type": "opi",
		"hwids": [
			[
				"0x303A",
				"0x1001"
			]
		],
		"mcu": "esp32s3",
		"variant": "esp32s3"
	},
	"connectivity": [
		"wifi",
		"bluetooth"
	],
	"debug": {
		"default_tool": "esp-builtin",
		"onboard_tools": [
			"esp-builtin"
		],
		"openocd_target": "esp32s3.cfg"
	},
	"frameworks": [
		"arduino",
		"espidf"
	],
	"name": "Espressif ESP32-S3-DevKitC-1-N16R8 (16 MB QD, 8MB PSRAM)",
	"upload": {
		"flash_size": "16MB",
		"maximum_ram_size": 327680,
		"maximum_size": 16777216,
		"require_upload_port": true,
		"speed": 460800
	},
	"url": "https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html",
	"vendor": "Espressif"
}
```

I think it would fail with any board tbh...

### Additional info

...

ivankravets, 2024-09-07:
Could you try?
```ini
platform_packages =
	espressif/toolchain-xtensa-esp32s3@^12.2.0
```

AgainPsychoX, 2024-09-07:
Oh, it works that way! Thanks. 

However the message could be more descriptive if so...

closed 2024-09-07 by ivankravets