platformio/platformio-core #4980: Switch to or allow HTTPS for checking the internet is on.

fifieldt, 2024-09-20:
- [x] **PlatformIO Core**.

In platformio/platformio-core/http.py  `_internet_on` method, HTTP port 80 is used to check if the internet is on:
https://github.com/platformio/platformio-core/blob/4230b223d2cf307408fdd4b1e077f0b149221f13/platformio/http.py#L201

I'm working with PlatformIO on networks where unencrypted traffic is heavily restricted. Even outbound traffic like web requests. So, HTTPS is very much preferred to HTTP.

Would it be possible to switch to using HTTPS hosts to check the internet is working? Or add one HTTPS host to the rotation?

ivankravets, 2024-09-23:
Thanks for the report. Please re-test with `pio upgrade --dev`. Does it work now?

closed 2024-09-23 by ivankravets

fifieldt, 2024-09-23:
Thank you very much @ivankravets !!

commit e6ae18ab (2024-09-23): Enhanced internet connection checks by falling back to HTTPS protocol when HTTP (port 80) fails // Resolve #4980