platformio/platformio-core #4983: Python 3.13: ModuleNotFoundError: No module named 'cgi' (updating bottle should fix this)

hroncok, 2024-09-23:
### Configuration

**Operating system**: Fedora Linux 41

**PlatformIO Version** (`platformio --version`): 3.3.3

### Description of problem

On Fedora Linux 41, we have [upgraded to Python 3.13](https://fedoraproject.org/wiki/Changes/Python3.13). Fedora Linux 41 is still in Beta, but users are already using PlatformIO and they report bugs:

 - Fedora Linux 41 user-reported bug report: [Code platformio extension does not work with Python 3.13](https://bugzilla.redhat.com/show_bug.cgi?id=2314013)
 
I've reproduced the issue on Fedora 41. I do not use PlatformIO, I am merely the Python maintainer for Fedora.

#### Steps to Reproduce

1. Get Fedora 41 Workstation, https://fedoraproject.org/workstation/download/?beta
2. Install it to a VM, as there is not enough space on the live system to install Code and PlatformIO
3. Follow https://code.visualstudio.com/docs/setup/linux#_rhel-fedora-and-centos-based-distributions
4. Follow https://platformio.org/install/ide?install=vscode
5. Upon error, follow the advice given (VSCode > Help > Toggle Developer Tools > Console) to get the full traceback

Alternatively, get another system where Python 3.13 is the Python used by Code and follow from step 4.

### Actual Results

```

Error: Error: Compatible PlatformIO Core not found.
Reason: Could not run `/home/test/.platformio/penv/bin/platformio --help`.
Error: ModuleNotFoundError: Traceback (most recent call last):
  File "/home/test/.platformio/penv/lib64/python3.13/site-packages/platformio/__main__.py", line 103, in main
    cli()  # pylint: disable=no-value-for-parameter
    ~~~^^
  File "/home/test/.platformio/penv/lib64/python3.13/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ~~~~~~~~~^^^^^^^^^^^^^^^^^
  File "/home/test/.platformio/penv/lib64/python3.13/site-packages/click/core.py", line 1077, in main
    with self.make_context(prog_name, args, **extra) as ctx:
         ~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/test/.platformio/penv/lib64/python3.13/site-packages/click/core.py", line 943, in make_context
    self.parse_args(ctx, args)
    ~~~~~~~~~~~~~~~^^^^^^^^^^^
  File "/home/test/.platformio/penv/lib64/python3.13/site-packages/click/core.py", line 1644, in parse_args
    rest = super().parse_args(ctx, args)
  File "/home/test/.platformio/penv/lib64/python3.13/site-packages/click/core.py", line 1408, in parse_args
    value, args = param.handle_parse_result(ctx, opts, args)
                  ~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^
  File "/home/test/.platformio/penv/lib64/python3.13/site-packages/click/core.py", line 2400, in handle_parse_result
    value = self.process_value(ctx, value)
  File "/home/test/.platformio/penv/lib64/python3.13/site-packages/click/core.py", line 2362, in process_value
    value = self.callback(ctx, self, value)
  File "/home/test/.platformio/penv/lib64/python3.13/site-packages/click/core.py", line 1300, in show_help
    echo(ctx.get_help(), color=ctx.color)
         ~~~~~~~~~~~~^^
  File "/home/test/.platformio/penv/lib64/python3.13/site-packages/click/core.py", line 704, in get_help
    return self.command.get_help(self)
           ~~~~~~~~~~~~~~~~~~~~~^^^^^^
  File "/home/test/.platformio/penv/lib64/python3.13/site-packages/click/core.py", line 1325, in get_help
    self.format_help(ctx, formatter)
    ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
  File "/home/test/.platformio/penv/lib64/python3.13/site-packages/click/core.py", line 1358, in format_help
    self.format_options(ctx, formatter)
    ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
  File "/home/test/.platformio/penv/lib64/python3.13/site-packages/click/core.py", line 1564, in format_options
    self.format_commands(ctx, formatter)
    ~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
  File "/home/test/.platformio/penv/lib64/python3.13/site-packages/click/core.py", line 1617, in format_commands
    cmd = self.get_command(ctx, subcommand)
  File "/home/test/.platformio/penv/lib64/python3.13/site-packages/platformio/cli.py", line 94, in get_command
    module = importlib.import_module(commands[cmd_name])
  File "/usr/lib64/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1022, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/home/test/.platformio/penv/lib64/python3.13/site-packages/platformio/commands/ci.py", line 24, in <module>
    from platformio.project.commands.init import project_init_cmd, validate_boards
  File "/home/test/.platformio/penv/lib64/python3.13/site-packages/platformio/project/commands/init.py", line 31, in <module>
    from platformio.project.integration.generator import ProjectGenerator
  File "/home/test/.platformio/penv/lib64/python3.13/site-packages/platformio/project/integration/generator.py", line 19, in <module>
    import bottle
  File "/home/test/.platformio/penv/bin/bottle.py", line 38, in <module>
    import base64, cgi, email.utils, functools, hmac, itertools, mimetypes,\
            os, re, subprocess, sys, tempfile, threading, time, warnings, hashlib
ModuleNotFoundError: No module named 'cgi'
```

### Expected Results

PlatformIo install and works with Python 3.13.

### Additional info

This is caused by https://github.com/bottlepy/bottle/issues/1403 which was patched upstream in https://github.com/bottlepy/bottle/pull/1438 and released as bottle 0.13.0. Updating the presumably vendored bottle should fix this.

ivankravets, 2024-09-23:
Fixed in https://github.com/platformio/platformio-core/commit/4230b223d2cf307408fdd4b1e077f0b149221f13

Please re-test with `pio upgrade --dev`. Does it work now?

closed 2024-09-23 by ivankravets

hroncok, 2024-09-23:
How do I do `pio upgrade --dev`? I am not a user of Code and I need a very specific set of instructions, sorry.

ivankravets, 2024-09-23:
```
$ /home/test/.platformio/penv/bin/pip install -U https://github.com/platformio/platformio-core/archive/refs/heads/develop.zip
```

WITHOUT SUDO

hroncok, 2024-09-23:
The pip install --upgrade succeeds. After that, PlatformIO appears to be installed in Code. Thanks.

ivankravets, 2024-09-23:
Thanks for the confirmation. We will bump the new PIO Core release this week.