platformio/platformio-core #4996: ESP8266 `xtensa-lx106-elf-gcc` and ESP32S3 `xtensa-esp32s3-elf-g++` not found on Alpine docker

ptr727, 2024-10-11:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to the
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues) repository

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Docker `python:alpine`

**PlatformIO Version** (`platformio --version`): `PlatformIO Core, version 6.1.16`

### Description of problem
`xtensa` build chain is not found on Alpine docker containers.  
Suspect the wrong tool binaries or tools not working with `musl` are being downloaded when running on Alpine.

#### Steps to Reproduce

1. Create docker container using `python:alpine` (Python 3.13, Alpine 3.20)
2. `pip install platformio`
3. `pio project init --board esp8285`
4. `pio run`

### Actual Results

```
a2c1ffeabb89:/config/pio-esp8266# pio run
Processing esp8285 (platform: espressif8266; board: esp8285; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp8285.html
PLATFORM: Espressif 8266 (4.2.1) > Generic ESP8285 Module
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
PACKAGES:
 - framework-arduinoespressif8266 @ 3.30102.0 (3.1.2)
 - tool-esptool @ 1.413.0 (4.13)
 - tool-esptoolpy @ 1.30000.201119 (3.0.0)
 - toolchain-xtensa @ 2.100300.220621 (10.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 37 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/esp8285/src/main.cpp.o
sh: xtensa-lx106-elf-g++: not found
Generating LD script .pio/build/esp8285/ld/local.eagle.app.v6.common.ld
*** [.pio/build/esp8285/src/main.cpp.o] Error 127
sh: xtensa-lx106-elf-gcc: not found
*** [.pio/build/esp8285/ld/local.eagle.app.v6.common.ld] Error 127
```

### Expected Results

Compile works.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

Dockerfile example:

```yaml
# docker buildx create --name "test" --use
# docker buildx build --load --platform linux/amd64 --tag test:latest .
# docker run -it --rm --name Test test:latest /bin/bash

FROM python:alpine AS final

RUN \
    apk update && apk upgrade \
    && apk add --no-cache \
        bash \
        build-base \
        git \
        python3-dev

RUN git config --system --add safe.directory '*'

RUN pip install platformio

VOLUME /config
WORKDIR /config

RUN <<EOF
mkdir /config/pio-esp8266
cd /config/pio-esp8266
pio project init --board esp8285
EOF

COPY <<EOF /config/pio-esp8266/src/main.cpp
#include "Arduino.h"

#ifndef LED_BUILTIN
#define LED_BUILTIN 13
#endif

void setup()
{
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
  digitalWrite(LED_BUILTIN, HIGH);

  delay(1000);

  digitalWrite(LED_BUILTIN, LOW);

  delay(1000);
}
EOF

RUN <<EOF
mkdir /config/pio-esp32
cd /config/pio-esp32
pio project init --board esp32-s3-devkitc-1
EOF

COPY <<EOF /config/pio-esp32/src/main.cpp
#include "driver/gpio.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "esp_check.h"

static const char *tagName = "Test";
static const gpio_num_t oneWireGPIO = GPIO_NUM_15;

esp_err_t Config1WireIO()
{
    ESP_LOGI(tagName, "Config1WireIO()");
    gpio_config_t gpioConfig = {};
    gpioConfig.intr_type = GPIO_INTR_DISABLE;
    gpioConfig.mode = GPIO_MODE_INPUT_OUTPUT;
    gpioConfig.pull_down_en = GPIO_PULLDOWN_DISABLE;
    gpioConfig.pull_up_en = GPIO_PULLUP_ENABLE;
    gpioConfig.pin_bit_mask = 1ULL << oneWireGPIO;
    ESP_RETURN_ON_ERROR(gpio_config(&gpioConfig), tagName, "gpio_config()");
    gpio_dump_io_configuration(stdout, gpioConfig.pin_bit_mask);
    return ESP_OK;
}

esp_err_t SetIOState(gpio_num_t gpio_num, bool state)
{
    ESP_LOGI(tagName, "SetIOState(%d, %d)", gpio_num, state);
    ESP_RETURN_ON_ERROR(gpio_set_level(gpio_num, state == true ? 1 : 0), tagName, "gpio_set_level()");
    return ESP_OK;
}

extern "C" void app_main()
{
    static const TickType_t xDelay = 1000 / portTICK_PERIOD_MS;

    ESP_LOGI(tagName, "app_main()");
    if (Config1WireIO() != ESP_OK)
    {
        return;
    }

    bool state = true;
    while (1)
    {
        SetIOState(oneWireGPIO, state);
        vTaskDelay(xDelay);
        state = !state;
    }
}
EOF

# Compiling .pio/build/esp8285/src/main.cpp.o
# sh: xtensa-lx106-elf-g++: not found
# Generating LD script .pio/build/esp8285/ld/local.eagle.app.v6.common.ld
# *** [.pio/build/esp8285/src/main.cpp.o] Error 127
# sh: xtensa-lx106-elf-gcc: not found
# *** [.pio/build/esp8285/ld/local.eagle.app.v6.common.ld] Error 127
RUN <<EOF
cd /config/pio-esp8266
pio run || true
EOF

# Compiling .pio/build/esp32-s3-devkitc-1/src/main.cpp.o
# sh: xtensa-esp32s3-elf-g++: not found
# Compiling .pio/build/esp32-s3-devkitc-1/FrameworkArduino/Esp.cpp.o
# *** [.pio/build/esp32-s3-devkitc-1/src/main.cpp.o] Error 127
# sh: xtensa-esp32s3-elf-g++: not found
# *** [.pio/build/esp32-s3-devkitc-1/FrameworkArduino/Esp.cpp.o] Error 127
RUN <<EOF
cd /config/pio-esp32
pio run || true
EOF
```

ptr727, 2024-10-12:
Per suggestion from [forum thread](https://community.platformio.org/t/is-alpine-docker-a-supported-platform/43700/4) looks like glibc vs. musl as guessed.

```
fc3b9f224885:~/.platformio/packages/toolchain-xtensa/bin# ldd xtensa-lx106-elf-gcc
        /lib64/ld-linux-x86-64.so.2 (0x7fbf8a328000)
        libc.so.6 => /lib64/ld-linux-x86-64.so.2 (0x7fbf8a328000)
Error loading shared library ld-linux-x86-64.so.2: No such file or directory (needed by xtensa-lx106-elf-gcc)
Error relocating xtensa-lx106-elf-gcc: __strcat_chk: symbol not found
Error relocating xtensa-lx106-elf-gcc: __snprintf_chk: symbol not found
Error relocating xtensa-lx106-elf-gcc: __vfprintf_chk: symbol not found
Error relocating xtensa-lx106-elf-gcc: __read_chk: symbol not found
Error relocating xtensa-lx106-elf-gcc: __printf_chk: symbol not found
Error relocating xtensa-lx106-elf-gcc: __fprintf_chk: symbol not found
Error relocating xtensa-lx106-elf-gcc: __sprintf_chk: symbol not found
```

```
fc3b9f224885:~/.platformio/packages/toolchain-xtensa-esp32s3/bin# ldd xtensa-esp32s3-elf-g++
        /lib64/ld-linux-x86-64.so.2 (0x7f5d93469000)
        libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x7f5d931d0000)
        libm.so.6 => /lib64/ld-linux-x86-64.so.2 (0x7f5d93469000)
        libc.so.6 => /lib64/ld-linux-x86-64.so.2 (0x7f5d93469000)
        libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x7f5d931ac000)
Error relocating xtensa-esp32s3-elf-g++: __strdup: symbol not found
```

Per some google [help](https://github.com/rails/tailwindcss-rails/issues/115#issuecomment-1004896047) and [wiki](https://wiki.alpinelinux.org/wiki/Running_glibc_programs) I also installed `gcompat`, and it seems to work.

```
c40f855a1d9c:/config/pio-esp8266# pio run
Processing esp8285 (platform: espressif8266; board: esp8285; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp8285.html
PLATFORM: Espressif 8266 (4.2.1) > Generic ESP8285 Module
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
PACKAGES:
 - framework-arduinoespressif8266 @ 3.30102.0 (3.1.2)
 - tool-esptool @ 1.413.0 (4.13)
 - tool-esptoolpy @ 1.30000.201119 (3.0.0)
 - toolchain-xtensa @ 2.100300.220621 (10.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 37 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Retrieving maximum program size .pio/build/esp8285/firmware.elf
Checking size .pio/build/esp8285/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  34.3% (used 28104 bytes from 81920 bytes)
Flash: [===       ]  34.3% (used 261415 bytes from 761840 bytes)
======================================================================================================================================= [SUCCESS] Took 0.52 seconds =======================================================================================================================================
c40f855a1d9c:/config/pio-esp8266# 
```

```
766a35c5191a:/config/pio-esp32# pio run
Processing esp32-s3-devkitc-1 (platform: espressif32; board: esp32-s3-devkitc-1; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (6.9.0) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20017.0 (2.0.17)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 33 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Retrieving maximum program size .pio/build/esp32-s3-devkitc-1/firmware.elf
Checking size .pio/build/esp32-s3-devkitc-1/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   5.5% (used 18032 bytes from 327680 bytes)
Flash: [=         ]   7.0% (used 233073 bytes from 3342336 bytes)
======================================================================================================================================= [SUCCESS] Took 0.93 seconds =======================================================================================================================================
766a35c5191a:/config/pio-esp32# 
```

ivankravets, 2024-10-15:
> In short for somebody landing here apk add build-base gcompat.

Thanks for the solution.

closed 2024-10-15 by ivankravets