platformio/platformio-core #4997: Add arduino support for the ESP32-C6

SomeSpaceNerd, 2024-10-14:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to the
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues) repository

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [X] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

Could arduino framework support please be added for the ESP32-C6? The latest ESP32 arduino core supports the ESP32-C6 and it is becoming an increasingly popular chip.

znmeb, 2024-10-14:
This is a fortunate coincidence - just today I was testing an ESP32-C6-DevKitC-1 and discovered it was supported in Arduino but not PlatformIO. But why does PlatformIO need to support this board via Arduino? Doesn't the Espressif toolchain support it? Their documentation seems to suggest so. Is an Arduino build the only way to get debugging on it?

SomeSpaceNerd, 2024-10-14:
>But why does PlatformIO need to support this board via Arduino? Doesn't the Espressif toolchain support it? 

I was doing some testing with custom board JSON for a Seeeduino XIAO esp32c6 and the main issue seems to be the absence of a "platformio-build-esp32c6.py" build script. I am not sure if that's an issue with platformio or the ESP32 toolchain itself because it's looking for a PIO dedicated script that's contained in the tools directory of the esp32 arduino core.

znmeb, 2024-10-14:
> > But why does PlatformIO need to support this board via Arduino? Doesn't the Espressif toolchain support it?
> 
> I would like to have arduino support added because it allows you to make projects that support many microcontrollers outside of the ESP32 ecosystem.

I don't know the internals of either PlatformIO or Arduino well enough to tell if this is a documentation issue, a bug, or a feature that needs to be planned and implemented. I do see my board in the output of `pio boards > board_list.txt` so I'm inclined to say documentation / bug, not a missing feature.

[board_list.txt](https://github.com/user-attachments/files/17360839/board_list.txt)

```
esp32-c6-devkitc-1                   ESP32C6  160MHz       8MB      320KB   Espressif ESP32-C6-DevKitC-1
```

SomeSpaceNerd, 2024-10-14:
I was doing some testing with custom board JSON for a Seeeduino XIAO esp32c6 and the main issue seems to be the absence of a "platformio-build-esp32c6.py" build script. I am not sure if that's an issue with platformio or the ESP32 toolchain itself because it's looking for a PIO dedicated script that's contained in the tools directory of the esp32 arduino core.

znmeb, 2024-10-14:
I opened an issue on the compile failure - see https://github.com/platformio/platformio-core/issues/4999 if you're interested in the gory details.