platformio/platformio-core #4998: Space in compilator path not escaped in compilation_db

HanyzPAPU, 2024-10-14:
- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**:

Windows, with a username who has a space in the username.

**PlatformIO Version** (`platformio --version`):

PlatformIO Core, version 6.1.16

### Description of problem

When the path to the `.platformio` contains a space (for example in the username), the `compile_commands.json` produced by running `pio run -t compiledb` does not escape the compiler executable path in the `command` fields. 

#### Steps to Reproduce

1. Have a user with a space in the username
2. Install PlatformIO, create a new project.
3. Run `pio run -t compiledb`

### Actual Results

Example of a malformed command:

```
{"command": "C:\\Users\\Karel Mezera\\.platformio\\packages\\toolchain-gccarmnoneeabi\\bin\\arm-none-eabi-g++.exe -o .pio\\build\\nrf52840_dk\\FrameworkArduinoVariant\\variant.cpp.o -c -fno-rtti -fno-exceptions -std=gnu++11 -fno-threadsafe-statics -Os -ffunction-sections -fdata-sections -Wall -mthumb -nostdlib --param max-inline-insns-single=500 -mcpu=cortex-m4 -mfloat-abi=softfp -mfpu=fpv4-sp-d16 -DPLATFORMIO=60116 -DARDUINO_NRF52_DK -DARDUINO=10805 -DF_CPU=16000000L -DARDUINO_ARCH_NRF5 -DNRF5 -DNRF52 -DUSE_LFXO \"-IC:\\Users\\Karel Mezera\\.platformio\\packages\\framework-arduinonordicnrf5\\cores\\nRF5\" \"-IC:\\Users\\Karel Mezera\\.platformio\\packages\\framework-arduinonordicnrf5\\cores\\nRF5\\SDK\\components\\drivers_nrf\\delay\" \"-IC:\\Users\\Karel Mezera\\.platformio\\packages\\framework-arduinonordicnrf5\\cores\\nRF5\\SDK\\components\\device\" \"-IC:\\Users\\Karel Mezera\\.platformio\\packages\\framework-arduinonordicnrf5\\cores\\nRF5\\SDK\\components\\toolchain\" \"-IC:\\Users\\Karel Mezera\\.platformio\\packages\\framework-arduinonordicnrf5\\cores\\nRF5\\SDK\\components\\toolchain\\CMSIS\\Include\" \"-IC:\\Users\\Karel Mezera\\.platformio\\packages\\framework-arduinonordicnrf5\\variants\\nRF52DK\" \"C:\\Users\\Karel Mezera\\.platformio\\packages\\framework-arduinonordicnrf5\\variants\\nRF52DK\\variant.cpp\"",
        "directory": "C:\\Users\\Karel Mezera\\Documents\\CLionProjects\\untitled",
        "file": "C:\\Users\\Karel Mezera\\.platformio\\packages\\framework-arduinonordicnrf5\\variants\\nRF52DK\\variant.cpp",
        "output": ".pio\\build\\nrf52840_dk\\FrameworkArduinoVariant\\variant.cpp.o"
    }
```

### Expected Results

```
{"command": "\"C:\\Users\\Karel Mezera\\.platformio\\packages\\toolchain-gccarmnoneeabi\\bin\\arm-none-eabi-g++.exe\" -o .pio\\build\\nrf52840_dk\\FrameworkArduinoVariant\\variant.cpp.o -c -fno-rtti -fno-exceptions -std=gnu++11 -fno-threadsafe-statics -Os -ffunction-sections -fdata-sections -Wall -mthumb -nostdlib --param max-inline-insns-single=500 -mcpu=cortex-m4 -mfloat-abi=softfp -mfpu=fpv4-sp-d16 -DPLATFORMIO=60116 -DARDUINO_NRF52_DK -DARDUINO=10805 -DF_CPU=16000000L -DARDUINO_ARCH_NRF5 -DNRF5 -DNRF52 -DUSE_LFXO \"-IC:\\Users\\Karel Mezera\\.platformio\\packages\\framework-arduinonordicnrf5\\cores\\nRF5\" \"-IC:\\Users\\Karel Mezera\\.platformio\\packages\\framework-arduinonordicnrf5\\cores\\nRF5\\SDK\\components\\drivers_nrf\\delay\" \"-IC:\\Users\\Karel Mezera\\.platformio\\packages\\framework-arduinonordicnrf5\\cores\\nRF5\\SDK\\components\\device\" \"-IC:\\Users\\Karel Mezera\\.platformio\\packages\\framework-arduinonordicnrf5\\cores\\nRF5\\SDK\\components\\toolchain\" \"-IC:\\Users\\Karel Mezera\\.platformio\\packages\\framework-arduinonordicnrf5\\cores\\nRF5\\SDK\\components\\toolchain\\CMSIS\\Include\" \"-IC:\\Users\\Karel Mezera\\.platformio\\packages\\framework-arduinonordicnrf5\\variants\\nRF52DK\" \"C:\\Users\\Karel Mezera\\.platformio\\packages\\framework-arduinonordicnrf5\\variants\\nRF52DK\\variant.cpp\"",
        "directory": "C:\\Users\\Karel Mezera\\Documents\\CLionProjects\\untitled",
        "file": "C:\\Users\\Karel Mezera\\.platformio\\packages\\framework-arduinonordicnrf5\\variants\\nRF52DK\\variant.cpp",
        "output": ".pio\\build\\nrf52840_dk\\FrameworkArduinoVariant\\variant.cpp.o"
    }
```

Notice the added `\"` around the compiler executable.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:nrf52840_dk]
platform = nordicnrf52
board = nrf52840_dk
framework = arduino
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>
void setup(){}
void loop(){}
```

ivankravets, 2024-10-15:
Could you point to the specification that documents `{"command": "\"C:\`?

HanyzPAPU, 2024-10-16:
[JSON Compilation Database Format Specification](https://clang.llvm.org/docs/JSONCompilationDatabase.html#format) allows shell escaping inside of the command string.

[Windows Command Prompt syntax remarks](https://learn.microsoft.com/en-us/windows-server/administration/windows-commands/cmd#remarks) states that a path with spaces should be enclosed in quotation marks.

To achieve this, the quotation marks have to be inside of the JSON string, and according to the [JSON specification](https://www.json.org/json-en.html) should be escaped as `\"`.

ivankravets, 2024-10-16:
How do we reproduce this issue? With which type of software do you use `compilation_db`?

Indeed, the end command runner (3rd party software, shell, etc) is responsible to handle "spaces" in the path.

HanyzPAPU, 2024-10-16:
> How do we reproduce this issue?

Parsing the `compilation_db` by any JSON parser (to correctly unescape the strings) and consequently trying to run the content of the `"command"` field in Command Prompt should be able to reproduce.

> With which type of software do you use compilation_db?

The `compilation_db` is used by the plugin for CLion.

closed 2024-10-17 by ivankravets

HanyzPAPU, 2024-10-18:
Just tried it out, and the problem's gone. Thank you.

proxict, 2025-05-04:
Can we please re-open this issue to confirm/reject [this comment](https://github.com/platformio/platformio-core/commit/f5ab0e5ddd7568de06f7df7f616e4c2f04abb02d#diff-1bb43c4f1cec6e861e38a5a2f9e292a6343625e530256c3f993d627ffedd3971R61)?

commit f5ab0e5d (2024-10-17): Resolve an issue where the ``compiledb`` target failed to properly escape compiler executable paths containing spaces // Resolve #4998