platformio/platformio-core #5004: Relative Path Issue: PlatformIO 6.1.16 Incorrect Path Resolution When Linking Library Files

AliveGh0st, 2024-10-23:
### Description

While using PlatformIO 6.1.16 for project development, I encountered an issue with relative path resolution. Specifically, when using the `-Wl,-static` flag and relative paths to link library files, the path `./` incorrectly points to `C:\Users\admin\.platformio\platforms\ststm32\builder` instead of the project root directory, resulting in an error during the linking process.

### Steps to Reproduce

1. Configure `platformio.ini` with the following `build_flags`:
   ```ini
   build_flags = 
       -Wl,-static
       ./ExternalLibs/Library1/lib1.a
       ./ExternalLibs/Library2/lib2.a
       ./ExternalLibs/Library3/lib3.a
       ./App/Main/libmain.a

2.Run the pio run command in the project root directory.
3.During the linking of firmware.elf, observe the path resolution error and subsequent linking failure.
Expected Behavior
The relative path `./` should point to the project root directory, allowing the library files to be correctly linked. Although using absolute paths can solve the issue, it is not conducive to project portability.

### Actual Behavior
The relative path `./` points to `C:\Users\admin\.platformio\platforms\ststm32\builder` , causing the library file paths to be incorrectly resolved and the linking process to fail.

### Error Log
The following error log is produced during the linking process:
 ```ini
  *** [.pio\build\genericSTM32F103ZE\firmware.elf] Implicit dependency `C:\Users\admin\.platformio\platforms\ststm32\builder\ExternalLibs\Library1\lib1.a' not found, needed by target `.pio\build\genericSTM32F103ZE\firmware.elf'.
```
### System Information
- PlatformIO Version: 6.1.16
- Operating System: Windows 10
- PLATFORM: ST STM32 (17.6.0) > STM32F103ZE (64k RAM. 512k Flash)
- HARDWARE: STM32F103ZET6 72MHz, 64KB RAM, 512KB Flash
- DEBUG: Current (stlink) External (blackmagic, cmsis-dap, jlink, stlink)
- PACKAGES: toolchain-gccarmnoneeabi @ 1.120301.0 (12.3.1)

### Additional Information
Using absolute paths resolves the issue, but it would be preferable to use relative paths, with the relative paths correctly pointing to the project root directory.

Thank you for your support and assistance!

ivankravets, 2024-10-23:
Could you try a temporary solution?
```ini
build_flags = 
  $PROJECT_DIR/ExternalLibs/Library1/lib1.a
``

AliveGh0st, 2024-10-23:
not work.
 ```ini
 *** [.pio\build\genericSTM32F103ZE\firmware.elf] Implicit dependency `C:\Users\admin\.platformio\platforms\ststm32\builder\$PROJECT_DIR\ExternalLibs\Library1\lib1.a' not found, needed by target `.pio\build\genericSTM32F103ZE\firmware.elf'.

commit 014ac79c (2025-02-13): Resolved an issue with incorrect path resolution when linking static libraries via the `build_flags` // Resolve #5004
closed 2024-12-12 by ivankravets