platformio/platformio-core #5007: pio pkg publish 6.1.16 issue ? "ManifestValidationError: Invalid manifest fields:"

mathieucarbou, 2024-10-26:
I am unable to publish the new version of MycilaJSY library:

https://github.com/mathieucarbou/MycilaJSY/blob/main/library.json

I am within the bounds of the field length though.

```
~/.../me/MycilaJSY ( main → origin {1} ✓ )
❯  pio --version
PlatformIO Core, version 6.1.16

~/.../me/MycilaJSY ( main → origin {1} ✓ )
❯  pio pkg publish
Preparing a package...
ManifestValidationError: Invalid manifest fields: {'examples': {315: {'name': ['Length must be between 1 and 255.']}, 316: {'name': ['Length must be between 1 and 255.']}, 317: {'name': ['Length must be between 1 and 255.']}, 318: {'name': ['Length must be between 1 and 255.']}, 319: {'name': ['Length must be between 1 and 255.']}, 320: {'name': ['Length must be between 1 and 255.']}, 321: {'name': ['Length must be between 1 and 255.']}, 322: {'name': ['Length must be between 1 and 255.']}, 324: {'name': ['Length must be between 1 and 255.']}, 325: {'name': ['Length must be between 1 and 255.']}, 327: {'name': ['Length must be between 1 and 255.']}, 328: {'name': ['Length must be between 1 and 255.']}, 329: {'name': ['Length must be between 1 and 255.']}, 330: {'name': ['Length must be between 1 and 255.']}, 331: {'name': ['Length must be between 1 and 255.']}, 332: {'name': ['Length must be between 1 and 255.']}, 333: {'name': ['Length must be between 1 and 255.']}, 334: {'name': ['Length must be between 1 and 255.']}, 335: {'name': ['Length must be between 1 and 255.']}, 336: {'name': ['Length must be between 1 and 255.']}, 337: {'name': ['Length must be between 1 and 255.']}, 338: {'name': ['Length must be between 1 and 255.']}, 339: {'name': ['Length must be between 1 and 255.']}, 340: {'name': ['Length must be between 1 and 255.']}, 341: {'name': ['Length must be between 1 and 255.']}, 342: {'name': ['Length must be between 1 and 255.']}, 343: {'name': ['Length must be between 1 and 255.']}, 344: {'name': ['Length must be between 1 and 255.']}, 345: {'name': ['Length must be between 1 and 255.']}, 346: {'name': ['Length must be between 1 and 255.']}, 347: {'name': ['Length must be between 1 and 255.']}, 348: {'name': ['Length must be between 1 and 255.']}, 349: {'name': ['Length must be between 1 and 255.']}, 350: {'name': ['Length must be between 1 and 255.']}, 351: {'name': ['Length must be between 1 and 255.']}, 352: {'name': ['Length must be between 1 and 255.']}, 353: {'name': ['Length must be between 1 and 255.']}, 354: {'name': ['Length must be between 1 and 255.']}, 355: {'name': ['Length must be between 1 and 255.']}, 356: {'name': ['Length must be between 1 and 255.']}, 357: {'name': ['Length must be between 1 and 255.']}, 358: {'name': ['Length must be between 1 and 255.']}, 359: {'name': ['Length must be between 1 and 255.']}, 360: {'name': ['Length must be between 1 and 255.']}, 361: {'name': ['Length must be between 1 and 255.']}, 362: {'name': ['Length must be between 1 and 255.']}, 363: {'name': ['Length must be between 1 and 255.']}, 364: {'name': ['Length must be between 1 and 255.']}, 365: {'name': ['Length must be between 1 and 255.']}}}. 
Please check specification -> https://docs.platformio.org/page/librarymanager/config.html
```

ivankravets, 2024-10-27:
Did you solve the issue? I see that the library is published https://registry.platformio.org/libraries/mathieucarbou/MycilaJSY

mathieucarbou, 2024-10-27:
No it is not solved. I am trying to publish the one in main : 11.0.4 and the json file matches the specs. I don't understand.

I am published yesterday and this morning several versions of other libraries (AsyncTCP, ESPAsyncWebSerevr ,etc).

Only this one blocks.

ivankravets, 2024-10-27:
I can't reproduce the issue. See the screenshot below. Could you try downloading this archive: https://github.com/mathieucarbou/MycilaJSY/archive/refs/tags/v11.0.4.zip and repeating it?

[image]

mathieucarbou, 2024-10-27:
Oh you are using the zipped version from a tag! I will try. Didn't think of this workaround!

mathieucarbou, 2024-10-27:
Yes it works when going through the archive.
I used to do `pio pkg publish` at the root location of the project.

ivankravets, 2024-10-27:
Great. It seems that your local version (cloned) != the source code hosted on GitHub.

Happy coding with PlatformIO!

closed 2024-10-27 by ivankravets

mathieucarbou, 2024-10-27:
Thanks a lot!