platformio/platformio-core #5013: Unsupported board

DecXdrones, 2024-11-06:
- [ ] **Feature Request**.
      Please add Anycubic Kobra Neo motherboard to the supported boards.
------------------------------------------------------------------

### Configuration

**Windows 10**:

**PlatformIO Version 3.3.3**:

### Description of problem
I've spent the last 4 days trying to update my Anycubic Kobra Neo to the most recent version of Marlin and all I get are different message errors either for unknown board or directory and files not existing. I've downloaded the files from the Anycubic github repository and the only way I found to get close to something working so far is by adding their files one by one to the marlin files based on what error message I'm getting from VSCode.

#### Steps to Reproduce

1. Download the Kobra Neo Master file from 
https://github.com/ANYCUBIC-3D/Kobra_Neo
2. Download the latest version of Marlin
3. Unzip both files
4. In the directory (C:\Marlin-2.1.2.4\Marlin) change the configuration.h , configuration.adv.h and version.h files for the ones found in the Kobra neo master file. 
5. Try compiling
6. You should get an unknown board error. 
7. You can then go to (Kobra_Neo-master\source\Marlin\src\core) to find the boards.h file. Copy it and replace the one in the marlin file (Marlin-2.1.2.4\Marlin\src\core)
8. Then go to (Kobra_Neo-master\source\Marlin\src\pins) and copy the .h files visible in that file. Paste them in (Marlin-2.1.2.4\Marlin\src\pins)
9. You then get an unlimited subscription to error messages for differents files missing from the repository. everytime you add a file you get another message stating another one is missing.

### Actual Results
Error messages stating that some files are missing. If I add the files asked by VSCode in the correct directory I get another message for a missing file. I am now stuck at a "Stream.h" missing file eventhough it is in the same directory as all the previous missing files that are now not giving me an error message

### Expected Results
Compiling the fimware to obtain a working firmware.bin file to insert into my printer

### Additional info
I don't know if you are the right people to ask for that kind of problem and I am very sorry to bother you if you aren't. I'm just currently running out of solution. Have a nice day

ivankravets, 2024-11-06:
Please forward to https://github.com/MarlinFirmware/Marlin

closed 2024-11-06 by ivankravets

DecXdrones, 2024-11-06:
Okay perfect. Thank you for redirecting me in the right direction. I appreciate it.