platformio/platformio-core #5018: FYI - Build speed optimization - ccache massively speeds up build time

zackees, 2024-11-12:
Hi there, I'm making an emscripten docker file for platformio for a web compile of FastLED.

I've been grinding on build speed for the last week. I want to report that one of the biggest wins was slapping on ccache in front of the compiler.

As many have mentioned before, PlatformIO has a tendency to rebuild everything whenever order of deps/defines changes, which makes sense. ccache is a great way to mitigate that effect.

Today I discovered auto clean and disabling it. This makes it even faster. However I'm not sure what the failure condition is going to be with disabling auto-clean.

This compiler project is kind of a special case. Users pretty much just compile against FastLED. So my use case is how much can I pre-warm the ccache in release/debug/quick modes, stash that as a docker image layer. Then when a file is sent it just compiles then link against fastled. Right now I'm still dealing with linking against a bunch of object files. I've tried to combine this into a giant lib but no luck so far.

zackees, 2024-11-12:
By the way, a super fast cloud compiler for platformio would be amazing. There is so much caching that can be done between all these different projects. There is also the benefit that the user doesn't have to install a giant framework to get something to compile.

In my case, I have a tool called fastled-wasm that just zips up a sketch (which is tiny) and the backed compiles it all in 4 seconds then sends it back. My 4 second compile is on a super weak cloud server at $7 a month.

Platformio, if it did this, could just install the necessary tools to upload the hex/bin to the actual device, and not do the 1.6GB download for RP2350 or the ESP32S3 for the framework dependencies.

Anyway, probably outside the scope of what PlatformIO wants to do. So I just wanted to share the result of the benefits that I'm seeing.

ivankravets, 2024-12-04:
So, and how is bad https://docs.platformio.org/en/latest/projectconf/sections/platformio/options/directory/build_cache_dir.html ?

Jason2866, 2024-12-04:
@robertlipe I tried cchache with Platformio a while ago. There is a speed gain even when Platformios cache function is active. Since now and than unexplainable strange compiler and/or linker error occurred i trashed this approach to integrate in my fork.
You can try (the not anymore maintained) branch https://github.com/Jason2866/platform-espressif32/tree/ccache

zackees, 2024-12-05:
> Platformio has a [build cache](https://docs.platformio.org/en/latest/projectconf/sections/platformio/options/directory/build_cache_dir.html) but for small files, I'm not convinced that it's any faster than just recompiling the objects anyway - which it does completely too much.
> 
> I'm intrigued by "slapping on ccache". Did you modify scons directly to do this? I once tried the trick of making a front-end wrapper for g++ and friends that called ccache with argv[] but pio build outsmarts that by not running, e.g. xtensa-esp32-elf-g++ from the $PATH (as it has to be clever about calling xtensa-esp32- or xtensa-esp32s3- or riscv-) and instead reaches directly into /.platformio/packages/toolchain- and calls them from there. I wasn't thrilled about replacing those tools with front-ends to ccache.
> 
> Since this thing wants to check dependencies in Python, it's often slower than just building the stupid code.
> 
> Have you successfully built a maintainable configuration that calls distcc and/or ccache?
> 
> As you learned when building nightdriverled, platformio fetches and builds 39 nearly identical copies of most object when building, so those hour-long builds (which almost nobody but us maintainers do) are worth some pain to help.

What I mean by slap it on I mean I do a swap of environmental variables at "CC", "CXX", and "LINK" with scons so that instead of the "compiler" it's "ccache compiler". I AM doing this for the emscripten compiler instead of something like avr-gcc, but I think it works the same.

This is for a web compiler so it's MUCH more sensitive on CPU speed than a home computer. This web compiler was getting 8 seconds for a simple re-compile of the 8 or so object files for FastLED, and this dropped to less than a second for all 8 as I'm assuming it was hitting the ccache.

@ivankravets I am overriding the build directory, but what I'm doing is deploying client code into a specific directory and trying to use the previous compilation of shared code again. Something about scons is triggering a rebuild. I turned off the auto-clean feature and this massively sped it up and solved the build time problem, but then I started getting weird errors with the linker so I think I would have to do an in-depth manual clean step to be able to use this feature.

I haven't whether determined putting ccache in front of avr-gcc would work, but I strongly suspect that's the case because ccache is essentially just a kv database mapping pre-processed C/C++ code to object blobs.

The use case this project is pretty optimal, since the code I'm compiling is a user sketch vs FastLED. The ccache is great here because it solves about 90% of the problem. Scons is still over-calculating what needs to be compiled but ccache makes this a non issue. The fast path for a cached file is 30ms vs 1000ms without ccache per object file. FastLED does have some nasty header issues and tends to pull the whole world for each translation unit. So our library is very sensitive to re-compilations and that's not going to change any time soon.

To see how I did it, see our custom build flags file for platformio you can see it here:

https://github.com/FastLED/FastLED/blob/master/src/platforms/wasm/compiler/wasm_compiler_flags.py

Keep in mind we are doing an emscripten build for the upcoming web compile feature for FastLED. It turns out that platformio is extremely compatible in this with the only downside being that platformIO seems to have internal locks that prevent concurrent builds. I've mitigated this so far by using GCC syntax checking to fast fail invalid sketches so only valid C++ makes it to the critical section. And later, I may rip out the platformio build system altogether and just go with a CMake build system instead to unlock concurrent builds

Whats also surprising is how easy it is to use the emscripten compiler. All you have to do is swap out CC/CXX/LINK in the env variables and that's pretty much it. Obviously a lot of code that's platform specific especially with ASM won't compile, but all of the logic code pretty much does as clang is nearly identical to the familiar gcc toolchain behavior.

puzrin, 2025-06-29:
> So, and how is bad https://docs.platformio.org/en/latest/projectconf/sections/platformio/options/directory/build_cache_dir.html ?

This does not help with esp32 idf-sdk, BUT(!) this ini modifications caches everything:

```
[platformio]
build_cache_dir = .pio/cache

[env:main]
build_flags =
   ${env.build_flags}
   -D IDF_CCACHE_ENABLE=1
#board_build.cmake_extra_args = -DCCACHE_ENABLE=ON
```

Build time 4m30s => 30s. The difference is huge, 9x.

Also tried commented out `-DCCACHE_ENABLE=ON` instead of `IDF_CCACHE_ENABLE`. It helped partially, but not too much (1.5x-2x). Without `build_cache_dir`, ccache does not work.

No idea what happens inside, and whether it will be correct for any situation. Currently, the provided config helps with two cases:

- `platformio.ini` changed
- sdk config changed

Both cause a complete rebuild, but all files are taken from cache.

Note: env "Clean" does not touch the cache. Not sure is it good or bad.

@ivankravets, what do you think? Can this be used safely? Maybe this can be added to defaults or recommended in the docs?

zackees, 2025-06-30:
arabia for this write up. I'm going to try it

valeros, 2025-06-30:
Hi @puzrin, why put the PlatformIO cache inside the `.pio` folder? It won't work because the cache gets deleted along with the build files when you clean the project or change configuration files, so everything has to be rebuilt from scratch.

valeros, 2025-06-30:
@puzrin you can use the PlatformIO cache the same way as ccache, just place the cache in a persistent folder (for example near the `.pio` folder).

puzrin, 2025-06-30:
> [@puzrin](https://github.com/puzrin) you can use the PlatformIO cache the same way as ccache, just place the cache in a persistent folder (for example near the `.pio` folder).

I don't understand your point. What problem do you try to solve? The posted config works. My point is "probably, it can be improved, but I don't know how".

valeros, 2025-06-30:
I mean there is no need to use an additional caching mechanism (`IDF_CCACHE_ENABLE`, `CCACHE_ENABLE=ON`, etc.) if you are already using the PlatformIO cache, it works the same way:
```ini
[platformio]
build_cache_dir = .cache

[env:main]
build_flags =
   ${env.build_flags}

# Disabled
;   -D IDF_CCACHE_ENABLE=1
; board_build.cmake_extra_args = -DCCACHE_ENABLE=ON
```
Then run your project two times, first is the clean build from scractch, second is the build with cache.

puzrin, 2025-06-30:
You are right. My mistake was placing the cache in the `.pio` folder. With your snippet, the secondary rebuild is approximately as fast as mine.

I think. It's worth clarifying the docs https://docs.platformio.org/en/latest/projectconf/sections/platformio/options/directory/build_cache_dir.html.

- Add a concrete value `.cache` to the example, instead of an empty one.
- Add a note, the cache should be outside of the `.pio` folder
- (?) Promote using this option, especially for esp-idf.

In my opinion, the issue can be considered resolved (although I'm not an author).

/cc @zackees

zackees, 2025-06-30:
I'm running it right now. The first pass is without ccache and just using .pio_cache

Jason2866, 2025-07-01:
I have a script written which analyses the first compile and stores all Infos LDF generates.
In second run LDF is switched off and all compiled files in `.pio/build/<env>` are used without the time consuming LDF run. The only time consuming step is the Linker process now. BUT this LDF cache logic is ONLY working and possible with LDF mode `chain`.
Needs setting Platformio Core setting `--disable-auto-clean` too keep files in folder `.pio` since  the setting `lib_ldf_mode` is automatically changed to off from chain for second run.
The script (no explanations no support!!) is here https://github.com/Jason2866/Tasmota/blob/ldf_clean_up/pio-tools/ldf_cache.py

zackees, 2025-07-01:
I weird - and yes i've thought about doing the exact same thing.

@ivankravets are you using Cursor IDE or Windsurf IDE yet?

These are one of those problems a background agent with max mode could just solve with the info we've provided so far.

Jason2866, 2025-07-02:
> The entire FastLED build pipeline is built on top of platformio. Show me another command line tool that can build every product on the market.

Don't get me wrong i like Platformio and yes that's the big nice thing. But there is no relevant development done the last months. The buggy slow LDF is still there, which needs a complete refactor. This was promised long time ago.

ivankravets, 2025-07-02:
> But there is no relevant development done the last months. 

The 6.x branch is currently in bug-fix mode. We are actively working on version 7.0 internally and do not plan to refactor LDF in the 6.x series.

Jason2866, 2025-07-02:
That are good news, planned release date for 7.x?

zackees, 2025-07-02:
> > I'm seeing esp32s3 rebuild constantly.
> 
> Could you try comparing `pio run -v` outputs between build iterations? If you have volatile flags (e.g., macros), they could be causing a full project rebuild. For example: `-D CURRENT_TIME=$UNIX_TIME`.

pio run on project directories runs great, i just tried it on teensy.

I think the problem is that i'm using `pio ci` and specifying project flags via the runtime. I suspect this is like modifying the platformio.ini - it's going to trigger a build. So I'm putting everything under unit test and I'm going to go low level:

  * board definition -> platformio.ini file -> strip comments and white space -> fingerprint-> re-initialize on fingerprint change
  * Then selectively apply disable auto clean.
  * I might have to pull out libfastled and other archives and symlink them and do -I back to the library.

ivankravets, 2025-07-03:
See https://docs.platformio.org/en/latest/core/userguide/cmd_ci.html#cmdoption-pio-ci-keep-build-dir

Please note that `pio ci` is not intended for the incremental builds.

zackees, 2025-07-03:
@Jason2866 a big part of the slow down with the esp platforms is that the official link to the platform files is pointing at an absolutely gigantic repo.

Have you ever created an orphan branch with no history but all the files?

I think this would go very fast.

zackees, 2025-07-04:
My hack seems to work very well. Most of the trick is about shutting down auto clean. There is of course way more optimization that can be done, but at this point most of the wasted time is recovered. Some things, like intercepting the lib dependencies is trivial, but the tool dependencies are more complicated. So it looks like my front end is going to be more of a hack than a replacement.

Keep in mind, I'm vibe coding most of this. This might seem like madness to many developers but Vibe coding is incredibly effective if you stick to strict TDD and use types everywhere.

After two days drilling away on this of I've come to the following conclusions:

  * The native "dev" platform that platformio has gently tucked away IS the test case for lightning-fast unit tests. After that it's the uno platform. Everyone else is slow. Esp32s3 is probably 30 minutes for a platform/tool download.
  * PlatformIO is not ready for AI coding yet because
    * It's using typeless python. Humans hate typeless python and so does AI. AI will fuck up this code base if let lose.
      * @ivankravets my recommendation is to skip past mypy and go directly to pyright, which handles types between modules and external packages. Mypy is already done a linter because surprisingly it's a single file type checker.
    * Missing unit tests. This is not a TDD project. Hence all these bugs with the LDF. Without full coverage the AI will inject entropy and cause lots of regressions.

For AI to turn this project around @ivankravets would need about 1 day of using Cursor AI + Background agents and about $100 spend on AI to put in typing. After that, he can start instructing the AI agents to add unit tests. This would go quickly, probably one or two days with multiple background agents running in parallel, with snags on unexpected bugs emerging that the tests expose.

If the repo was converted this way then the entire code base would be like a giant AI where you just ask for features and 5 minutes later it's done. That's where FastLED is at this point. It's mesmerizing to just babysit and watch 5 different AI's write code, test their assumptions, pull open the binaries with tools like `nm` and just go to town figuring out why a symbol isn't linking in. I dont' even have to overdefine the task, it knows how to find the avr toolset for example and start grepping for symbols.

I can't even tell most of the FastLED followers on reddit about the AI stuff. I made one post about how FastLED is compatible with AI and the amount of hate this triggered was stunning. Most people downvoted and are brigading other mentions of AI in the reddit forums.

Though it's not entirely uncalled for. AI is like a nuke. The amount of propaganda out there now saying that "AI" will make you a bad coder has to be one of the top propaganda wins by the state I've ever seen. The AI criticism about code quality should have evaporated by now because it's all mitigatable with TDD. Any place that doesn't have TDD, like a process boundary for communication, becomes surface area for AI entropy and weird breakages if you don't watch it like a hawk.

My opinion is that AI is like wielding a nuke and any software project that doesn't use nukes are going to be re-written by someone else using AI.

For now, i'm not going to fork platformio and fix it's problems. That seems like stepping on feet. I'm going to stick with a front end because it solves FastLED's build latency nightmare, where our S3 platform takes 30 minutes on linux and over an hour on Windows to run 48 ino tests. That's unacceptable.

And this isn't a threat to fork platformio by me, it's just an observation that if Ivan doesn't get on the bandwagon of AI, *someone* will just naturally fork the project, add types, add tests, and then fix all the bugs and then make it blazing fast.

zackees, 2025-07-04:
It would be outdated if something were to replace it. The only other tool is the arduino tool stack, which is even worse.

zackees, 2025-07-04:
@Jason2866 I hope espressif is able to mock out a native version of their HAL or can emulate a one of the simpler boards. Or you will face the same pressure very soon.

@ivankravets 

I'm putting in inlined build settings into the ino in the same way that astral-uv is doing it. I humbly suggest you add this style of embedded attributes into your build system. It solves a lot of problems.

*Sketch.ino*
```
/// SKETCH-INFO
/// dependencies = ["FastLED"]
/// defines = ["FASTLED_USE_PROGMEM=0"]
/// SKETCH-INFO
```

Right now I've got the dependencies section working, which will put the deps into my so called "turbo deps" side loader which gives a nice performance boost.

zackees, 2025-07-07:
These tests are going to take more than an hour to run. Holy shit.

ivankravets, 2025-07-08:
> I realized last night there's no way to set build_flags for individual
files

You can do much more than defining macros https://docs.platformio.org/en/latest/scripting/middlewares.html

TD-er, 2025-07-08:
> > I realized last night there's no way to set build_flags for individual
> > files
> 
> You can do much more than defining macros https://docs.platformio.org/en/latest/scripting/middlewares.html

I have used that in the past, but it sometimes did result in odd build results which magically disappeared when doing a clean build.

Now I call a [pre: Python script to generate a specific .cpp file](https://github.com/letscontrolit/ESPEasy/blob/mega/tools/pio/generate-compiletime-defines.py) which does contain all my compile time defined strings and numbers like the build filename, build environment, Git branch and more.
This way there is only a single .cpp file to rebuild on each build, apart from the other changes I made of course.

For ESP8266 builds I still append these defines to the `CXXFLAGS`, since I have to build my project completely different for ESP8266.
Problem with ESP8266 is that I still need to use an older toolchain which doesn't really work well in Windows as it tries to add all .o files on a single command line when linking.
So for ESP8266 I just concatenate all .cpp files into a temp .cpp file and add the rest to the ignore list.
This way it is still a compile time which is perfectly acceptable and my project files are all being processed as a single .cpp file.

My ESP32-xx builds are now also quite acceptable regarding build speed as I just tried to (re)build a 2.6 MB binary for ESP32P4 in 1m06sec.

```
Environment                         Status    Duration
----------------------------------  --------  ------------
max_ESP32p4_16M8M_LittleFS_CDC_ETH  SUCCESS   00:01:06.683
```

N.B. this is using WSL2 on Windows 11.
Building on a Windows file system with everything build-related excluded by the virusscanner etc. takes 10+ minutes. Easily way more than a factor 5 - 10x slower on exactly the same hardware.

zackees, 2025-07-10:
> Consider replacing it.
> 
> 
> 
> The value is in reading a user's Platformio.ini. ini parsers are a dime a
> 
> dozen. Consider reading that, perform the (sad) interpolation that it does,
> 
> build a Cmakefiles or Ninja or something that's actually fast. Bonus to get
> 
> tooling and that does dependcy checking accurately (and fast) that knows
> 
> about distcc and cache and other 20+ y/o industry norms.
> 
> 
> 
> Oh, I suppose it needs to know how to parse  the boards tree, too, but even
> 
> if that's a manual setup stage (MOST people aren't us and building on 40+
> 
> boards - and those that do wouldn't mind a bit of manual one-time setup)
> 
> that's not totally terrible.
> 
> 
> 
> Its not trivial but it might be easier than trying to fix a foundation of
> 
> python and axons.
> 
> 
> 
> I realized last night there's no way to set build_flags for individual
> 
> files. This means even if only one file needs a define, if you touch
> 
> build_flags for a define in one file, that shoots down the build cache for
> 
> the entire project,. Including third-party projects that know nothing about
> 
> that flag. Sad. There's a terrible workaroud proposed by Max, but it
> 
> triggers a pio-core bug that's been open for years.
> 
> 
> 
> 
> 
> 
> 
> 
> 
> >
> 
> > These tests are going to take more than an hour to run. Holy shit.

So now that I put 12 hours of my time.

And about $100 sunk into sonnet MAX thinking mode (the state of art programming AI).

Let me just me be, I'm not going to go forward 

Ivan created platform IO before python got fun. When he staged torijecg When Ivan started on this the concept 

This is an ancient code base from the python 3.6 days, when pylint was the best linter in town.

Theres a lot code. And started in python 3.6 days.

This is an ancient code base back when 

I now have a different appreciation for this code than I did a week ago.

First off, platformio is about a decade old, from the python 3.6 or 3.7 era. This was when pylint was THE linter 

This was before mypu for pillar no, when most linting was done by pylint

I want to let  money with the sonnet 4.0 MAX model I now have a different opinion and I understand the dcale

 I'm going not go , I now have an appreciation 

getting types in and experiment  to fix the typing

zackees, 2025-07-10:
> Well.... I did have to tune and tweak a bit and mainly also copy all used libs into my repo (ESPEasy) to make sure not every other GHActions build failed.

This is the way. once you get to a certain scale just copy everything into third party. Deps are great for prototyping or libraries. Not for products. It's just the wrong solution.

> Do you have a link describing those 'thin archives' ?

You just pass `T` into your linker and thats it. Easiest optimization I've ever done.

From ChatGPT

```

A **thin archive** is simply a static‐library archive that contains only:

1. A symbol index, and
2. References (paths) to the original object files

instead of embedding full object‐file contents inside the `.a` file itself.

---

## How Thin Archives Differ from “Normal” Archives

* **Normal archives** (`.a`)

  * Store the complete contents of each object file as members of the archive.
  * Can be linked anywhere without needing the original object file.
* **Thin archives**

  * Store only a lookup table (symbol index) plus the paths to each member file.
  * Do *not* embed the full object data; when you link against a thin archive, the linker reads the original `.o` files directly ([[Sourceware](https://sourceware.org/binutils/docs/binutils/ar.html?utm_source=chatgpt.com)][1]).
  * Are “flattened” when nested—adding a thin archive into another thin archive expands its members rather than nesting archives ([[Sourceware](https://sourceware.org/binutils/docs/binutils/ar.html?utm_source=chatgpt.com)][1]).
  * Use the special magic string `!<thin>` in their header to distinguish them ([[Wikipedia](https://en.wikipedia.org/wiki/Ar_%28Unix%29?utm_source=chatgpt.com)][2]).

---

## Benefits

* **Reduced disk I/O & space**

  * No duplication of object-file contents if you’re building multiple intermediate libraries in the same tree.
* **Faster archive creation**

  * The archiver need only build the symbol index, not copy bytes from each object.
* **Better for incremental builds**

  * When you rebuild an object file, you don’t have to re‐copy it into every static library that uses it—just rebuild the symbol index.

---

## Limitations

* **Must keep original object files around**

  * If you move or delete an object file, the thin archive will break at link time (“file not found”).
* **Not universally supported**

  * Most modern linkers (GNU ld, LLVM’s lld) handle thin archives, but some toolchains may not.
* **Cannot convert in place**

  * You can’t switch a normal archive to thin (or vice versa) without recreating it from scratch ([[Sourceware](https://sourceware.org/binutils/docs/binutils/ar.html?utm_source=chatgpt.com)][1]).

---

## Creating & Using Thin Archives

* **GNU `ar`** (since Binutils \~2.18):

  ```bash
  ar crT libfoo.a foo.o bar.o
  ```

  The `T` modifier means “create a thin archive” ([[LLVM Reviews](https://reviews.llvm.org/D116979?utm_source=chatgpt.com)][3]).
* **LLVM’s `llvm-ar`**:

  ```bash
  llvm-ar --thin -cr libfoo.a foo.o bar.o
  ```
* **Linking** is identical to a normal static library:

  ```bash
  g++ main.o -L. -lfoo -o myapp
  ```

---

### When to Reach for Thin Archives

* Large codebases where many intermediate static libs share the same objects.
* High‐performance build systems aiming to minimize file copying.
* Environments where disk space or I/O are at a premium (e.g., embedded‐Linux SDKs, Docker layers).

Beyond these scenarios, the extra bookkeeping of ensuring object‐file paths stay valid may outweigh the benefits—but in the right context, thin archives can shave significant time and space off your C++ builds.

[1]: https://sourceware.org/binutils/docs/binutils/ar.html?utm_source=chatgpt.com "ar (GNU Binary Utilities) - Sourceware"
[2]: https://en.wikipedia.org/wiki/Ar_%28Unix%29?utm_source=chatgpt.com "Ar (Unix)"
[3]: https://reviews.llvm.org/D116979?utm_source=chatgpt.com "D116979 [llvm-ar] Add --thin for creating a thin archive"r

```

zackees, 2025-07-10:
The allsrc build was harder to conjure up. But now I can build fastled for wasm for release, quick, debug in 14 seconds instead of 60.

If it's just quick build it's like 3 seconds, cold compile full link on linux. Everything in fastled compiles into one source file.

zackees, 2025-07-10:
@ivankravets 

Here's all the work i did to modernize your codebase. It's about 12 hours.

https://github.com/platformio/platformio-core/pull/5211

Hopefully you find some value in it.

zackees, 2025-07-11:
> > > Do you have a link describing those 'thin archives' ?
> > 
> > You just pass `T` into your linker and thats it. Easiest optimization I've ever done.
> 
> Do you have an example of where to add this in the pio build flags?
> Just `-T` ?
> That doesn't seem right, as it also seem to require a 'script' argument.
> See: https://gcc.gnu.org/onlinedocs/gcc/Link-Options.html#index-T

Yes I have an example.

Here is an example of configuring platformio for thin archives

https://github.com/zackees/fastled-wasm-compiler/blob/main/assets/wasm_compiler_flags.py#L198

Jason2866, 2025-08-07:
ccache is not faster as Platformio cache, which needs to be activated by a setting. It is not default

zackees, 2025-08-07:
Wow this everything that I wanted

Jason2866, 2025-08-07:
just add as pre script in your platformio.ini like `pre:ldf_cache.py` and add / change the ignore files and directorys (it is documented in the script) And yes the script was made with the help of AI but without clear prompts whats needed it would have never reached a working state.

zackees, 2025-08-08:
Look at @Jason2866's pre: script in scons format. It does everything I was look to do. It saves the build dependency information, then shuts off LDF. Next build LDF is shut off and everything knows what to link against and which headers to put on the src path.

zackees, 2025-08-08:
`--disable-auto-clean` doesn't disable auto clean

When i run this at FastLED project root:

`bash compile uno Blink Audio Async`

I can see that the folder get's nuked after program close.

`bash compile uno Blink` does not do this

zackees, 2025-08-08:
per board build_cache_dir for the win.

Now it's fast.

This should be the default.