platformio/platformio-core #5029: --project-dir flag does not work correctly for check and debug commands

brawner, 2024-11-21:
What kind of issue is this?

- [X ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

### Configuration

**Operating system**:
Ubuntu 20.04

**PlatformIO Version** (`platformio --version`):
PlatformIO Core, version 6.1.16

### Description of problem
I generally work out of the root of my repo, but my platformio projects are nested a couple of levels. So I often run platformio like:

```
platformio run -d boards/my_project -e deployment
```

which works great for the `run` command. But I've noticed it fails on `check` and `debug` commands. Whereas moving to that directory and calling check/debug without the `-d` parameter works fine

#### Steps to Reproduce

```
$ platformio check -d boards/my_project -e deployment
$ platformio debug -d boards/my_project -e debug --interface=gdb -x .pioinit

```
### Actual Results
```
$ platformio check -d boards/my_project -e deployment
Checking deployment > clangtidy (platform: espressif32@6.4.0; board: esp-wrover-kit; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------
FileNotFoundError: Traceback (most recent call last):
  File "/home/brawner/.local/lib/python3.8/site-packages/platformio/__main__.py", line 103, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/usr/local/lib/python3.8/dist-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/home/brawner/.local/lib/python3.8/site-packages/platformio/cli.py", line 85, in invoke
    return super().invoke(ctx)
  File "/usr/local/lib/python3.8/dist-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.8/dist-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.8/dist-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/home/brawner/.local/lib/python3.8/site-packages/platformio/check/cli.py", line 152, in cli
    ct = CheckToolFactory.new(
  File "/home/brawner/.local/lib/python3.8/site-packages/platformio/check/tools/__init__.py", line 33, in new
    return cls(project_dir, config, envname, options)
  File "/home/brawner/.local/lib/python3.8/site-packages/platformio/check/tools/base.py", line 44, in __init__
    self._load_cpp_data()
  File "/home/brawner/.local/lib/python3.8/site-packages/platformio/check/tools/base.py", line 60, in _load_cpp_data
    data = load_build_metadata(self.project_dir, self.envname)
  File "/home/brawner/.local/lib/python3.8/site-packages/platformio/project/helpers.py", line 140, in load_build_metadata
    with fs.cd(project_dir):
  File "/home/brawner/.local/lib/python3.8/site-packages/platformio/fs.py", line 37, in __enter__
    os.chdir(self.new_path)
FileNotFoundError: [Errno 2] No such file or directory: 'boards/my_project'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `python -m pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

```

brawner, 2024-11-21:
Stacktrace for running `platformio debug`

```
FileNotFoundError: Traceback (most recent call last):
  File "/home/brawner/.local/lib/python3.8/site-packages/platformio/__main__.py", line 103, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/usr/local/lib/python3.8/dist-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/home/brawner/.local/lib/python3.8/site-packages/platformio/cli.py", line 85, in invoke
    return super().invoke(ctx)
  File "/usr/local/lib/python3.8/dist-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.8/dist-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.8/dist-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/brawner/.local/lib/python3.8/site-packages/platformio/debug/cli.py", line 109, in cli
    _run(project_dir, debug_config, client_extra_args)
  File "/home/brawner/.local/lib/python3.8/site-packages/platformio/debug/cli.py", line 177, in _run
    loop.run_until_complete(coro)
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "/home/brawner/.local/lib/python3.8/site-packages/platformio/debug/process/gdb.py", line 57, in run
    await self.spawn(*args, cwd=self.project_dir, wait_until_exit=True)
  File "/home/brawner/.local/lib/python3.8/site-packages/platformio/debug/process/base.py", line 76, in spawn
    await loop.subprocess_exec(
  File "/usr/lib/python3.8/asyncio/base_events.py", line 1630, in subprocess_exec
    transport = await self._make_subprocess_transport(
  File "/usr/lib/python3.8/asyncio/unix_events.py", line 197, in _make_subprocess_transport
    transp = _UnixSubprocessTransport(self, protocol, args, shell,
  File "/usr/lib/python3.8/asyncio/base_subprocess.py", line 36, in __init__
    self._start(args=args, shell=shell, stdin=stdin, stdout=stdout,
  File "/usr/lib/python3.8/asyncio/unix_events.py", line 789, in _start
    self._proc = subprocess.Popen(
  File "/usr/lib/python3.8/subprocess.py", line 858, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.8/subprocess.py", line 1704, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'boards/my_project'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `python -m pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

```

ivankravets, 2024-12-04:
Have you solved this issue?

commit d1531468 (2024-12-13): Resolved an issue where the ``--project-dir`` flag did not function correctly with the check and debug commands // Resolve #5029
closed 2024-12-13 by ivankravets