platformio/platformio-core #5034: eu2.contabostorage.com is blocked in Poland, consider using DNS-over-HTTPS

toriningen, 2024-12-01:
### Configuration

**Operating system**: Ubuntu 20.04

**PlatformIO Version** (`platformio --version`): `PlatformIO Core, version 6.1.16`

### Description of problem

Domain `eu2.contabostorage.com` is [blacklisted](https://hole.cert.pl/domains/v2/domains.txt) by cert.pl, which makes it unavailable in Poland, and this is the only mirror `platformio pkg` wants to work with, even after "trying a new mirror".

#### Steps to Reproduce

1. Be in Poland
2. Run `pio pkg install --force`

### Actual Results
```
Resolving uno dependencies...
Tool Manager: Installing platformio/toolchain-atmelavr @ ~1.70300.0
Tool Manager: Error: Please read https://bit.ly/package-manager-ioerror
Tool Manager: Warning! Package Mirror: HTTPSConnectionPool(host='eu2.contabostorage.com', port=443): Max retries exceeded with url: /c0a5502915964bc09d2607baa2984f98:pioregmirroreu/tools/cc/65/b2cd783a42ebb8d5148b287e67982efd054049001234afd0de29d0917297/toolchain-atmelavr-linux_x86_64-1.70300.191015.tar.gz (Caused by SSLError(SSLCertVerificationError(1, "[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: Hostname mismatch, certificate is not valid for 'eu2.contabostorage.com'. (_ssl.c:1145)")))
Tool Manager: Looking for another mirror...
Tool Manager: Error: Please read https://bit.ly/package-manager-ioerror
Tool Manager: Warning! Package Mirror: HTTPSConnectionPool(host='eu2.contabostorage.com', port=443): Max retries exceeded with url: /c0a5502915964bc09d2607baa2984f98:pioregmirroreu/tools/cc/65/b2cd783a42ebb8d5148b287e67982efd054049001234afd0de29d0917297/toolchain-atmelavr-linux_x86_64-1.70300.191015.tar.gz (Caused by SSLError(SSLCertVerificationError(1, "[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: Hostname mismatch, certificate is not valid for 'eu2.contabostorage.com'. (_ssl.c:1145)")))
Tool Manager: Looking for another mirror...
Tool Manager: Error: Please read https://bit.ly/package-manager-ioerror
Tool Manager: Warning! Package Mirror: HTTPSConnectionPool(host='eu2.contabostorage.com', port=443): Max retries exceeded with url: /c0a5502915964bc09d2607baa2984f98:pioregmirroreu/tools/cc/65/b2cd783a42ebb8d5148b287e67982efd054049001234afd0de29d0917297/toolchain-atmelavr-linux_x86_64-1.70300.191015.tar.gz (Caused by SSLError(SSLCertVerificationError(1, "[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: Hostname mismatch, certificate is not valid for 'eu2.contabostorage.com'. (_ssl.c:1145)")))
Tool Manager: Looking for another mirror...
^C
Aborted!
```

### Expected Results

Packages should be installed without issues.

### Additional info

DNS-over-HTTPS might help against "benevolent" censorship. Alternatively, let client ask for another mirror if the chosen one is blocked.

toriningen, 2024-12-02:
> If it fails, PlatformIO should pick another.

This doesn't happen. It keeps trying the same address over and over without success. I changed my system resolver to use DoH, this helped (since the address isn't blocked anymore). But no, if encountered error is SSL error (like I have quoted above), it just keeps going indefinitely trying to reach the same page.

More transparency about choosing mirrors might help.

ivankravets, 2024-12-03:
Could you navigate to `~/.platformio/packages` folder and remove `toolchain-atmelavr` folder.

Open PlatformIO Core CLI, change directory `cd` to the project folder and type:
```
pio run
```

Please provide the full output or screenshot.

toriningen, 2024-12-03:
```
ai-user@ai-user:~/CLionProjects/test-pio-serial$ rm -rf ~/.platformio/.cache ~/.platformio/packages/*
ai-user@ai-user:~/CLionProjects/test-pio-serial$ pio run
Processing uno (platform: atmelavr; board: uno; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing platformio/toolchain-atmelavr @ ~1.70300.0
Tool Manager: Error: Please read https://bit.ly/package-manager-ioerror
Tool Manager: Warning! Package Mirror: HTTPSConnectionPool(host='eu2.contabostorage.com', port=443): Max retries exceeded with url: /c0a5502915964bc09d2607baa2984f98:pioregmirroreu/tools/cc/65/b2cd783a42ebb8d5148b287e67982efd054049001234afd0de29d0917297/toolchain-atmelavr-linux_x86_64-1.70300.191015.tar.gz (Caused by SSLError(SSLCertVerificationError(1, "[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: Hostname mismatch, certificate is not valid for 'eu2.contabostorage.com'. (_ssl.c:1145)")))
Tool Manager: Looking for another mirror...
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: toolchain-atmelavr@1.70300.191015 has been installed!
Tool Manager: Installing platformio/framework-arduino-avr @ ~5.2.0
Tool Manager: Error: Please read https://bit.ly/package-manager-ioerror
Tool Manager: Warning! Package Mirror: HTTPSConnectionPool(host='eu2.contabostorage.com', port=443): Max retries exceeded with url: /c0a5502915964bc09d2607baa2984f98:pioregmirroreu/tools/8b/86/1f5fe99318d1656665f392581991f26121744a3599f58caaaca94bf8450d/framework-arduino-avr-5.2.0.tar.gz (Caused by SSLError(SSLCertVerificationError(1, "[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: Hostname mismatch, certificate is not valid for 'eu2.contabostorage.com'. (_ssl.c:1145)")))
Tool Manager: Looking for another mirror...
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: framework-arduino-avr@5.2.0 has been installed!
Tool Manager: Installing platformio/tool-scons @ ~4.40801.0
Tool Manager: Error: Please read https://bit.ly/package-manager-ioerror
Tool Manager: Warning! Package Mirror: HTTPSConnectionPool(host='eu2.contabostorage.com', port=443): Max retries exceeded with url: /c0a5502915964bc09d2607baa2984f98:pioregmirroreu/tools/7d/42/b0fefe6c6223b85c3e0375d07c400e0548dfbbfa2445860fcc475f37c909/tool-scons-4.40801.0.tar.gz (Caused by SSLError(SSLCertVerificationError(1, "[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: Hostname mismatch, certificate is not valid for 'eu2.contabostorage.com'. (_ssl.c:1145)")))
Tool Manager: Looking for another mirror...
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-scons@4.40801.0 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/uno.html
PLATFORM: Atmel AVR (5.1.0) > Arduino Uno
HARDWARE: ATMEGA328P 16MHz, 2KB RAM, 31.50KB Flash
DEBUG: Current (avr-stub) External (avr-stub, simavr)
PACKAGES: 
 - framework-arduino-avr @ 5.2.0 
 - toolchain-atmelavr @ 1.70300.191015 (7.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 6 compatible libraries
Scanning dependencies...
Dependency Graph
|-- avr-debugger @ 1.4.0
Building in release mode
Checking size .pio/build/uno/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   9.6% (used 196 bytes from 2048 bytes)
Flash: [=         ]   5.1% (used 1650 bytes from 32256 bytes)
======================================================================================= [SUCCESS] Took 9.17 seconds =======================================================================================
ai-user@ai-user:~/CLionProjects/test-pio-serial$ pio pkg install --force
Resolving uno dependencies...
Tool Manager: Removing framework-arduino-avr @ 5.2.0
Tool Manager: framework-arduino-avr@5.2.0 has been removed!
Tool Manager: Removing toolchain-atmelavr @ 1.70300.191015
Tool Manager: toolchain-atmelavr@1.70300.191015 has been removed!
Platform Manager: Removing atmelavr @ 5.1.0
Platform Manager: atmelavr@5.1.0 has been removed!
Platform Manager: Installing atmelavr
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Platform Manager: atmelavr@5.1.0 has been installed!
Tool Manager: Installing platformio/toolchain-atmelavr @ ~1.70300.0
Tool Manager: Error: Please read https://bit.ly/package-manager-ioerror
Tool Manager: Warning! Package Mirror: HTTPSConnectionPool(host='eu2.contabostorage.com', port=443): Max retries exceeded with url: /c0a5502915964bc09d2607baa2984f98:pioregmirroreu/tools/cc/65/b2cd783a42ebb8d5148b287e67982efd054049001234afd0de29d0917297/toolchain-atmelavr-linux_x86_64-1.70300.191015.tar.gz (Caused by SSLError(SSLCertVerificationError(1, "[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: Hostname mismatch, certificate is not valid for 'eu2.contabostorage.com'. (_ssl.c:1145)")))
Tool Manager: Looking for another mirror...
Tool Manager: Error: Please read https://bit.ly/package-manager-ioerror
[... cut ...]
Tool Manager: Looking for another mirror...
^C
Aborted!
ai-user@ai-user:~/CLionProjects/test-pio-serial$ 
```

Doing `pio run` or `pio pkg install` from clean cache and no preinstalled packages works.

Doing `pio pkg install --force` never works (keeps retrying endlessly).

Doing `pio pkg install --force`, then Ctrl+C, then `pio run` or `pio pkg install` never works (keeps retrying endlessly).

I was able to replicate the issue outside of Poland by adding this line to `/etc/hosts`:
```
0.0.0.0 eu2.contabostorage.com
```

ivankravets, 2024-12-10:
I can't reproduce this issue. Please check your system configuration. See the screenshot below (`0.0.0.0 eu2.contabostorage.com`):

[image]

Also, please run `pio system prune` and try again.

closed 2024-12-10 by ivankravets

YukiSuter, 2025-01-09:
I'm not in Poland, but similar issue with unresolved package mirror. It appears to only want to use usc1.contabostorage.com. I think this is a core bug since it never tries a different mirror. I can confirm manually trying this URL in browser fails, but a different mirror (https://dl.registry.nm1.platformio.org/) functions perfectly fine and downloads the relevant tool. It's also worth noting even when downloaded and placed into .platformio/packages, it doesnt recognise this and attempts to download from contabo aswell.

```

Tool Manager: Warning! Package Mirror: HTTPSConnectionPool(host='usc1.contabostorage.com', port=443): Max retries exceeded with url: /250040e607e8431abaff8727621872f9:pioregmirrorus/tools/07/a1/2ec67e82f538756d6aa62fa3519435fdd266e9cc16eec52fd6a36e1db113/tool-mkfatfs-windows_amd64-2.0.1.tar.gz (Caused by NameResolutionError("<urllib3.connection.HTTPSConnection object at 0x000001D23B0C2C90>: Failed to resolve 'usc1.contabostorage.com' ([Errno 11001] getaddrinfo failed)"))
Tool Manager: Looking for another mirror...
Tool Manager: Error: Please read https://bit.ly/package-manager-ioerror
Tool Manager: Warning! Package Mirror: HTTPSConnectionPool(host='usc1.contabostorage.com', port=443): Max retries exceeded with url: /250040e607e8431abaff8727621872f9:pioregmirrorus/tools/07/a1/2ec67e82f538756d6aa62fa3519435fdd266e9cc16eec52fd6a36e1db113/tool-mkfatfs-windows_amd64-2.0.1.tar.gz (Caused by NameResolutionError("<urllib3.connection.HTTPSConnection object at 0x000001D23B0D5190>: Failed to resolve 'usc1.contabostorage.com' ([Errno 11001] getaddrinfo failed)"))
Tool Manager: Looking for another mirror...
Tool Manager: Error: Please read https://bit.ly/package-manager-ioerror
Tool Manager: Warning! Package Mirror: HTTPSConnectionPool(host='usc1.contabostorage.com', port=443): Max retries exceeded with url: /250040e607e8431abaff8727621872f9:pioregmirrorus/tools/07/a1/2ec67e82f538756d6aa62fa3519435fdd266e9cc16eec52fd6a36e1db113/tool-mkfatfs-windows_amd64-2.0.1.tar.gz (Caused by NameResolutionError("<urllib3.connection.HTTPSConnection object at 0x000001D23B0D10D0>: Failed to resolve 'usc1.contabostorage.com' ([Errno 11001] getaddrinfo failed)"))
Tool Manager: Looking for another mirror...
Tool Manager: Error: Please read https://bit.ly/package-manager-ioerror
Tool Manager: Warning! Package Mirror: HTTPSConnectionPool(host='usc1.contabostorage.com', port=443): Max retries exceeded with url: /250040e607e8431abaff8727621872f9:pioregmirrorus/tools/07/a1/2ec67e82f538756d6aa62fa3519435fdd266e9cc16eec52fd6a36e1db113/tool-mkfatfs-windows_amd64-2.0.1.tar.gz (Caused by NameResolutionError("<urllib3.connection.HTTPSConnection object at 0x000001D23B033090>: Failed to resolve 'usc1.contabostorage.com' ([Errno 11001] getaddrinfo failed)"))
Tool Manager: Looking for another mirror...
Tool Manager: Error: Please read https://bit.ly/package-manager-ioerror
Tool Manager: Warning! Package Mirror: HTTPSConnectionPool(host='usc1.contabostorage.com', port=443): Max retries exceeded with url: /250040e607e8431abaff8727621872f9:pioregmirrorus/tools/07/a1/2ec67e82f538756d6aa62fa3519435fdd266e9cc16eec52fd6a36e1db113/tool-mkfatfs-windows_amd64-2.0.1.tar.gz (Caused by NameResolutionError("<urllib3.connection.HTTPSConnection object at 0x000001D23B0D3950>: Failed to resolve 'usc1.contabostorage.com' ([Errno 11001] getaddrinfo failed)"))
Tool Manager: Looking for another mirror...
Tool Manager: Error: Please read https://bit.ly/package-manager-ioerror
Tool Manager: Warning! Package Mirror: HTTPSConnectionPool(host='usc1.contabostorage.com', port=443): Max retries exceeded with url: /250040e607e8431abaff8727621872f9:pioregmirrorus/tools/07/a1/2ec67e82f538756d6aa62fa3519435fdd266e9cc16eec52fd6a36e1db113/tool-mkfatfs-windows_amd64-2.0.1.tar.gz (Caused by NameResolutionError("<urllib3.connection.HTTPSConnection object at 0x000001D23B0D6610>: Failed to resolve 'usc1.contabostorage.com' ([Errno 11001] getaddrinfo failed)"))
Tool Manager: Looking for another mirror...
```

smegg99, 2025-02-21:
still happens to me today
```bash
❯ pio pkg install
Resolving esp32-s3-mini-1 dependencies...
Tool Manager: Installing platformio/toolchain-riscv32-esp @ 14.2.0+20241119
Tool Manager: Error: Please read https://bit.ly/package-manager-ioerror
Tool Manager: Warning! Package Mirror: HTTPSConnectionPool(host='eu2.contabostorage.com', port=443): Max retries exceeded with url: /c0a5502915964bc09d2607baa2984f98:pioregmirroreu/tools/e7/dd/b4c4ecf515bd16cd8d70c483093fc40f3892344bc4fae5c563e92e6ed722/toolchain-riscv32-esp-linux_x86_64-14.2.0+20241119.tar.gz (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:992)')))
Tool Manager: Looking for another mirror...
Downloading  [------------------------------------]    0%  01:27:31^C
```

ivankravets, 2025-02-24:
Have you resolved the issue? Could you check your firewall/hosts?

msm-cert, 2025-04-28:
Hi,

Sorry about your trouble.

>DNS-over-HTTPS might help against "benevolent" censorship. Alternatively, let client ask for another mirror if the chosen one is blocked.

In the future, consider also raising a ticket when encountering a false-positive.

That subdomain was just taken down from the cert.pl's list. But there are downstream consumers which may not update frequently, in which case the domain may be unreachable for everyone for a while.