platformio/platformio-core #5047: Python error caused by empty args

PRugg-Cap, 2024-12-12:
I have a custom platform and I "start debugging" from the VSCode PlatformIO extension. I get a Python error caused by the following line:

https://github.com/platformio/platformio-core/blob/c1afb364e9039fac2838bd000227b0a1871a07a6/platformio/package/commands/exec.py#L57

It seems `args` is an empty list. This is fixed by changing the condition to `if args and args[0].endswith(...)`

I note in the rest of this function, there are checks that `args` is non-empty, e.g. https://github.com/platformio/platformio-core/blob/c1afb364e9039fac2838bd000227b0a1871a07a6/platformio/package/commands/exec.py#L41

ivankravets, 2024-12-12:
Thanks for the report! Please re-test with `pio upgrade --dev`.

closed 2024-12-12 by ivankravets

PRugg-Cap, 2024-12-13:
Thanks! All fixed

commit dd3fe909 (2024-12-12): Better handling of the missed args for exec command // Resolve #5047