platformio/platformio-core #5055: Bug: ws_esp32_s3_matrix.json appears invalid on linux builders only

zackees, 2024-12-20:
Hi there, FastLED developer here. Our tests are busted for esp32dev and esp32h2.

Both of these are failing only on the github runners but run fine locally on Windows and MacOS

ESP32H2
https://github.com/FastLED/FastLED/actions/runs/12438371142/job/34730155972

ESP32DEV:
https://github.com/FastLED/FastLED/actions/runs/12438371100

Output:

```
Using CPython 3.11.11 interpreter at: /opt/hostedtoolcache/Python/3.11.11/x64/bin/python3.11
Creating virtual environment at: .venv
Resolved 81 packages in 1.58s
Prepared 80 packages in 4.94s
Installed 81 packages in 364ms
 + ajsonrpc==1.2.0
 + anyio==4.7.0
 + appdirs==1.4.4
 + backports-tarfile==1.2.0
 + black==24.10.0
 + bottle==0.[13](https://github.com/FastLED/FastLED/actions/runs/12438371100/job/34730158292#step:5:14).2
 + certifi==2024.12.14
 + cffi==1.17.1
 + charset-normalizer==3.4.0
 + ci==0.1.0 (from file:///home/runner/work/FastLED/FastLED)
 + clang-format==19.1.5
 + click==8.1.7
 + cmake==3.31.2
 + colorama==0.4.6
 + cryptography==44.0.0
 + disklru==2.0.4
 + docker==7.1.0
 + download==0.3.5
 + fastled==1.2.3
 + fastled-wasm==1.0.16
 + filelock==3.16.1
 + fpvgcc==1.1.2
 + greenlet==3.1.1
 + h11==0.[14](https://github.com/FastLED/FastLED/actions/runs/12438371100/job/34730158292#step:5:15).0
 + httpcore==1.0.7
 + httpx==0.28.1
 + idna==3.10
 + importlib-metadata==8.5.0
 + iniconfig==2.0.0
 + isort==5.13.2
 + jaraco-classes==3.4.0
 + jaraco-context==6.0.1
 + jaraco-functools==4.1.0
Tool Manager: tool-mklittlefs@3.2.0 has been installed!
Tool Manager: Installing platformio/tool-scons @ ~4.40801.0
Downloading 0% 10% [20](https://github.com/FastLED/FastLED/actions/runs/12438371100/job/34730158292#step:5:21)% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: tool-scons@4.40801.0 has been installed!
Project has been successfully initialized!
*** Finished initializing environment for board esp32dev ***
*** Initializing environment for esp[32](https://github.com/FastLED/FastLED/actions/runs/12438371100/job/34730158292#step:5:33)dev ***

Running command:
  pio project init --project-dir .build/esp32dev --board esp32dev --project-option=platform=https://github.com/pioarduino/platform-espressif32.git#develop --project-option=build_flags=-Wl,-Map,firmware.map --project-option=build_flags=-fopt-info-all=optimization_report.txt --project-option=extra_scripts=/home/runner/work/FastLED/FastLED/pre:lib/ci/ci-flags.py
InvalidBoardManifest: Invalid board JSON manifest '/home/runner/.platformio/platforms/espressif32/boards/ws_esp32_s3_matrix.json'
*** Error setting up board esp32dev ***
Error initializing build_dir for board esp32dev:
InvalidBoardManifest: Invalid board JSON manifest '/home/runner/.platformio/platforms/espressif32/boards/ws_esp32_s3_matrix.json'
Compilation finished in 00m:[47](https://github.com/FastLED/FastLED/actions/runs/12438371100/job/34730158292#step:5:48)s.
Error: Process completed with exit code 1
```

zackees, 2024-12-21:
I had the json printout on our builder and found the issue:

from https://github.com/FastLED/FastLED/actions/runs/12440535169/job/34736196267

```json
{
  "build": {
    "core": "esp32",
    "f_cpu": "240000000L",
    "f_flash": "80000000L",
    "memory_type": "qio_qspi"   <--------- MISSING COMMA!
    "mcu": "esp32s3",
    "variant": "ws_esp32_s3_matrix"
  },
  "connectivity": [
    "bluetooth",
    "wifi"
  ],
  "debug": {
    "openocd_target": "esp32s3.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "Waveshare ESP32-S3-Matrix",
  "upload": {
    "flash_size": "4MB",
    "maximum_ram_size": 327680,
    "maximum_size": 4194304,
    "wait_for_upload_port": true,
    "require_upload_port": true,
    "speed": 460800
  },
  "url": "https://www.waveshare.com/wiki/ESP32-S3-Matrix",
  "vendor": "Waveshare"
}
```

ivankravets, 2024-12-21:
We can't find this board in the official integration https://github.com/platformio/platform-espressif32/tree/develop/boards

closed 2024-12-21 by ivankravets

zackees, 2024-12-23:
Can you please have platformio **not** break if one of the json files being (assumingly) injected by espressif has a missing comma?

This is like a try catch statement. Or can you point me to the code and I'll issue a PR.

Thanks,
~Zach

zackees, 2024-12-23:
Okay I got the issue resolved. I had one of the espressif dependencies pinned for arduino core for idf5.1. Nothing was changed there. But this problem started happening when the new idf core 5.3 was released earlier this week.

I went ahead and upgraded all our pinned dependencies to the new version and the build failure went away.

zackees, 2024-12-23:
But please, for the love god, don't let platformio break because some third party injects a malformed json file. That just seems like you are going to have more problems down the line.

ivankravets, 2024-12-23:
Could you share your `platformio.ini`?

zackees, 2024-12-23:
> Could you share your `platformio.ini`?

*working*
```
[env:esp32dev]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.10/platform-espressif32.zip
board = esp32dev
framework = arduino
build_flags = -fopt-info-all=optimization_report.txt
extra_scripts = C:\Users\niteris\dev\fastled\pre:lib\ci\ci-flags.py
```

*broken on github runner linux*
```
[env:esp32dev]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.04/platform-espressif32.zip
board = esp32dev
framework = arduino
build_flags = -fopt-info-all=optimization_report.txt
extra_scripts = C:\Users\niteris\dev\fastled\pre:lib\ci\ci-flags.py
```