platformio/platformio-core #5057: PIO tries to give the contents of #ifdef/#endif blocks to the compiler even if they should not be compiled.

hhrhhr, 2024-12-25:
### Configuration

**Operating system**: Microsoft Windows [Version 10.0.19045.5131]

**PlatformIO Version** PlatformIO Core, versions **v6.1.16** and **6.1.17b2**

### Description of problem

PIO tries to give the contents of #ifdef/#endif blocks to the compiler even if they should not be compiled.
The same file compiles without errors in Arduino IDE versions 1.8 and 2.3.

#### Steps to Reproduce

1. `pio run -j1 -v`

### Actual Results
```
Processing d1_mini (board: d1_mini; board_build.filesystem: littlefs; build_type: debug; framework: arduino; platform: espressif8266 @ 4.2.1; platform_packages: platformio/toolchain-xtensa @ 2.100300.220621, platformio/tool-esptool     @ 1.413.0, platformio/tool-esptoolpy   @ 1.30000.201119)
----------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 (4.2.1) > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 @ 3.30102.0 (3.1.2)
 - tool-esptool @ 1.413.0 (4.13)
 - tool-esptoolpy @ 1.30000.201119 (3.0.0)
 - toolchain-xtensa @ 2.100300.220621 (10.3.0)
xtensa-lx106-elf-g++ -o "I:\fail\fail.ino.cpp" -x c++ -fpreprocessed -dD -E "i:\tmp\hhr\tmphe3nida3"
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 37 compatible libraries
Scanning dependencies...
No dependencies
Building in debug mode
xtensa-lx106-elf-g++ -o .pio\build\d1_mini\src\fail.ino.cpp.o -c -fno-rtti -std=gnu++17 -fno-exceptions -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -ffunction-sections -fdata-sections -Wall -Werror=return-type -free -fipa-pta -Og -g2 -ggdb2 -DPLATFORMIO=60117 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_WEMOS_D1MINI -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -D_GNU_SOURCE -DARDUINO=10805 -DARDUINO_BOARD=\"PLATFORMIO_D1_MINI\" -DARDUINO_BOARD_ID=\"d1_mini\" -DFLASHMODE_DIO -DLWIP_OPEN_SRC -DNONOSDK22x_190703=1 -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DVTABLES_IN_FLASH -DMMU_IRAM_SIZE=0x8000 -DMMU_ICACHE_SIZE=0x8000 -D__PLATFORMIO_BUILD_DEBUG__ -I. -ID:\Users\hhr\.platformio\packages\framework-arduinoespressif8266\tools\sdk\include -ID:\Users\hhr\.platformio\packages\framework-arduinoespressif8266\cores\esp8266 -ID:\Users\hhr\.platformio\packages\toolchain-xtensa\include -ID:\Users\hhr\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lwip2\include -ID:\Users\hhr\.platformio\packages\framework-arduinoespressif8266\variants\d1_mini fail.ino.cpp
I:/fail/fail.ino:7:6: error: variable or field 'func2' declared void
    7 | #if (USE_LIB1 == 1)
      |      ^~~~~
I:/fail/fail.ino:7:12: error: 'lib2_t' was not declared in this scope
    7 | #if (USE_LIB1 == 1)
      |            ^~~~~~
I:/fail/fail.ino:7:20: error: 'p' was not declared in this scope
    7 | #if (USE_LIB1 == 1)
      |                    ^
I:/fail/fail.ino:8:1: error: 'lib3_t' does not name a type
    8 | void func1() {
      | ^~~~~~
*** [.pio\build\d1_mini\src\fail.ino.cpp.o] Error 1
```

### Expected Results

No errors.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[platformio]
src_dir = .
default_envs = d1_mini

[env]
framework = arduino
platform = espressif8266 @ 4.2.1
platform_packages = 
    platformio/toolchain-xtensa @ 2.100300.220621
    platformio/tool-esptool     @ 1.413.0
    platformio/tool-esptoolpy   @ 1.30000.201119

[env:d1_mini]
board = d1_mini
board_build.filesystem = littlefs
build_type = debug
```

**Source file to reproduce issue:**
```cpp
#define USE_LIB1 0
#define USE_LIB2 0
#define USE_LIB3 0
void setup() {}
void loop() {}

#if (USE_LIB1 == 1)
void func1() { // no errors from compiler, but linker is fail
    Serial.println("message from func1()");
}
#endif

#if (USE_LIB2 == 1)
void func2(lib2_t *p) { // error
    Serial.println("message from func2()");
}
#endif

#if (USE_LIB3 == 1)
lib3_t func3() { // error
    Serial.println("message from func3()");
}
#endif
```

### Additional info
If I move the `#ifdef` blocks BEFORE the _setup()_ and _loop()_ functions, then no compilation errors occur.
However, in a real project, such functions and libraries are defined in other .ino files, so I cannot control the parsing order.

vortigont, 2024-12-26:
`ino` files are Arduino IDE's creation, it only works there for their own reason. They event call is _sketches_ :) not a projects.
You can refactor your project to use standard `.h` and `.cpp` files with  and there won't be any problems neither in pio nor in Arduino IDE.

hhrhhr, 2024-12-26:
You are right, I meant specifically about projects with .ino files, I updated the issue.

However, the [documentation](https://docs.platformio.org/en/latest/faq/ino-to-cpp.html) says only that:
_Some Cloud & Desktop IDEs don’t support Arduino files (*.ino and .pde) because they are not valid C/C++ based source files..._
But nothing is said about the lack of support for such files in PlatformIO itself. Therefore, I consider this behavior (compile error) to be a bug.

ivankravets, 2024-12-26:
You are looking for `chain+` https://docs.platformio.org/en/latest/librarymanager/ldf.html#dependency-finder-mode

ivankravets, 2024-12-26:
Could you provide a simple PlatformIO-project to reproduce this issue? You can attach it there as a ZIP archive.

hhrhhr, 2024-12-26:
This archive contains the config and source from the first post.
[fail.zip](https://github.com/user-attachments/files/18254867/fail.zip)
Command - `pio run -v -j1` (-j1 so as not to wait long for the error to appear).

hhrhhr, 2024-12-26:
As I understand it, the problem is in the preprocessor of .ino files. Here is a diff between the output of arduino-cli and the source, which I managed to catch before it was deleted during compilation via pio:
```
--- fail_arduino.ino.cpp        2024-12-26 23:07:22.566316800 +0300
+++ fail_pio.ino.cpp    2024-12-26 23:07:19.686106700 +0300
@@ -2,6 +2,9 @@

 void setup();
 void loop();
+void func1();
+void func2(lib2_t *p);
+lib3_t func3();
 void setup() {}
 void loop() {}
```
As you can see, the pio preprocessor added declarations of unused functions, but the compiler does not know the types that use these functions.

ivankravets, 2024-12-27:
Can't reproduce the issue with this type of project. Please forward to https://community.platformio.org/

closed 2024-12-27 by ivankravets

hhrhhr, 2024-12-28:
> Can't reproduce...

v6.1.16 - reproduced, v6.1.17b2 - reproduced. What version did you use and couldn't reproduce the problem and where can I get it?

ivankravets, 2024-12-29:
The demo project misses `lib2_t` and other declarations. Please provide a complete project where we can reproduce this issue.

hhrhhr, 2024-12-29:
Well, that's the problem! These libraries and definitions should not be in this project.
If the project config specifically specifies `build_flags = -D USE_LIB2=0`, then I expect the compiler to ignore the `#if (USE_LIB2 == 1) /* */ #endif` block.
But the preprocessor inserts unused definitions without declaring them.

ivankravets, 2025-01-03:
I see. We don't plan to improve INO->CPP converter due to its limitations and distortion of C++. Please switch to the C++ https://docs.platformio.org/en/latest/faq/ino-to-cpp.html