platformio/platformio-core #5058: Windows Command size exceeds 32k causing failure to execute command

zackees, 2024-12-27:
What kind of issue is this?

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

Windows

**Version**

6.1.16 and also 6.1.17 latest

### Board

esp32 c6

### Description of problem

The generated command in the tmp file is 34k, exceeding the 32k limit

#### Steps to Reproduce

  * git clone https://github.com/fastled/fastled
  * checkout commit 5442a57c6603baac823bac885f2f10753bca2ff9
  * pio run

### Actual Results

```
Compiling .pio\build\dev\lib354\fastled\FastLED.cpp.o
Compiling .pio\build\dev\lib354\fastled\bitswap.cpp.o
Compiling .pio\build\dev\lib354\fastled\cled_controller.cpp.o
Compiling .pio\build\dev\lib354\fastled\colorpalettes.cpp.o
Compiling .pio\build\dev\lib354\fastled\colorutils.cpp.o
Compiling .pio\build\dev\lib354\fastled\crgb.cpp.o
riscv32-esp-elf-g++: fatal error: cannot execute 'C:/Users/niteris/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory
compilation terminated.
Compiling .pio\build\dev\lib354\fastled\five_bit_hd_gamma.cpp.o
*** [.pio\build\dev\lib354\fastled\bitswap.cpp.o] Error 1
riscv32-esp-elf-g++: fatal error: cannot execute 'C:/Users/niteris/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory
compilation terminated.
*** [.pio\build\dev\lib354\fastled\cled_controller.cpp.o] Error 1
riscv32-esp-elf-g++: fatal error: cannot execute 'C:/Users/niteris/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory
compilation terminated.
*** [.pio\build\dev\lib354\fastled\colorpalettes.cpp.o] Error 1
riscv32-esp-elf-g++: fatal error: cannot execute 'C:/Users/niteris/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory
compilation terminated.
*** [.pio\build\dev\lib354\fastled\FastLED.cpp.o] Error 1
riscv32-esp-elf-g++: fatal error: cannot execute 'C:/Users/niteris/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory
compilation terminated.
*** [.pio\build\dev\lib354\fastled\crgb.cpp.o] Error 1
riscv32-esp-elf-g++: fatal error: cannot execute 'C:/Users/niteris/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory
compilation terminated.
*** [.pio\build\dev\lib354\fastled\five_bit_hd_gamma.cpp.o] Error 1
riscv32-esp-elf-g++: fatal error: cannot execute 'C:/Users/niteris/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory
compilation terminated.
*** [.pio\build\dev\lib354\fastled\colorutils.cpp.o] Error 1
```

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini

[platformio]
src_dir = dev ; target is ./dev/dev.ino

[env:dev]
; This is the espressif32 platform which is the 4.1 toolchain as of 2024-Aug-23rd
; platform = espressif32
; The following platform enables the espressif32 platform to use the 5.1 toolchain, simulating
; the new Arduino 2.3.1+ toolchain.

# Developement branch of the open source espressif32 platform
platform =  https://github.com/pioarduino/platform-espressif32/releases/download/53.03.10/platform-espressif32.zip

framework = arduino
board = esp32-c6-devkitc-1

upload_protocol = esptool

monitor_filters = 
	default
	esp32_exception_decoder  ; Decode exceptions so that they are human readable.

; Symlink in the FastLED library so that changes to the library are reflected in the project
; build immediatly.
lib_deps = 
  FastLED=symlink://./

check_tool = clangtidy

```

### Additional info

According to the internet, the include paths can be shorted to be an environmental variable. Therefore, a fix would include using variable substitution for the top common prefix (Or just %HOME%, since everything is coming from the home directory) resulting in  a massive reduction in the generated GCC command.

hwacookie, 2025-01-12:
I have no solution, but at least I have the same problem :)
Could someone look into this?

zackees, 2025-01-13:
Honestly at this point, someone is going to have to volunteer to fox it and submit a PR.

tao4091, 2025-01-14:
`Compiling .pio\build\client_laser_guide\lib8b8\FastLED\third_party\espressif\led_strip\src\led_strip_spi_dev.c.o
Compiling .pio\build\client_laser_guide\lib8b8\FastLED\third_party\object_fled\src\OjectFLED.cpp.o
Compiling .pio\build\client_laser_guide\lib8b8\FastLED\transpose8x1_noinline.cpp.o
Compiling .pio\build\client_laser_guide\lib8b8\FastLED\wiring.cpp.o
Compiling .pio\build\client_laser_guide\src\client_laser_guide\button\button_adc.c.o
Compiling .pio\build\client_laser_guide\src\client_laser_guide\button\button_gpio.c.o
Compiling .pio\build\client_laser_guide\src\client_laser_guide\button\button_matrix.c.o
Compiling .pio\build\client_laser_guide\src\client_laser_guide\button\iot_button.c.o
Compiling .pio\build\client_laser_guide\src\client_laser_guide\charge.c.o
Compiling .pio\build\client_laser_guide\src\client_laser_guide\i2c_hal.c.o
riscv32-esp-elf-gcc: fatal error: cannot execute 'C:/Users/24084/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory
compilation terminated.
riscv32-esp-elf-gcc: fatal error: cannot execute 'C:/Users/24084/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory
compilation terminated.
riscv32-esp-elf-gcc: fatal error: cannot execute 'C:/Users/24084/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory
compilation terminated.
Compiling .pio\build\client_laser_guide\src\client_laser_guide\main.cpp.o
*** [.pio\build\client_laser_guide\src\client_laser_guide\button\button_gpio.c.o] Error 1`

I have the same question, but I can't figure out the solution

ivankravets, 2025-01-14:
@tao4091 , could you share your `platformio.ini`?

zackees, 2025-01-14:
This was posted in a cross post of this bug:

```python
import os

from platformio import fs

Import("env")

platform = env.PioPlatform()

FRAMEWORK_DIR = fs.to_unix_path(
    platform.get_package_dir("framework-arduinoespressif32")
)

IS_INTEGRATION_DUMP = env.IsIntegrationDump()

def shorten_includes(env, node):
    if IS_INTEGRATION_DUMP:
       # Don't shorten include paths for IDE integrations
       return node
       
    includes = [fs.to_unix_path(inc) for inc in env.get("CPPPATH", [])]
    shortened_includes = []
    generic_includes = []
    for inc in includes:
        if inc.startswith(FRAMEWORK_DIR):
            shortened_includes.append(
                "-iwithprefix/"
                + fs.to_unix_path(os.path.relpath(inc, FRAMEWORK_DIR))
            )
        else:
            generic_includes.append(inc)
 
    return env.Object(
        node,
        CPPPATH=generic_includes,
        CCFLAGS=env["CCFLAGS"] + ["-iprefix", FRAMEWORK_DIR] + shortened_includes
    )

env.AddBuildMiddleware(shorten_includes)
```

ivankravets, 2025-02-08:
Please forward to https://github.com/pioarduino/platform-espressif32/issues

closed 2025-02-08 by ivankravets