platformio/platformio-core #5060: Library from Registry won't install with username specified. Err:  Could not find the package with requirements for your system 'windows_amd64'

DjordjeMandic, 2025-01-01:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to the
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues) repository

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 10

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 6.1.16

### Description of problem

Custom library that I published won't install with username specified. 

#### Steps to Reproduce

1. Run `pio pkg install --library "djordjemandic/UIRBcorelib@^1.2.1"`
2. Run `pio pkg install --library "UIRBcorelib@^1.2.1"`

### Actual Results

```
pio pkg install --library "djordjemandic/UIRBcorelib@^1.2.1"                    
Resolving uirb-v02 dependencies...
Library Manager: Installing djordjemandic/UIRBcorelib @ ^1.2.1
UnknownPackageError: Could not find the package with 'djordjemandic/UIRBcorelib @ ^1.2.1' requirements for your system 'windows_amd64'

pio pkg install --library "UIRBcorelib@^1.2.1"              
Resolving uirb-v02 dependencies...
Library Manager: Installing UIRBcorelib @ ^1.2.1
Unpacking  [####################################]  100%
```

### Expected Results
Install library with `<username>/<libraryname>`

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

ivankravets, 2025-01-01:
`pio system prune`. Try again.

closed 2025-01-01 by ivankravets

DjordjeMandic, 2025-01-01:
Worked, thanks!