platformio/platformio-core #5064: Stuck at Installing platformio/toolchain-riscv

noobwei, 2025-01-14:
What kind of issue is this?

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

Ubuntu 24.10

**PlatformIO Version** (`platformio --version`):

PlatformIO Core, version 6.1.16  Home, version 3.4.4

### Description of problem

Stuck at `PlatformIO: Configuring project: Tool Manager: Installing platformio/toolchain-riscv @ ~1.80300.0`

#### Steps to Reproduce
Refer to the *LinuxFoundationX LFD119xComputer Architecture with an Industrial RISC-V Core [RVfpga]*

[image]

[image]

[image]

### Actual Results

Stuck at Installing platformio/toolchain-riscv, and configuring for hours.

[image]

### Expected Results

Takes only several minutes, then the project normally loaded

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter, extra scripting
;   Upload options: custom port, speed and extra flags
;   Library options: dependencies, extra library storages
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html

[env:swervolf_nexys]
platform = chipsalliance
board = swervolf_nexys
framework = wd-riscv-sdk

monitor_speed = 115200

#debug_tool = whisper

#RVfpga-Nexys
board_build.bitstream_file = /home/weiwei/RVfpga_v2/src/rvfpganexys.bit

#RVfpga-ViDBo
board_debug.verilator.binary = /home/weiwei/RVfpga_v2/Simulators/verilatorSIM_ViDBo/OriginalBinaries/RVfpga-ViDBo_Ubuntu22

#RVfpga-Pipeline
#board_debug.verilator.binary = /home/rvfpga/RVfpga/Simulators/verilatorSIM_Pipeline/OriginalBinaries/RVfpga-Pipeline_Ubuntu

#RVfpga-Trace
#board_debug.verilator.binary = /home/rvfpga/RVfpga/Simulators/verilatorSIM_Trace/OriginalBinaries/RVfpga-Trace_Ubuntu
```

ivankravets, 2025-01-14:
Do you use a proxy or any specific network configuration? Firewalls?

noobwei, 2025-01-15:
I am running all these on VM and there is no specific configuration on it

My physical host is in mainland China but my IP address shows in Singapore.  I believe it's an enterprise-level VPN. I don’t know if it will have any impact.

ivankravets, 2025-01-15:
Please open [PlatformIO CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli), navigate to the project folder and type `pio pkg install` command. Do you see any issues?

noobwei, 2025-01-15:
Thanks for your reply!

The output is:
```
Resolving swervolf_nexys dependencies...
Tool Manager: Installing platformio/toolchain-riscv @ ~1.80300.0
```

But there is no progress related information

ivankravets, 2025-01-15:
What is your OS? If Unix, please provide the output of the following command:
```
curl -v https://api.registry.platformio.org/v3/packages/platformio/tool/toolchain-riscv
```

noobwei, 2025-01-15:
About OS

```
cat /proc/version
Linux version 6.11.0-13-generic (buildd@lcy02-amd64-117) (x86_64-linux-gnu-gcc-14 (Ubuntu 14.2.0-4ubuntu2) 14.2.0, GNU ld (GNU Binutils for Ubuntu) 2.43.1) #14-Ubuntu SMP PREEMPT_DYNAMIC Sat Nov 30 23:51:51 UTC 2024
```

And the output is 

```
curl -v https://api.registry.platformio.org/v3/packages/platformio/tool/toolchain-riscv
* Host api.registry.platformio.org:443 was resolved.
* IPv6: (none)
* IPv4: 88.198.170.159
*   Trying 88.198.170.159:443...
* Connected to api.registry.platformio.org (88.198.170.159) port 443
* ALPN: curl offers h2,http/1.1
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
*  CAfile: /etc/ssl/certs/ca-certificates.crt
*  CApath: /etc/ssl/certs
* TLSv1.3 (IN), TLS handshake, Server hello (2):
* TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):
* TLSv1.3 (IN), TLS handshake, Certificate (11):
* TLSv1.3 (IN), TLS handshake, CERT verify (15):
* TLSv1.3 (IN), TLS handshake, Finished (20):
* TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1):
* TLSv1.3 (OUT), TLS handshake, Finished (20):
* SSL connection using TLSv1.3 / TLS_AES_256_GCM_SHA384 / x25519 / id-ecPublicKey
* ALPN: server accepted h2
* Server certificate:
*  subject: CN=*.platformio.org
*  start date: Nov 24 08:46:08 2024 GMT
*  expire date: Feb 22 08:46:07 2025 GMT
*  subjectAltName: host "api.registry.platformio.org" matched cert's "*.registry.platformio.org"
*  issuer: C=US; O=Let's Encrypt; CN=E5
*  SSL certificate verify ok.
*   Certificate level 0: Public key type EC/prime256v1 (256/128 Bits/secBits), signed using ecdsa-with-SHA384
*   Certificate level 1: Public key type EC/secp384r1 (384/192 Bits/secBits), signed using sha256WithRSAEncryption
*   Certificate level 2: Public key type RSA (4096/152 Bits/secBits), signed using sha256WithRSAEncryption
* using HTTP/2
* [HTTP/2] [1] OPENED stream for https://api.registry.platformio.org/v3/packages/platformio/tool/toolchain-riscv
* [HTTP/2] [1] [:method: GET]
* [HTTP/2] [1] [:scheme: https]
* [HTTP/2] [1] [:authority: api.registry.platformio.org]
* [HTTP/2] [1] [:path: /v3/packages/platformio/tool/toolchain-riscv]
* [HTTP/2] [1] [user-agent: curl/8.9.1]
* [HTTP/2] [1] [accept: */*]
> GET /v3/packages/platformio/tool/toolchain-riscv HTTP/2
> Host: api.registry.platformio.org
> User-Agent: curl/8.9.1
> Accept: */*
> 
* Request completely sent off
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
< HTTP/2 200 
< server: nginx
< date: Wed, 15 Jan 2025 09:18:39 GMT
< content-type: application/json
< content-length: 4880
< vary: Accept-Encoding
< vary: Accept-Encoding
< access-control-allow-origin: *
< access-control-expose-headers: ETag, Content-Type, Content-Length, Location
< 
{"id":8217,"type":"tool","tier":"community","owner":{"username":"platformio","firstname":"PlatformIO","lastname":null,"is_org":true},"name":"toolchain-riscv","description":"GNU toolchain for RISC-V, including GCC","added_at":"2019-03-07T21:34:04Z","updated_at":"2024-02-16T13:13:24Z","modified_at":"2024-02-15T20:59:17Z","downloads_trend":-39,"popularity_trend":3,"popularity_rank":89,"open_issues_count":30,"stars_count":2970,"examples_count":0,"title":null,"license":"GPL-2.0-or-later","homepage":null,"repository_url":"https://github.com/riscv/riscv-gnu-toolchain.git","keywords":["build tools","compiler","assembler","linker","preprocessor","risc-v"],"authors":[{"name":"RISC-V Collaboration","email":"info@riscv.org","url":"https://github.com/riscv-collab"}],"dependencies_count":0,"dependents_count":5,"top_dependencies":[],"top_dependents":[{"id":8032,"type":"platform","owner":{"username":"platformio"},"name":"sifive"},{"id":8036,"type":"platform","owner":{"username":"platformio"},"name":"chipsalliance"},{"id":12112,"type":"platform","owner":{"username":"bonfirecpu"},"name":"Bonfire"}],"version":{"id":59922,"name":"1.130200.2","unpacked_size":2386624751,"released_at":"2024-02-16T12:52:43Z","files":[{"name":"toolchain-riscv-windows_x86-1.130200.2.tar.gz","size":567933546,"checksum":{"sha256":"57b21376f3b3b6fabc59899d5c586e02c71fa9a66aca54f4bcebd8fe4b9d861e"},"system":["windows_amd64","windows_x86","windows_arm64"],"download_url":"https://dl.registry.platformio.org/download/platformio/tool/toolchain-riscv/1.130200.2/toolchain-riscv-windows_x86-1.130200.2.tar.gz"},{"name":"toolchain-riscv-linux_x86_64-1.130200.2.tar.gz","size":488385399,"checksum":{"sha256":"dca5153206bb27c26513d444e1ee0e5308a6b08cf14a39fcd758684e8d00eee4"},"system":["linux_x86_64"],"download_url":"https://dl.registry.platformio.org/download/platformio/tool/toolchain-riscv/1.130200.2/toolchain-riscv-linux_x86_64-1.130200.2.tar.gz"},{"name":"toolchain-riscv-darwin_x86_64-1.130200.2.tar.gz","size":481170615,"checksum":{"sha256":"fa2f1f90bde9efcac6afe5760a85975a24b9867b000d1ab0e1a90e7e20a74ee3"},"system":["darwin_x86_64","darwin_arm64"],"download_url":"https://dl.registry.platformio.org/download/platformio/tool/toolchain-riscv/1.130200.2/toolchain-riscv-darwin_x86_64-1.130200.2.tar.gz"}],"is_latest":true},"versions":[{"id":59922,"name":"1.130200.2","unpacked_size":2386624751,"released_at":"2024-02-16T12:52:43Z","files":[{"name":"toolchain-riscv-windows_x86-1.130200.2.tar.gz","size":567933546,"checksum":{"sha256":"57b21376f3b3b6fabc59899d5c586e02c71fa9a66aca54f4bcebd8fe4b9d861e"},"system":["windows_amd64","windows_x86","windows_arm64"],"download_url":"https://dl.registry.platformio.org/download/platformio/tool/toolchain-riscv/1.130200.2/to [... cut ...] .190927.tar.gz","size":191479773,"checksum":{"sha256":"2846aeddfc60bb60305cd97c1b45a73f054be463cbefe6b52b51f2a47f87ac97"},"system":["windows_amd64","windows_x86","windows_arm64"],"download_url":"https://dl.registry.platformio.org/download/platformio/tool/toolchain-riscv/1.80300.190927/toolchain-riscv-windows_x86-1.80300.190927.tar.gz"}],"is_latest":false}],"readme_url":"https://dl.registry.nm1.platformio.org/tools/ee/dc/d8c52066089431b229d73d6c28a6a02273bae4bc516de76cf427fee1fae1/README.md"}
```

ivankravets, 2025-01-15:
Please open PlatformIO Core CLI as mentioned above and type:
```
python3 -c "import requests; print(requests.get('https://platformio.org').text)"
```

noobwei, 2025-01-15:
```
<!doctype html><html lang="en" ng-app="siteApp" ng-controller="MainController as vm"><head><base href="/"><meta charset="utf-8"><title ng-bind="(viewTitle ? viewTitle : 'Your Gateway to Embedded Software Development Excellence') + ' Â· PlatformIO'">Your Gateway to Embedded Software Development Excellence Â· PlatformIO</title><meta name="author" content="PlatformIO"><meta name="robots" content="index, follow"><meta name="viewport" content="width=device-width,initial-scale=1"><meta name="google-site-verification" content="Zl6qDONgzdAFCj4uEbUNSuJdIKdk3MZYBcrc79fj0BY"><meta name="msvalidate.01" content="5CD2E0D7BC5235EDBE29F65F0DD0A180"><meta name="twitter:image:src" content="/images/platformio-ide-laptop.93fc8e69.png"><meta name="twitter:site" content="@PlatformIO_Org"><meta name="twitter:title" content="PlatformIO: Your Gateway to Embedded Software Development Excellence"><meta property="og:site_name" content="PlatformIO"><meta property="og:title" content="PlatformIO: Your Gateway to Embedded Software Development Excellence"><meta property="og:description" content="Unlock the true potential of embedded software development with PlatformIO's collaborative ecosystem, embracing declarative principles, test-driven methodologies, and modern toolchains for unrivaled success."><meta property="og:image" content="/images/platformio-ide-laptop.93fc8e69.png"><meta property="og:url" content="https://platformio.org"><link rel="shortcut icon" href="https://cdn.platformio.org/favicon.ico" type="image/x-icon"><link rel="apple-touch-icon" href="https://cdn.platformio.org/apple-touch-icon.png"><link href="https://maxcdn.bootstrapcdn.com/font-awesome/4.7.0/css/font-awesome.min.css" rel="stylesheet"><link rel="stylesheet" href="https://cdn.platformio.org/styles/vendor.f3749da9.css"><link rel="stylesheet" href="https://cdn.platformio.org/styles/main.84ce941a.css"></head><body><header><nav role="navigation" class="navbar navbar-inverse"><div class="container container-fluid"><div class="navbar-header"><button class="navbar-toggle" type="button" ng-click="vm.isNavBarCollapsed = !vm.isNavBarCollapsed"><span class="sr-only">Toggle navigation</span> <span class="icon-bar"></span> <span class="icon-bar"></span> <span class="icon-bar"></span></button> <a class="navbar-brand" href="/" title="PlatformIO"></a></div><div uib-collapse="vm.isNavBarCollapsed" class="collapse navbar-collapse"><ul class="nav navbar-nav"><li ng-class="{active: vm.isRouteActive('/platformio-ide')}" class="dropdown" uib-dropdown><a role="button" class="dropdown-toggle" uib-dropdown-toggle>Get Started<span class="caret"></span></a><ul class="dropdown-menu" role="menu"><li role="menuitem"><a href="https://docs.platformio.org/en/latest/what-is-platformio.html"><i class="fa fa-heart"></i>What is PlatformIO?</a></li><li role="separator" class="divider"></li><li role="menuitem"><a href="/platformio-ide"><i class="fa fa-cube"></i>PlatformIO IDE</a></li><li role="menuitem"><a href="/install/cli"><i class="fa fa-cogs"></i>PlatformIO Core (CLI)</a></li><li role="separator" class="divider"></li><li role="menuitem"><a href="https://docs.platformio.org/en/latest/tutorials/index.html" target="_blank"><i class="fa fa-graduation-cap"></i>Tutorials</a></li><li role="menuitem"><a href="https://github.com/platformio/platformio-examples" target="_blank"><i class="fa fa-code"></i>Project Examples</a></li></ul></li><li class="dropdown" uib-dropdown><a role="button" class="dropdown-toggle" uib-dropdown-toggle>Solutions<span class="caret"></span></a><ul class="dropdown-menu" role="menu"><li role="menuitem"><a href="/platformio-ide"><i class="fa fa-cube"></i>PlatformIO IDE</a></li><li role="menuitem"><a href="/install/cli"><i class="fa fa-cogs"></i>PlatformIO Core (CLI)</a></li><li role="separator" class="divider"></li><li role="menuitem"><a href="https://docs.platformio.org/en/latest/plus/debugging.html"><i class="fa fa-bug"></i>Debugging</a></li><li role="menuitem"><a href="https://docs.platformio.org/en/latest/plus/unit-testing.html"><i class="fa fa-puzzle-piece"></i>Unit Testing</a></li><li role="menuitem"><a href="https://docs.platformio.org/en/latest/plus/pio-check.html"><i class="fa fa-check-circle-o"></i>Static Code Analysis</a></li><li role="menuitem"><a href="https://docs.platformio.org/en/latest/plus/pio-remote.html"><i class="fa fa-globe"></i>Remote Development</a></li><li role="separator" class="divider"></li><li role="menuitem"><a href="https://docs.platformio.org/en/latest/librarymanager/index.html"><i class="fa fa-code"></i>Library Management</a></li><li role="separator" class="divider"></li><li role="menuitem"><a href="/install/integration"><i class="fa fa-cube"></i>Desktop IDEs Integration</a></li><li role="menuitem"><a href="/install/integration"><i class="fa fa-cloud"></i>Cloud IDEs Integration</a></li><li role="menuitem"><a href="https://docs.platformio.org/en/latest/ci/index.html"><i class="fa fa-refresh"></i>Continuous Integration</a></li></ul></li><li><a href="https://registry.platformio.org/">Registry</a></li></ul><ul class="nav navbar-nav navbar-right"><li><a href="https://docs.platformio.org">Docs</a></li><li><a href="https://community.platformio.org/">Community</a></li><li ng-class="{active: vm.isRouteActive('/support')}"><a href="/support">Support</a></li><li class="nav-pio-tech"><a href="https://piolabs.com/"><i class="fa fa-rocket fa-lg"></i> TECHNOLOGY</a></li></ul></div></div></nav></header><div ng-view="" autoscroll="vm.viewAutoScroll"><div c [... cut ...]
    <script src="https://code.jquery.com/jquery-1.10.2.min.js"></script>
  <![endif]--><!--[if lt IE 9]>
  <script src="scripts/oldieshim.d41d8cd9.js"></script>
  <![endif]--><script src="https://cdn.platformio.org/scripts/vendor.2c546285.js"></script><script src="https://cdn.platformio.org/scripts/scripts.5cb35eca.js"></script><script src="https://cdn.platformio.org/scripts/templates.a6371e15.js"></script><script async src="https://ad.platformio.org/delivery/revasyncjs.dll"></script></body></html>
```

ivankravets, 2025-01-15:
How are you in Python? Could you help us to debug the code? The source code of PlatformIO Core is located in `~/.platformio/penv/lib/python3.XXXXX/site-packages/platformio`

Try to uncomment [this line](https://github.com/platformio/platformio-core/blob/develop/platformio/http.py#L61) and see where it sticks.

noobwei, 2025-01-15:
ABSOLUTELY, but I'm not in front of the computer right now. I will send you the debug result tomorrow.🥰

noobwei, 2025-01-16:
After uncommenting `print("HTTPSession::request", self._x_base_url, method, url, args, kwargs)`

The output is

```
XXXXXX: $ pio pkg install
Resolving swervolf_nexys dependencies...
Tool Manager: Installing platformio/toolchain-riscv @ ~1.80300.0
HTTPSession::request https://api.registry.platformio.org GET /v3/packages/platformio/tool/toolchain-riscv () {'params': None, 'headers': {}, 'allow_redirects': True}
HTTPSession::request https://dl.registry.platformio.org HEAD /download/platformio/tool/toolchain-riscv/1.80300.190927/toolchain-riscv-linux_x86_64-1.80300.190927.tar.gz () {'allow_redirects': False, 'params': None, 'headers': {}}
```

noobwei, 2025-01-16:
And the other try is

```
Resolving swervolf_nexys dependencies...
Tool Manager: Installing platformio/toolchain-riscv @ ~1.80300.0
HTTPSession::request None GET https://sin1.contabostorage.com/fd833e6c4db04192b0ad7e561907b935:pioregmirrorsg/tools/b7/a0/68fe5d04ba31db73e37aac14c02e7a416b6ff18578e74b1e4e23634304fb/toolchain-riscv-linux_x86_64-1.80300.190927.tar.gz () {'stream': True, 'allow_redirects': True}
```

noobwei, 2025-01-16:
Seems the download speed of the toolchain is too slow

ivankravets, 2025-01-16:
Could you try to add to hosts:
```
127.0.0.1     sin1.contabostorage.com
```

You will be able to skip this mirror.

noobwei, 2025-01-17:
```
~/RVfpga_v2/Chapters/01/LedsSwitches_C-Lang$ pio pkg install
Resolving swervolf_nexys dependencies...
Tool Manager: Installing platformio/toolchain-riscv @ ~1.80300.0
HTTPSession::request https://api.registry.platformio.org GET /v3/packages/platformio/tool/toolchain-riscv () {'params': None, 'headers': {}, 'allow_redirects': True}
HTTPSession::request https://dl.registry.platformio.org HEAD /download/platformio/tool/toolchain-riscv/1.80300.190927/toolchain-riscv-linux_x86_64-1.80300.190927.tar.gz () {'allow_redirects': False, 'params': None, 'headers': {}}
HTTPSession::request None GET https://sin1.contabostorage.com/fd833e6c4db04192b0ad7e561907b935:pioregmirrorsg/tools/b7/a0/68fe5d04ba31db73e37aac14c02e7a416b6ff18578e74b1e4e23634304fb/toolchain-riscv-linux_x86_64-1.80300.190927.tar.gz () {'stream': True, 'allow_redirects': True}
HTTPSession::request None GET https://sin1.contabostorage.com/fd833e6c4db04192b0ad7e561907b935:pioregmirrorsg/tools/b7/a0/68fe5d04ba31db73e37aac14c02e7a416b6ff18578e74b1e4e23634304fb/toolchain-riscv-linux_x86_64-1.80300.190927.tar.gz () {'stream': True, 'allow_redirects': True}
Tool Manager: Error: Please read https://bit.ly/package-manager-ioerror
Tool Manager: Warning! Package Mirror: HTTPSConnectionPool(host='sin1.contabostorage.com', port=443): Max retries exceeded with url: /fd833e6c4db04192b0ad7e561907b935:pioregmirrorsg/tools/b7/a0/68fe5d04ba31db73e37aac14c02e7a416b6ff18578e74b1e4e23634304fb/toolchain-riscv-linux_x86_64-1.80300.190927.tar.gz (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x76e6c639dfd0>: Failed to establish a new connection: [Errno 111] Connection refused'))
Tool Manager: Looking for another mirror...
HTTPSession::request https://dl.registry.platformio.org HEAD /download/platformio/tool/toolchain-riscv/1.80300.190927/toolchain-riscv-linux_x86_64-1.80300.190927.tar.gz () {'allow_redirects': False, 'params': {'bypass': '687dea43b437e678e2366eb62cd259c4ccdd1511'}, 'headers': {}}
HTTPSession::request None GET https://github.com/ivankravets/pioreg-ci-mirror/releases/download/toolchain-riscv-linux_x86_64/b7a068f-toolchain-riscv-linux_x86_64-1.80300.190927.tar.gz () {'stream': True, 'allow_redirects': True}
Downloading  [------------------------------------]    1%  06:14:04
```

The good news is that there is progress, but it still takes a long time and the download speeds I tried on my current physical host are also very slow.

Is there any alternative download source, or do I need to configure it manually after downloading?

noobwei, 2025-01-17:
Thank you very much for your help, I have completed the environment configuration. 

But I still have some problems here, about `libwebsockets`, but they have nothing to do with this issue.

I will close this issue soon. Have a good day!

closed 2025-01-17 by noobwei

ivankravets, 2025-01-17:
This is the problem of your VPN provider. Try to disable VPN, we have the mirror in China. 

Please reopen the issue if it doesn’t work with disabled VPN.