platformio/platformio-core #5076: Log2File - override default log file name

FarbakyPal, 2025-01-30:
I would like to have an option for the logfile, to have fixed, user specified name. 
Everytime the serial port is opened, it creates a new log file with the specific name, or if the file allready exsist, it deletes the original content and starts from the empty log file.

ivankravets, 2025-02-08:
See https://docs.platformio.org/en/latest/core/userguide/device/cmd_monitor.html#custom-filters

You can modify the original filter to fit your needs https://github.com/platformio/platformio-core/blob/develop/platformio/device/monitor/filters/log2file.py

closed 2025-02-08 by ivankravets

FarbakyPal, 2025-02-11:
Thank your for your response!
In the end I managed to make my own filter work.
The demo filter returned "Warning! Skipping unknown filters"... error for a long time.
For me it wasn't intuitive to use the variable NAME in the .ini "monitor_filter = ", so I used the .py file name.
In hindsight it is reasonable.
If I can make a suggestion for the documentation, in the custon filter section, next to the filter_demo.py code, I would also include a snippet like "monitor_filter = demo" to make it more intuitive for less experienced users.

ivankravets, 2025-02-11:
Could you provide the source code of your monitor here? So, the others can use it.

FarbakyPal, 2025-02-11:
Sure, maybe it proves usefull to others. Here is the .py code:

```python

import io
import os
from datetime import datetime

from platformio.public import DeviceMonitorFilterBase

class filter_own(DeviceMonitorFilterBase):
    NAME = "filter_own"

    def __init__(self, *args, **kwargs):
        super().__init__(*args, **kwargs)
        self._bufferOut = ""
        self._bufferIn = ""
        if self.options.get("eol") == "CR":
            self._eol = "\r"
        elif self.options.get("eol") == "LF":
            self._eol = "\n"
        else:
            self._eol = "\r\n"
        

    def __del__(self):
        if self._log_fp:
            self._log_fp.close()

    def rx(self, text):
        self._bufferIn += text
        if self._bufferIn.endswith(self._eol):
            timestamp = datetime.now().strftime("%H:%M:%S.%f")[:-3]
            text = "Recieved at %s -- %s" % (timestamp, self._bufferIn)
            self._bufferIn = ""
            self._log_fp.write(text)
            self._log_fp.flush()
            return text
        return ""

    def tx(self, text):
        self._bufferOut += text
        if self._bufferOut.endswith(self._eol):
            timestamp = datetime.now().strftime("%H:%M:%S.%f")[:-3]
            logText = "Sent     at %s -- %s" % (timestamp, self._bufferOut)
            text=self._bufferOut
            print(logText)
            self._bufferOut = ""
            self._log_fp.write(logText)
            self._log_fp.flush()
            return text
        return ""

    def __call__(self):
        if not os.path.isdir("logs"):
            os.makedirs("logs")
        log_file_name = os.path.join(
            "logs", "serial-monitor.log"
        )
        print("--- Logging an output to %s" % os.path.abspath(log_file_name))
        self._log_fp = io.open(log_file_name, "w", encoding="utf-8")
        self._log_fp.write("--- Log file date: %s ---\r\n" % datetime.now().strftime("20%y. %m. %d. -- %H : %M"))
        self._log_fp.flush()
        return self
``` 

In The paltformio.ini file you need to add:
```
monitor_filters =  filter_own
```

This filter saves the serial communication, both the data received and sent by the computer into a log file called "serial-monitor.log". It also adds a timestamp to every message for extra information.