platformio/platformio-core #5090: PIO Core 6.1.17 brakes `compile_commands.json` from `compiledb`

Nathan-ma, 2025-02-16:
Hi folks, no big deal, reporting an issue with 6.1.17.
After upgrading to 6.1.17, the clangd stopped working as expected; I just downgraded the core version back to `6.1.16` and everything is now working again.

OS: macOS
Basic .ini config:
```
[device]
platform = espressif32@6.4.0
board = esp32dev
framework = arduino
``` 
`.clangd`: 
```
CompileFlags:
  Remove: [-fno-tree-switch-conversion, -mtext-section-literals, -mlongcalls, -fstrict-volatile-bitfields]
```

My LSP when on 6.1.17, fails to find any Arduino framework lib like: WiFi, EEPROM, WiFiProv, etc.

Downgrading to 6.1.16 solves it right away:
```
pip install platformio==6.1.16
```

This is a great opportunity to start supporting version selection with the brew package manager.

ivankravets, 2025-03-11:
Sorry, I can't reproduce the issue. Could you provide a simple PlatformIO project?

MacMannes, 2025-03-30:
@ivankravets I have the same problems. I'm using a Mac with NeoVim and the clangd LSP. Here's my project: https://github.com/MacMannes/homey-sonoff-tx

When I use PIO 6.1.18 I get these kind of errors:

[image]

When I downgrade to PIO **6.1.16** and run `pio run -t compiledb` and open the `main.cpp` file of my project, the LSP errors dissappear. The `compile_commands.json` that are generated between versions **6.1.16** and **6.1.18** are different.

For **6.1.8**, these library paths are missing in the command for `main.cpp`:

-I/Users/andre/.platformio/packages/framework-arduinoespressif8266/libraries/Ethernet/src
-I/Users/andre/.platformio/packages/framework-arduinoespressif8266/libraries/SPI
-I/Users/andre/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src

proxict, 2025-05-04:
I believe this issue is related to this: https://github.com/platformio/platformio-core/issues/4998

commit d787648e (2025-05-07): Fixed a regression issue where custom build flags were not properly reflected in the `compile_commands.json` // Resolve #5090 Resolve #5147
closed 2025-05-07 by ivankravets