platformio/platformio-core #5091: PIO Core 6.1.17 brakes unit testing with googletest

TeamPlatinum1, 2025-02-16:
Hi everyone, first I would like to thank you for your work on this project.

After the upgrade to PIO Core 6.1.17 the testing fails in the build stage and throws this error:

```
.pio/libdeps/native/googletest/googletest/src/gtest-filepath.cc: In static member function ‘static testing::internal::FilePath testing::internal::FilePath::GetCurrentDir()’:
.pio/libdeps/native/googletest/googletest/src/gtest-filepath.cc:60:25: error: ‘_POSIX_PATH_MAX’ was not declared in this scope; did you mean ‘_PC_PATH_MAX’?
   60 | #define GTEST_PATH_MAX_ _POSIX_PATH_MAX
      |                         ^~~~~~~~~~~~~~~
.pio/libdeps/native/googletest/googletest/src/gtest-filepath.cc:114:12: note: in expansion of macro ‘GTEST_PATH_MAX_’
  114 |   char cwd[GTEST_PATH_MAX_ + 1] = {'\0'};
      |            ^~~~~~~~~~~~~~~
.pio/libdeps/native/googletest/googletest/src/gtest-filepath.cc:115:25: error: ‘cwd’ was not declared in this scope
  115 |   char* result = getcwd(cwd, sizeof(cwd));
      |                         ^~~
*** [.pio/build/native/lib109/googletest/googletest/src/gtest-filepath.o] Error 1
Building stage has failed, see errors above. Use `pio test -vvv` option to enable verbose output.
--------------- native:test_native [ERRORED] Took 36.92 seconds ---------------
```

The .ini file looks like this:

```
[env:teensy40]
platform = teensy
board = teensy40
framework = arduino
build_flags = -Ofast
test_ignore = test_native
lib_extra_dirs = lib/custom, lib/public, lib/inbuilt, lib/edited
monitor_speed = 115200

[env:native]
platform = native
test_framework = googletest
build_flags = -D NATIVE
lib_compat_mode = off
lib_extra_dirs = lib/custom, lib/public
lib_deps = google/googletest@1.15.2
```

After downgrading to 6.1.16 everything works as expected. Let me know if I should provide more info.

ivankravets, 2025-03-03:
What is your OS?

ivankravets, 2025-03-10:
Could you try this example https://github.com/platformio/platformio-examples/tree/develop/unit-testing/googletest ? Does this work for you?

TeamPlatinum1, 2025-03-22:
Sorry for the late response.
Interestingly, the example provided by you worked on my machine. I tried several things to get it as similar as possible, like changing the platformio.ini, but it worked.

So I tried adding some includes to the file gtest-filepath.cc and after adding this:

```
#ifdef GTEST_OS_WINDOWS_MOBILE
#include <windows.h>
#elif defined(GTEST_OS_WINDOWS)
#include <direct.h>
#include <io.h>
#else
#include <limits.h>

#include <linux/limits.h> // Added by me!

#include <climits>  // Some Linux distributions define PATH_MAX here.
#endif              // GTEST_OS_WINDOWS_MOBILE
```
Everything worked as expected. It confuses me why your example didn't need this include.
I also found this: https://stackoverflow.com/questions/73168335/difference-between-include-limits-h-and-inlcude-linux-limits-h
So I think the issue here lies in the googletest includes.

ivankravets, 2025-03-25:
If you have no objections, I will close this issue since it is not related to PlatformIO Core.

closed 2025-03-25 by ivankravets