platformio/platformio-core #5103: It seems that M5Cardputer.BtnA becomes null once any function from M5Cardputer.Power gets called

kylxbn, 2025-03-11:
I tried my best but couldn't find any reason why that would be. Does anyone know how to avoid this problem?

Even without running an `M5Cardputer.Power`-related function, just the presence of that line of code in the file turns `M5Cardputer.BtnA` into `null` (invalid pointer).

Getting the memory address of `M5Cardputer.Power` is not enough to trigger the issue. A function from `Power` has to be called (like `getBatteryCurrent` or `getBatteryLevel`)

Some functions (like `M5Cardputer.Power.getType()`) does not cause any issue.

Attached below is the simplest possible proof of concept. This will crash even before printing `PRESSED` or `NOT PRESSED`. That's because `M5Cardputer.BtnA` is a null / invalid pointer. However, removing the line with `getBatteryLevel` will not cause `BtnA` to turn into a `null` pointer (yes, even though it's after the line that uses `BtnA`). Really weird.

It seems that this only happens on PlatformIO, and not on Arduino IDE. That's even more weird.

```
#include "M5Cardputer.h"

void setup() {
    Serial.begin(115200U);
    auto cfg = M5.config();
    M5Cardputer.begin(cfg);
}

void loop() {
    M5Cardputer.update();    

    bool pressed = M5Cardputer.BtnA.wasPressed();
    Serial.println(pressed ? "PRESSED" : "NOT PRESSED");

    int bat = M5Cardputer.Power.getBatteryLevel();
    Serial.println(bat);

    delay(1000);
}
```

ivankravets, 2025-03-11:
Please forward to https://community.platformio.org/

closed 2025-03-11 by ivankravets

kylxbn, 2025-03-12:
My apologies. Thank you for pointing me to the right place.