platformio/platformio-core #5128: Building a library without linking attempt

matthiashanns, 2025-04-04:
### Problem

Library projects contain their sources in `src/` and `include/` and **don't have a `main()` function**. Building such a project first compiles the code and **subsequently tries to link it unnecessarily**. The linker, however, fails by complaining about a missing `main()` or missing `setup()` and `loop()` (depending on whether, e.g., the arduino framework is in use).

There is no project parameter in `platformio.ini` to tell pio not to link a library project. There is also no build parameter to the compiler (eg. gcc) to suppress the linker since the linker is started exclusively by pio. A custom build script could be used, but it should not be required for a common task like building a library project.

### Expected behavior

Building a library project should contain the steps of compiling, optionally creating a library archive (.a), and optionally including an archive index, **but no linking step**.

- A project should be markable as a library project in `platformio.ini` (eg. `type = library` with `type` defaulting to `application`) to omit the linking step and avoid a linking error and hence a build error.
- Additionally, a *compile-only* build option should exist that compiles without linking to quickly build a library project or check the code for compile errors on any project, saving the linker time.

Jason2866, 2025-07-30:
+1

furthermore the in deep hard coded gcc structure should be made configurable to make it possible to use an other compile / linker than gcc. It is nearly impossible to use clang / llvm with Platformio

ivankravets, 2025-07-30:
Have you tried https://docs.platformio.org/manifests/library-json/fields/build/libarchive.html ?

matthiashanns, 2025-07-30:
@ivankravets Thanks for the link, Ivan. Unfortunately, the page doesn't exist (404).

ivankravets, 2025-07-30:
Sorry, the link has been fixed.