platformio/platformio-core #5134: Extremely slow download of platformio components

trizmark, 2025-04-13:
PlatformIO is taking a really-really long time to download the various components.

```
platformio --version
PlatformIO Core, version 6.1.16
``` 

platformio/toolchain-xtensa @ ~2.100300.220621 took about 16 minutes to download. Using https://dl.registry.platformio.org/download/platformio/tool/toolchain-xtensa/2.100300.220621/toolchain-xtensa-windows_amd64-2.100300.220621.tar.gz I can see that the donwload speed is about 60kbps.

I'm in the UK on a 900 Mbps connection. Speedtest.net returned 890 Mbps down and 750 Mbps up.

ivankravets, 2025-04-13:
Are you still experiencing this issue?

trizmark, 2025-04-13:
Still the same
```
INFO Installing platformio/framework-arduinoespressif8266 @ ~3.30102.0
Downloading  [####--------------------------------]   12%  00:06:37
```

With a VPN to France it's a bit better
```
INFO Installing platformio/framework-arduinoespressif8266 @ ~3.30102.0
Downloading  [#####-------------------------------]   16%  00:00:55
```

trizmark, 2025-04-13:
VPN to US East results in 60-70kbps download speed as well.

ivankravets, 2025-04-13:
I've just created the small Amazon AWS instance in the GB region and can't reproduce the issue. See the screenshot

[image]

ivankravets, 2025-04-13:
Could you try to disable the VPN? Are you sure your IP is in the GB zone? https://www.maxmind.com/en/geoip-demo

trizmark, 2025-04-13:
Definitely UK address

[image]

```
pi@rpi-prod:~/work/tmp $ wget https://dl.registry.platformio.org/download/platformio/tool/toolchain-xtensa/2.100300.220621/toolchain-xtensa-windows_amd64-2.100300.220621.tar.gz
--2025-04-13 21:09:35--  https://dl.registry.platformio.org/download/platformio/tool/toolchain-xtensa/2.100300.220621/toolchain-xtensa-windows_amd64-2.100300.220621.tar.gz
Resolving dl.registry.platformio.org (dl.registry.platformio.org)... 88.198.170.159
Connecting to dl.registry.platformio.org (dl.registry.platformio.org)|88.198.170.159|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://usc1.contabostorage.com/250040e607e8431abaff8727621872f9:pioregmirrorus/tools/2e/c0/8ad5caaacf8cb92ae04e8bcebe697857766b729fd0f88ee5c4bb97c0c6a9/toolchain-xtensa-windows_amd64-2.100300.220621.tar.gz [following]
--2025-04-13 21:09:35--  https://usc1.contabostorage.com/250040e607e8431abaff8727621872f9:pioregmirrorus/tools/2e/c0/8ad5caaacf8cb92ae04e8bcebe697857766b729fd0f88ee5c4bb97c0c6a9/toolchain-xtensa-windows_amd64-2.100300.220621.tar.gz
Resolving usc1.contabostorage.com (usc1.contabostorage.com)... 209.126.15.85, 209.126.15.84
Connecting to usc1.contabostorage.com (usc1.contabostorage.com)|209.126.15.85|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 74043018 (71M) [binary/octet-stream]
Saving to: ‘toolchain-xtensa-windows_amd64-2.100300.220621.tar.gz’

toolchain-xtensa-windows_amd64-2.100300.220621.t   1%[>                                                                                                       ]   1.11M  79.7KB/s    eta 15m 5s
```

ivankravets, 2025-04-21:
Do you still experience the issues?

trizmark, 2025-04-22:
Looks like things are back to normal. Not sure what it was. 🤷 
Thanks for looking into it!

```
(venv) pi@rpi-prod:~/work $ wget https://dl.registry.platformio.org/download/platformio/tool/toolchain-xtensa/2.100300.220621/toolchain-xtensa-windows_amd64-2.100300.220621.tar.gz
--2025-04-22 09:25:51--  https://dl.registry.platformio.org/download/platformio/tool/toolchain-xtensa/2.100300.220621/toolchain-xtensa-windows_amd64-2.100300.220621.tar.gz
Resolving dl.registry.platformio.org (dl.registry.platformio.org)... 88.198.170.159
Connecting to dl.registry.platformio.org (dl.registry.platformio.org)|88.198.170.159|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://usc1.contabostorage.com/250040e607e8431abaff8727621872f9:pioregmirrorus/tools/2e/c0/8ad5caaacf8cb92ae04e8bcebe697857766b729fd0f88ee5c4bb97c0c6a9/toolchain-xtensa-windows_amd64-2.100300.220621.tar.gz [following]
--2025-04-22 09:25:51--  https://usc1.contabostorage.com/250040e607e8431abaff8727621872f9:pioregmirrorus/tools/2e/c0/8ad5caaacf8cb92ae04e8bcebe697857766b729fd0f88ee5c4bb97c0c6a9/toolchain-xtensa-windows_amd64-2.100300.220621.tar.gz
Resolving usc1.contabostorage.com (usc1.contabostorage.com)... 209.126.15.84, 209.126.15.85
Connecting to usc1.contabostorage.com (usc1.contabostorage.com)|209.126.15.84|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 74043018 (71M) [binary/octet-stream]
Saving to: ‘toolchain-xtensa-windows_amd64-2.100300.220621.tar.gz’

toolchain-xtensa-windows_amd64-2.100300.220621.tar.gz         100%[==============================================================================================================================================>]  70.61M  27.2MB/s    in 2.6s    

2025-04-22 09:25:54 (27.2 MB/s) - ‘toolchain-xtensa-windows_amd64-2.100300.220621.tar.gz’ saved [74043018/74043018]
```

closed 2025-04-22 by trizmark