platformio/platformio-core #5135: TypeError: Tried to lookup RootDir '\\' as a File when importing project

ForrestFire0, 2025-04-14:
Using pio on clion. Latest version (did pio upgrade).

Tried deleting the contents of my .packages folder but that did not help. Seems to be some error deep in pio code which is beyond me.

```
[env:esp32cam]
; Arduino Release v3.0.0 ALPHA based on ESP-IDF v5, caused issues
;platform = https://github.com/Jason2866/platform-espressif32.git#Arduino/IDF5
; Arduino Release v2.0.14 based on ESP-IDF v4.4.6
platform = platformio/espressif32@6.10.0
board = esp32cam
framework = arduino

upload_speed = 921600
monitor_speed = 115200
; Needed for working serial
monitor_rts = 0
monitor_dtr = 0
; Nice to have...
monitor_filters = esp32_exception_decoder
;Library versions now frozen.

lib_deps =
    esp32async/AsyncTCP@3.3.8
    esp32async/ESPAsyncWebServer@3.7.6
    yoursunny/esp32cam@0.0.20240110
    bblanchon/ArduinoJson@7.0.3
    gilmaimon/ArduinoWebsockets@0.5.3

board_build.extra_flags =
    -D CONFIG_ASYNC_TCP_MAX_ACK_TIME=5000 // (keep default)
    -D CONFIG_ASYNC_TCP_PRIORITY=10 // (keep default)
    -D CONFIG_ASYNC_TCP_QUEUE_SIZE=64 // (keep default)
    -D CONFIG_ASYNC_TCP_RUNNING_CORE=1 // force async_tcp task to be on same core as the app (default is core 0)
    -D CONFIG_ASYNC_TCP_STACK_SIZE=4096 // reduce the stack size (default is 16K)
```

My error:
```
C:\Users\forre\.platformio\penv\Scripts\pio project config --json-output
C:\Users\forre\.platformio\penv\Scripts\pio project metadata --json-output -e esp32cam --json-output-path C:\Users\forre\AppData\Local\Temp\pio.json
UserSideException: Processing esp32cam (platform: platformio/espressif32@6.10.0; board: esp32cam; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32cam.html
PLATFORM: Espressif 32 (6.10.0) > AI Thinker ESP32-CAM
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20017.241212+sha.dcc1105b 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
TypeError: Tried to lookup RootDir '\\' as a File.:
  File "C:\Users\forre\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\forre\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Script\SConscript.py", line 620:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\forre\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\forre\.platformio\platforms\espressif32\builder\main.py", line 368:
    target_elf = env.BuildProgram()
  File "C:\Users\forre\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Util\envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "C:\Users\forre\.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 62:
    env.ProcessProgramDeps()
  File "C:\Users\forre\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Util\envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "C:\Users\forre\.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 136:
    env.ProcessFlags(env.BoardConfig().get("build.extra_flags"))
  File "C:\Users\forre\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Util\envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "C:\Users\forre\.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 238:
    env.Append(**env.ParseFlagsExtended(flags))
  File "C:\Users\forre\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Util\envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "C:\Users\forre\.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 195:
    for key, value in env.ParseFlags(str(raw)).items():
  File "C:\Users\forre\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Environment.py", line 1034:
    do_parse(arg)
  File "C:\Users\forre\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Environment.py", line 950:
    mapping['LIBS'].append(self.fs.File(arg))
  File "C:\Users\forre\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Node\FS.py", line 1458:
    return self._lookup(name, directory, File, create)
  File "C:\Users\forre\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Node\FS.py", line 1437:
    return root._lookup_abs(p, fsclass, create)
  File "C:\Users\forre\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Node\FS.py", line 2480:
    result.must_be_same(klass)
  File "C:\Users\forre\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Node\FS.py", line 2439:
    Base.must_be_same(self, klass)
  File "C:\Users\forre\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Node\FS.py", line 644:
    raise TypeError("Tried to lookup %s '%s' as a %s." %\
========================== [FAILED] Took 0.56 seconds ==========================
```

ForrestFire0, 2025-04-14:
New evidence - commenting out the build_board.extra_flags makes it continue on to compilation!!!

Succeeding:
```
[env:esp32cam]
; Arduino Release v3.0.0 ALPHA based on ESP-IDF v5, caused issues
;platform = https://github.com/Jason2866/platform-espressif32.git#Arduino/IDF5
; Arduino Release v2.0.14 based on ESP-IDF v4.4.6
platform = platformio/espressif32@6.10.0
board = esp32cam
framework = arduino

upload_speed = 921600
monitor_speed = 115200
; Needed for working serial
monitor_rts = 0
monitor_dtr = 0
; Nice to have...
monitor_filters = esp32_exception_decoder
;Library versions now frozen.

lib_deps =
    esp32async/AsyncTCP@3.3.8
    esp32async/ESPAsyncWebServer@3.7.6
    yoursunny/esp32cam@0.0.20240110
    bblanchon/ArduinoJson@7.0.3

board_build.extra_flags =
;    -D CONFIG_ASYNC_TCP_MAX_ACK_TIME=5000 // (keep default)
;    -D CONFIG_ASYNC_TCP_PRIORITY=10 // (keep default)
;    -D CONFIG_ASYNC_TCP_QUEUE_SIZE=64 // (keep default)
;    -D CONFIG_ASYNC_TCP_RUNNING_CORE=1 // force async_tcp task to be on same core as the app (default is core 0)
;    -D CONFIG_ASYNC_TCP_STACK_SIZE=4096 // reduce the stack size (default is 16K)
```

Failing:
```
[env:esp32cam]
; Arduino Release v3.0.0 ALPHA based on ESP-IDF v5, caused issues
;platform = https://github.com/Jason2866/platform-espressif32.git#Arduino/IDF5
; Arduino Release v2.0.14 based on ESP-IDF v4.4.6
platform = platformio/espressif32@6.10.0
board = esp32cam
framework = arduino

upload_speed = 921600
monitor_speed = 115200
; Needed for working serial
monitor_rts = 0
monitor_dtr = 0
; Nice to have...
monitor_filters = esp32_exception_decoder
;Library versions now frozen.

lib_deps =
    esp32async/AsyncTCP@3.3.8
    esp32async/ESPAsyncWebServer@3.7.6
    yoursunny/esp32cam@0.0.20240110
    bblanchon/ArduinoJson@7.0.3

board_build.extra_flags =
    -D CONFIG_ASYNC_TCP_MAX_ACK_TIME=5000 // (keep default)
;    -D CONFIG_ASYNC_TCP_PRIORITY=10 // (keep default)
;    -D CONFIG_ASYNC_TCP_QUEUE_SIZE=64 // (keep default)
;    -D CONFIG_ASYNC_TCP_RUNNING_CORE=1 // force async_tcp task to be on same core as the app (default is core 0)
;    -D CONFIG_ASYNC_TCP_STACK_SIZE=4096 // reduce the stack size (default is 16K)

```

ForrestFire0, 2025-04-14:
Alright... figured out that the comments after the build flags were breaking that section of the platformio.ini file.

Thats on me!

Working config:
```
[env:esp32cam]
; Arduino Release v3.0.0 ALPHA based on ESP-IDF v5, caused issues
;platform = https://github.com/Jason2866/platform-espressif32.git#Arduino/IDF5
; Arduino Release v2.0.14 based on ESP-IDF v4.4.6
platform = platformio/espressif32@6.10.0
board = esp32cam
framework = arduino

build_flags =
    -D CONFIG_ASYNC_TCP_MAX_ACK_TIME=5000
    -D CONFIG_ASYNC_TCP_PRIORITY=10
    -D CONFIG_ASYNC_TCP_QUEUE_SIZE=64
    -D CONFIG_ASYNC_TCP_RUNNING_CORE=1
    -D CONFIG_ASYNC_TCP_STACK_SIZE=4096

upload_speed = 921600
monitor_speed = 115200
; Needed for working serial
monitor_rts = 0
monitor_dtr = 0
; Nice to have...
monitor_filters = esp32_exception_decoder
;Library versions now frozen.

lib_deps =
    esp32async/AsyncTCP@3.3.8
    esp32async/ESPAsyncWebServer@3.7.6
    yoursunny/esp32cam@0.0.20240110
    bblanchon/ArduinoJson@7.0.3
```

Jason2866, 2025-04-21:
`//` does not indicate a comment in Platformio.

closed 2025-04-28 by ivankravets