platformio/platformio-core #5155: build defines suddenly not making it into build

zackees, 2025-05-08:
This is the ultimate command being run on our machine.

Do you have any idea what it suddenly stopped defining -DFASTLED_FORCE_NAMESPACE=1 in the build environment?

https://github.com/FastLED/FastLED/actions/runs/14908023440/job/41875013756

`pio project init --project-dir .build/esp32dev --board esp32dev --project-option=platform=https://github.com/pioarduino/platform-espressif32/releases/download/53.03.10/platform-espressif32.zip --project-option=build_flags=-Wl,-Map,firmware.map --project-option=build_flags=-fopt-info-all=optimization_report.txt "--project-option=build_flags=-DFASTLED_FORCE_NAMESPACE=1 -DFASTLED_FORCE_USE_NAMESPACE=1" --project-option=extra_scripts=/home/runner/work/FastLED/FastLED/pre:lib/ci/ci-flags.py`

ivankravets, 2025-05-09:
You have to quote values of the CLI options that contain the bash symbols. 

`--option=“value with = symbol”`

closed 2025-05-09 by ivankravets

zackees, 2025-05-10:
looks like it was a github issue on the runners.

It auto resolved itself.

Thanks for looking into it.