platformio/platformio-core #5240: Add flag to disable self-upgrades

vidplace7, 2025-08-29:
PlatformIO packaging efforts downstream in Fedora, Debian, Arch, OpenWrt, and probably others must currently patch `platformio-core` to remove the `platformio upgrade` command to ensure packaging is not broken. This results in additional maintainer overhead.

Example patch from Fedora: [platformio-short-circuit-upgrades.patch](https://src.fedoraproject.org/rpms/python-platformio/blob/rawhide/f/platformio-short-circuit-upgrades.patch)
CC: @musicinmybrain 

As the maintainer of the [OpenWrt](https://github.com/openwrt/packages/tree/master/lang/python/python-platformio) PlatformIO package, and adoptive-maintainer of the [Debian](https://salsa.debian.org/python-team/packages/platformio/-/merge_requests/4) packaging. This would be greatly appreciated.

Request: Provide build-time flag to disable `platformio upgrade` command

musicinmybrain, 2025-08-30:
> CC: [@musicinmybrain](https://github.com/musicinmybrain)

I’m only involved in Fedora’s [`python-platformio` package](https://src.fedoraproject.org/rpms/python-platformio) because I have to patch its strict dependency bounds every time I update [`python-starlette`](https://src.fedoraproject.org/rpms/python-starlette). I have no real interest in the package *per se*. The “real” maintainer of the Fedora package is @davide125.

ivankravets, 2026-09-05:
Resolved in the upcoming PIO Core 6.2.0. See https://docs.platformio.org/en/latest/envvars.html#envvar-PLATFORMIO_DISABLE_UPGRADE_CHECK

closed 2026-09-05 by ivankravets