platformio/platformio-core #5261: Library Name Collision: External Registry Libraries Override Built-in Framework Libraries

sylque, 2025-09-20:
## Problem

When using `lib_ldf_mode = off` and explicitly declaring dependencies, PlatformIO resolves library names to external registry libraries instead of built-in framework libraries, even when the intent is to use the framework's native implementation.

## Example

User specifies ESP32's built-in Preferences library:
```json
{
  "dependencies": {
    "Preferences": "*"
  }
}
```

PlatformIO incorrectly resolves this to `vshymanskyy/Preferences` from the registry instead of the native ESP32 framework library.

## Impact

- **Project Breakage**: Working projects can suddenly break when someone registers a library with a conflicting name (e.g., `Wire`, `SPI`, `Preferences`)
- **Unpredictable Builds**: Library resolution depends on external registry state rather than framework specifications

## Workaround

Specify exact versions that don't exist in external registry:
```json
{
  "dependencies": {
    "Preferences": "3.3.1"
  }
}
```

This is fragile and not a long-term solution: it will break if someone registers a library with the same name and with that version

## Possible Solutions

1. **Explicit Framework Syntax**: Add syntax to explicitly reference framework libraries:
   - `"framework:Preferences": "*"`
   - `"builtin:Preferences": "*"`

2. **Priority-based Resolution**: Built-in framework libraries should take precedence over registry libraries when names conflict

## References

Original discussion: https://community.platformio.org/t/how-to-explicitly-include-a-built-in-framework-library/52570

sylque, 2025-10-17:
Any news on this? It looks like a terrible issue. I'm tempted to register my own `Wire` and `SPI` library to see what happens...

ivankravets, 2026-09-05:
I can't reproduce the issue. Do you use the latest stable PIO Core? See the code that handles this https://github.com/platformio/platformio-core/blob/develop/platformio/package/manager/library.py#L89

I created a fake library with this `library.json`:
```json
...
  "dependencies": {
    "SPI": "*",
    "Preferences": "*"
  },
```

The SPI/Preferences were not installed.

If you still experience this issue, please create an independent PIO project and share it as ZIP/Github repository to reproduce it. Thanks.

closed 2026-09-05 by ivankravets